esphome/issues #1175: failed to parse protocol - http

tiagofreire-pt, 2020-04-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

PIP dev  

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32

**Affected component:**

```http_request```

**Description of problem:**

Error about the http url parsing.

**Problem-relevant YAML-configuration entries:**
```yaml
binary_sensor:
  - platform: esp32_touch
    name: "ESP32 Touch Pad GPIO27"
    pin: GPIO27
    threshold: 1000
    on_press:
      then:
        - http_request.post:
            url: http://online.oliveiradigital.net/user.php?cont=login&username=u123&password=u123&Submit=Login&md5=123
            verify_ssl: false  

```

**Logs (if applicable):**
```
[12:11:41][D][binary_sensor:036]: 'ESP32 Touch Pad GPIO27': Sending state ON
[12:11:41][E][HTTPClient.cpp:241] beginInternal(): failed to parse protocol
[12:11:41][E][HTTPClient.cpp:241] beginInternal(): failed to parse protocol
[12:11:41][W][http_request:032]: HTTP Request failed at the begin phase. Please check the configuration
```

**Additional information and things you've tried:**
N/A

glmnet, 2020-04-28:
@Anonym-tsk

Anonym-tsk, 2020-04-30:
Are you sure that you are using dev version? Please update. I think, this problem is already solved here https://github.com/esphome/esphome/pull/980

Anonym-tsk, 2020-04-30:
So, i reproduced these warnings on ESP32.

closed 2020-05-01 by glmnet