esphome/issues #1177: Slow response from INA3221

mboremski, 2020-04-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESP8266 / D1-Mini & ESP3221 via I2C

**Affected component:**

I tried to get my INA3221 running.
It is giving me data, but has really slow responde when setting update_interval to 1 or 2 seconds.

**Description of problem:**
Slow response. Data seem to be averaged.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: battery
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: *********
  password: *********

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: *********
    password: *********

ota:

web_server:
  port: 80

mqtt:
  broker: *********

i2c:
  - id: bus_a
    scan: true
    scl: 5
    sda: 4

sensor:
  - platform: ina3221
    i2c_id: bus_a
    address: 0x40
    channel_1:
      shunt_resistance: 0.1 ohm
      current:
        name: "Ch1 Current"
      power:
        name: "Ch1 Power"
      bus_voltage:
        name: "Ch1 Bus Voltage"
      shunt_voltage:
        name: "Ch1 Shunt Voltage"
    channel_2:
      shunt_resistance: 0.1 ohm
      current:
        name: "Ch2 Current"
      power:
        name: "Ch2 Power"
      bus_voltage:
        name: "Ch2 Bus Voltage"
      shunt_voltage:
        name: "Ch2 Shunt Voltage"
    channel_3:
      shunt_resistance: 0.1 ohm
      current:
        name: "Ch3 Current"
      power:
        name: "Ch3 Power"
      bus_voltage:
        name: "Ch3 Bus Voltage"
      shunt_voltage:
        name: "Ch3 Shunt Voltage"
    update_interval: 1s

```

**Additional information and things you've tried:**
Here you can see a response with 1s update-Interval with a load of aprox 3A switched ON/OFF on Ch1:
(1st running free / 2nd with a Reset of the node)
[image]

mboremski, 2020-04-30:
Can any body help me get averaging turned off?

brandond, 2020-04-30:
add `filters: []` to your component configuration.

mboremski, 2020-04-30:
Thank you.
I added this to my config like that:
````
sensor:
  - platform: ina3221
    i2c_id: bus_a
    address: 0x40
    channel_1:
      shunt_resistance: 1 ohm
      current:
        name: "Ch1 Current"
        filters: []
      power:
        name: "Ch1 Power"
        filters: []
      bus_voltage:
        name: "Ch1 Bus Voltage"
        filters: []
      shunt_voltage:
        name: "Ch1 Shunt Voltage"
        filters: []
````

Unfotunately there is no change in behavior of the values. They are still averaged.

Vxider, 2020-06-29:
This is caused by the default configuration of INA3221, it will use 1024 samples in average mode by default, you can change the `config` in `ina3221.cpp` (might be located in `python_location/site-packages/esphome/components`) according to [PR1102](https://github.com/esphome/esphome/pull/1102/files)

OttoWinter, 2020-07-13:
Closed in https://github.com/esphome/esphome/pull/1102

closed 2020-07-13 by OttoWinter