esphome/issues #1181: Crash when updating template binary sensor

declanshanaghy, 2020-04-30:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
ESP32

**Affected component:**

**Description of problem:**
Crash when updating template binary sensor

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  devicename: bbqsentry5
  topic_prefix: bbqsentry5
  devicename_friendly: BBQ Sentry5
  illumination_update_interval: 1s
  illumination_avg_window_size: '5'   # Average illumination readings over the last X readings
  illumination_send_every: '15'        # Multiply by interval to get how often the calculated illumination is sent
  temperature_update_interval: 1s
  temperature_avg_window_size: '30'   # Average temperature over the last X readings
  temperature_send_every: '15'        # Multiply by interval to get how often the calculated temperature is sent
  fire_illumination_level: '100'

esphome:
  name: "$devicename"
  platform: ESP32
  board: nodemcu-32s
  on_boot:
    priority: -10
    then:
      - switch.turn_off: chamberlights
      - switch.turn_off: diallights

<<: !include devices/common.yaml

spi:
  clk_pin: GPIO18
  miso_pin: GPIO19
  mosi_pin: GPIO23

i2c:
  scl: GPIO22
  sda: GPIO21

status_led:
  pin:
    number: GPIO32
    inverted: false

output:
  - platform: ledc
    pin: GPIO12
    id: buzzer

sensor:
  #  - platform: wifi_signal
  #    name: "$devicename_friendly Wifi Signal"
  #    update_interval: 60s

  - platform: uptime
    name: "$devicename_friendly Uptime"
    update_interval: 60s

  - platform: bh1750
    name: "$devicename_friendly Illumination"
    id: illumination
    address: 0x23
    resolution: 1.0
    icon: mdi:brightness-5
    unit_of_measurement: lx
    update_interval: ${illumination_update_interval}
    filters:
      - offset: 0
      - sliding_window_moving_average:
          window_size: ${illumination_avg_window_size}
          send_every: ${illumination_send_every}

  - platform: max31856
    name: "$devicename_friendly Probe 1 Temperature"
    icon: "mdi:hamburger"
    cs_pin: GPIO17
    update_interval: ${temperature_update_interval}
    filters:
      - offset: 0
      - sliding_window_moving_average:
          window_size: ${temperature_avg_window_size}
          send_every: ${temperature_send_every}

  - platform: max31856
    name: "$devicename_friendly Chamber Temperature"
    icon: "mdi:thermometer-lines"
    cs_pin: GPIO5
    update_interval: ${temperature_update_interval}
    filters:
      - offset: 0
      - sliding_window_moving_average:
          window_size: ${temperature_avg_window_size}
          send_every: ${temperature_send_every}
    on_value_range:
      # - above: 48.89 # 120F in C
      - above: 37.78 # 100F in C
        then:
          - switch.turn_on: diallights
      # - below: 48.889 # 120F in C
      - below: 37.78 # 100F in C
        then:
          - switch.turn_off: diallights

binary_sensor:
  - platform: status
    name: "$devicename_friendly Status"

  - platform: template
    name: "$devicename_friendly FIRE! &#128293;&#128293;"
    id: fire
    lambda: return id(hood).state && (id(illumination).state >= $fire_illumination_level);
    on_state:
      if:
        condition:
          binary_sensor.is_on: fire
        then:
          - output.ledc.set_frequency:
              id: buzzer
              frequency: 2000Hz

  - platform: gpio
    name: "$devicename_friendly Hood"
    id: hood
    device_class: opening
    pin:
      inverted: true
      number: GPIO13
      mode: INPUT_PULLUP
    on_state:
      then:
        - switch.toggle: chamberlights

  - platform: template
    name: "$devicename_friendly Chamber Lights"
    device_class: light
    # icon: "mdi:wall-sconce-flat"
    lambda: return id(chamberlights).state;

  - platform: template
    name: "$devicename_friendly Dial Lights"
    device_class: light
    # icon: "mdi:blur-radial"
    lambda: return id(diallights).state;

switch:
  - platform: gpio
    name: "$devicename_friendly Chamber Lights Switch"
    internal: true
    id: chamberlights
    pin:
      number: GPIO33

  - platform: gpio
    name: "$devicename_friendly Dial Lights Switch"
    internal: true
    id: diallights
    pin:
      number: GPIO25

```

**Logs (if applicable):**

```
[18:50:08][D][sensor:092]: 'BBQ Sentry5 Illumination': Sending state 2.33331 lx with 1 decimals of accuracy
[18:50:08][D][binary_sensor:036]: 'BBQ Sentry5 FIRE! &#128293;&#128293;': Sending state OFF
[18:50:08]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[18:50:08]Core 1 register dump:
[18:50:08]PC      : 0x400e649c  PS      : 0x00060c30  A0      : 0x800e653a  A1      : 0x3ffb1db0  
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d /Volumes/config/esphome/bbqsentry5 -t idedata
WARNING Decoded 0x400e649c: AsyncEventSourceClient::_runQueue() at /Users/dshanaghy/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
[18:50:20]A2      : 0x3ffd7204  A3      : 0x3ffdfff8  A4      : 0x89000000  A5      : 0x3ffd7300  
[18:50:20]A6      : 0x7d65736c  A7      : 0x0a0d0a0d  A8      : 0x800e64a8  A9      : 0x3ffb1d90  
[18:50:20]A10     : 0x00000001  A11     : 0x3ffd6040  A12     : 0x00000000  A13     : 0x4f983690  
[18:50:20]A14     : 0x00000000  A15     : 0x7061000a  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
[18:50:20]EXCVADDR: 0x00000009  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
[18:50:20]
[18:50:20]Backtrace: 0x400e649c:0x3ffb1db0 0x400e6537:0x3ffb1dd0 0x400e6557:0x3ffb1df0 0x400e65ad:0x3ffb1e10 0x400dafc3:0x3ffb1e40 0x4016b656:0x3ffb1e80 0x400d28c1:0x3ffb1ea0 0x400d28f1:0x3ffb1ed0 0x400da3a6:0x3ffb1ef0 0x4016b4dd:0x3ffb1f20 0x4016b5d5:0x3ffb1f40 0x400deba2:0x3ffb1f60 0x400e132a:0x3ffb1f90 0x400f2ead:0x3ffb1fb0 0x4008982d:0x3ffb1fd0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x400e649c: AsyncEventSourceClient::_runQueue() at /Users/dshanaghy/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
WARNING Decoded 0x400e6537: AsyncEventSourceClient::_queueMessage(AsyncEventSourceMessage*) at /Users/dshanaghy/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
WARNING Decoded 0x400e6557: AsyncEventSourceClient::write(char const*, unsigned int) at /Users/dshanaghy/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
WARNING Decoded 0x400e65ad: AsyncEventSource::send(char const*, char const*, unsigned int, unsigned int) at /Users/dshanaghy/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
WARNING Decoded 0x400dafc3: esphome::web_server::WebServer::on_binary_sensor_update(esphome::binary_sensor::BinarySensor*, bool) at /Volumes/config/esphome/bbqsentry5/src/esphome/components/web_server/web_server.cpp:63
WARNING Decoded 0x4016b656: std::_Function_handler<void (bool), esphome::Controller::setup_controller()::{lambda(bool)#1}>::_M_invoke(std::_Any_data const&, bool&&) at /Volumes/config/esphome/bbqsentry5/src/esphome/core/controller.cpp:11
 (inlined by) _M_invoke at /Users/dshanaghy/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1871
WARNING Decoded 0x400d28c1: std::function<void (bool)>::operator()(bool) const at /Volumes/config/esphome/bbqsentry5/src/esphome/components/binary_sensor/binary_sensor.cpp:66
 (inlined by) esphome::CallbackManager<void (bool)>::call(bool) at /Volumes/config/esphome/bbqsentry5/src/esphome/core/helpers.h:209
 (inlined by) esphome::binary_sensor::BinarySensor::send_state_internal(bool, bool) at /Volumes/config/esphome/bbqsentry5/src/esphome/components/binary_sensor/binary_sensor.cpp:41
WARNING Decoded 0x400d28f1: esphome::binary_sensor::BinarySensor::publish_state(bool) at /Volumes/config/esphome/bbqsentry5/src/esphome/components/binary_sensor/binary_sensor.cpp:66
WARNING Decoded 0x400da3a6: esphome::template_::TemplateBinarySensor::loop() at /Volumes/config/esphome/bbqsentry5/src/esphome/components/template/binary_sensor/template_binary_sensor.cpp:15
WARNING Decoded 0x4016b4dd: esphome::Component::call_loop() at /Volumes/config/esphome/bbqsentry5/src/esphome/core/component.cpp:61
WARNING Decoded 0x4016b5d5: esphome::Component::call() at /Volumes/config/esphome/bbqsentry5/src/esphome/core/component.cpp:82
WARNING Decoded 0x400deba2: esphome::Application::loop() at /Volumes/config/esphome/bbqsentry5/src/esphome/core/application.cpp:68
WARNING Decoded 0x400e132a: loop() at /Volumes/config/esphome/bbqsentry5/src/main.cpp:644
WARNING Decoded 0x400f2ead: loopTask(void*) at /Users/dshanaghy/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:19
WARNING Decoded 0x4008982d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

```

**Additional information and things you've tried:**

declanshanaghy, 2020-05-01:
Happened again:

```
[13:55:08][D][binary_sensor:036]: 'BBQ Sentry5 Fire Alarm': Sending state ON
[13:55:08][W][main:963]: I'm on fire!! Temperature is 111 °F. Illumination is 56 lx
[13:55:08]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[13:55:08]Core 1 register dump:
[13:55:08]PC      : 0x400f093a  PS      : 0x00060c30  A0      : 0x800f09da  A1      : 0x3ffb1db0  
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d /Volumes/config/esphome/bbqsentry5 -t idedata
WARNING Decoded 0x400f093a: AsyncEventSourceClient::_runQueue() at /Users/dshanaghy/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
[13:55:20]A2      : 0x3ffd7c64  A3      : 0x00000001  A4      : 0x0000005f  A5      : 0x3ffd81c6  
[13:55:20]A6      : 0x0000000a  A7      : 0x72743a22  A8      : 0x800f0948  A9      : 0x3ffb1d90  
[13:55:20]A10     : 0x0000005f  A11     : 0x0000005f  A12     : 0x0000005f  A13     : 0x0000005f  
[13:55:20]A14     : 0x00000000  A15     : 0x007d6575  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
[13:55:20]EXCVADDR: 0x00000001  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
[13:55:20]
[13:55:20]Backtrace: 0x400f093a:0x3ffb1db0 0x400f09d7:0x3ffb1dd0 0x400f09f7:0x3ffb1df0 0x400f0a4d:0x3ffb1e10 0x400e171b:0x3ffb1e40 0x40175926:0x3ffb1e80 0x400d84d9:0x3ffb1ea0 0x400d8509:0x3ffb1ed0 0x400e043e:0x3ffb1ef0 0x401757ad:0x3ffb1f20 0x401758a5:0x3ffb1f40 0x400e5862:0x3ffb1f60 0x400e84ae:0x3ffb1f90 0x400fd475:0x3ffb1fb0 0x4008982d:0x3ffb1fd0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x400f093a: AsyncEventSourceClient::_runQueue() at /Users/dshanaghy/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
WARNING Decoded 0x400f09d7: AsyncEventSourceClient::_queueMessage(AsyncEventSourceMessage*) at /Users/dshanaghy/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
WARNING Decoded 0x400f09f7: AsyncEventSourceClient::write(char const*, unsigned int) at /Users/dshanaghy/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
WARNING Decoded 0x400f0a4d: AsyncEventSource::send(char const*, char const*, unsigned int, unsigned int) at /Users/dshanaghy/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
WARNING Decoded 0x400e171b: esphome::web_server::WebServer::on_binary_sensor_update(esphome::binary_sensor::BinarySensor*, bool) at /Volumes/config/esphome/bbqsentry5/src/esphome/components/web_server/web_server.cpp:63
WARNING Decoded 0x40175926: std::_Function_handler<void (bool), esphome::Controller::setup_controller()::{lambda(bool)#1}>::_M_invoke(std::_Any_data const&, bool&&) at /Volumes/config/esphome/bbqsentry5/src/esphome/core/controller.cpp:11
 (inlined by) _M_invoke at /Users/dshanaghy/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1871
WARNING Decoded 0x400d84d9: std::function<void (bool)>::operator()(bool) const at /Users/dshanaghy/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:2271
 (inlined by) esphome::CallbackManager<void (bool)>::call(bool) at /Volumes/config/esphome/bbqsentry5/src/esphome/core/helpers.h:209
 (inlined by) esphome::binary_sensor::BinarySensor::send_state_internal(bool, bool) at /Volumes/config/esphome/bbqsentry5/src/esphome/components/binary_sensor/binary_sensor.cpp:41
WARNING Decoded 0x400d8509: esphome::binary_sensor::BinarySensor::publish_state(bool) at /Volumes/config/esphome/bbqsentry5/src/esphome/components/binary_sensor/binary_sensor.cpp:18
WARNING Decoded 0x400e043e: esphome::template_::TemplateBinarySensor::loop() at /Volumes/config/esphome/bbqsentry5/src/esphome/components/template/binary_sensor/template_binary_sensor.cpp:15
WARNING Decoded 0x401757ad: esphome::Component::call_loop() at /Volumes/config/esphome/bbqsentry5/src/esphome/core/component.cpp:61
WARNING Decoded 0x401758a5: esphome::Component::call() at /Volumes/config/esphome/bbqsentry5/src/esphome/core/component.cpp:82
WARNING Decoded 0x400e5862: esphome::Application::loop() at /Volumes/config/esphome/bbqsentry5/src/esphome/core/application.cpp:68
WARNING Decoded 0x400e84ae: loop() at /Volumes/config/esphome/bbqsentry5/src/main.cpp:1009
WARNING Decoded 0x400fd475: loopTask(void*) at /Users/dshanaghy/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:19
WARNING Decoded 0x4008982d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

glmnet, 2020-05-01:
This is related to the web_server component, looks like you are monitoring the binary sensor with the web_server page, not the HA api or mqtt, the bug is related to this library https://github.com/me-no-dev/ESPAsyncWebServer may be there is a bug in ESPHome miss using this library or a bug in the library itself.

So far this info should help you avoid getting this problem if you don't need to monitor the sensor via web server continuously.

Thanks for the report

egrekov, 2020-11-12:
Hi!
I see the same error in myself.
```c++
[15:36:24]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[15:36:24]Core 1 register dump:
[15:36:24]PC      : 0x400e01f6  PS      : 0x00060230  A0      : 0x800e0296  A1      : 0x3ffce5b0  
WARNING Decoded 0x400e01f6: AsyncEventSourceClient::_runQueue() at /root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
[15:36:24]A2      : 0x3fff5964  A3      : 0x00000002  A4      : 0x00000053  A5      : 0x3fff500a  
[15:36:24]A6      : 0x0000000a  A7      : 0x0d7d362e  A8      : 0x800e0204  A9      : 0x3ffce590  
[15:36:24]A10     : 0x00000053  A11     : 0x00000053  A12     : 0x00000053  A13     : 0x3fff5918  
[15:36:24]A14     : 0x00000000  A15     : 0x31333a22  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
[15:36:24]EXCVADDR: 0x00000002  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
[15:36:24]
[15:36:24]Backtrace: 0x400e01f6:0x3ffce5b0 0x400e0293:0x3ffce5d0 0x400e02b3:0x3ffce5f0 0x400e0309:0x3ffce610 0x400d63a5:0x3ffce640 0x401e1e29:0x3ffce680 0x400d59e5:0x3ffce6a0 0x400d5a6e:0x3ffce6d0 0x400d5ad1:0x3ffce730 0x400da3e2:0x3ffce760 0x400da405:0x3ffce830 0x400d355f:0x3ffce850 0x401e1d25:0x3ffce8f0 0x401e1db9:0x3ffce910 0x400da532:0x3ffce930 0x400dc662:0x3ffce960 0x400e87ed:0x3ffce980 0x4008e8c1:0x3ffce9a0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x400e01f6: AsyncEventSourceClient::_runQueue() at /root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
WARNING Decoded 0x400e0293: AsyncEventSourceClient::_queueMessage(AsyncEventSourceMessage*) at /root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
WARNING Decoded 0x400e02b3: AsyncEventSourceClient::write(char const*, unsigned int) at /root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
WARNING Decoded 0x400e0309: AsyncEventSource::send(char const*, char const*, unsigned int, unsigned int) at /root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
WARNING Decoded 0x400d63a5: esphome::web_server::WebServer::on_sensor_update(esphome::sensor::Sensor*, float) at /config/esphome/gateway_1/src/esphome/components/web_server/web_server.cpp:63
WARNING Decoded 0x401e1e29: std::_Function_handler<void (float), esphome::Controller::setup_controller()::{lambda(float)#1}>::_M_invoke(std::_Any_data const&, float&&) at /config/esphome/gateway_1/src/esphome/core/controller.cpp:29
 (inlined by) _M_invoke at /root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1871
WARNING Decoded 0x400d59e5: std::function<void (float)>::operator()(float) const at /root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_vector.h:932
WARNING Decoded 0x400d5a6e: esphome::CallbackManager<void (float)>::call(float) at /root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_vector.h:932
 (inlined by) esphome::sensor::Sensor::internal_send_state_to_frontend(float) at /config/esphome/gateway_1/src/esphome/components/sensor/sensor.cpp:93
WARNING Decoded 0x400d5ad1: esphome::sensor::Sensor::publish_state(float) at /root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_vector.h:932
WARNING Decoded 0x400da3e2: esphome::xiaomi_lywsdcgq::XiaomiLYWSDCGQ::parse_device(esphome::esp32_ble_tracker::ESPBTDevice const&) at /config/esphome/gateway_1/src/esphome/components/xiaomi_lywsdcgq/xiaomi_lywsdcgq.cpp:47
WARNING Decoded 0x400da405: non-virtual thunk to esphome::xiaomi_lywsdcgq::XiaomiLYWSDCGQ::parse_device(esphome::esp32_ble_tracker::ESPBTDevice const&)
WARNING Decoded 0x400d355f: esphome::esp32_ble_tracker::ESP32BLETracker::loop() at /config/esphome/gateway_1/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp:241
WARNING Decoded 0x401e1d25: esphome::Component::call_loop() at /config/esphome/gateway_1/src/esphome/core/component.cpp:156
WARNING Decoded 0x401e1db9: esphome::Component::call() at /config/esphome/gateway_1/src/esphome/core/component.cpp:156
WARNING Decoded 0x400da532: esphome::Application::loop() at /config/esphome/gateway_1/src/esphome/core/application.cpp:72
WARNING Decoded 0x400dc662: loop() at /config/esphome/gateway_1/src/main.cpp:253
WARNING Decoded 0x400e87ed: loopTask(void*) at /root/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:19
WARNING Decoded 0x4008e8c1: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
[15:36:25]
[15:36:25]Rebooting...
```

egrekov, 2020-11-12:
By the way, I would like to note that in one of my projects it was also noticed that events do not work as stably as websockets, I think if migrating to websockets this will solve this problem.

sjtrny, 2021-01-03:
I'm experiencing a similar crash when using https://github.com/trvrnrth/esphome-bsec-bme680/

```
[02:03:56][I][logger:166]: Log initialized
[02:03:56][I][app:029]: Running through setup()...
[02:03:56][I][wifi:194]: WiFi Connecting to '15 Rochdale'...
[02:03:59][I][wifi:457]: WiFi Connected!
[02:03:59][I][app:059]: setup() finished successfully!
[02:03:59][I][app:105]: ESPHome version 1.15.3 compiled on Jan  3 2021, 02:02:37
[02:03:59][I][i2c:033]: Scanning i2c bus for active devices...
[02:03:59][I][i2c:040]: Found i2c device at address 0x76
[02:05:56][I][ota:046]: Boot seems successful, resetting boot loop counter.
[02:51:34]Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
[02:51:34]Core 1 register dump:
[02:51:34]PC      : 0x400e6194  PS      : 0x00060930  A0      : 0x800e622a  A1      : 0x3ffb1cf0  
INFO Need to fetch platformio IDE-data, please stand by
INFO Running:  platformio run -d co2 -t idedata
WARNING Decoded 0x400e6194: AsyncEventSourceClient::_runQueue() at /Users/steve/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
[02:51:35]A2      : 0x3ffb9878  A3      : 0x5e43420f  A4      : 0x00000000  A5      : 0x3ffb2cf8  
[02:51:35]A6      : 0x0a0d0a0d  A7      : 0x7d323935  A8      : 0x00000000  A9      : 0x3ffb1cd0  
[02:51:35]A10     : 0x00000000  A11     : 0x00000054  A12     : 0x00000054  A13     : 0x3ffb2664  
[02:51:35]A14     : 0x00000000  A15     : 0x000a0d0a  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
[02:51:35]EXCVADDR: 0x00000008  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
[02:51:35]
[02:51:35]Backtrace: 0x400e6194:0x3ffb1cf0 0x400e6227:0x3ffb1d10 0x400e6243:0x3ffb1d30 0x400e6299:0x3ffb1d50 0x400db519:0x3ffb1d80 0x4016d999:0x3ffb1dc0 0x400da291:0x3ffb1de0 0x400da2b7:0x3ffb1e10 0x400d907a:0x3ffb1e40 0x400d90a2:0x3ffb1e60 0x400da304:0x3ffb1e90 0x400d6ce1:0x3ffb1ec0 0x400d70af:0x3ffb1ee0 0x4016d831:0x3ffb1f20 0x4016d8fd:0x3ffb1f40 0x400ded82:0x3ffb1f60 0x400e1262:0x3ffb1f90 0x400efbb1:0x3ffb1fb0 0x400896ad:0x3ffb1fd0
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x400e6194: AsyncEventSourceClient::_runQueue() at /Users/steve/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
WARNING Decoded 0x400e6227: AsyncEventSourceClient::_queueMessage(AsyncEventSourceMessage*) at /Users/steve/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
WARNING Decoded 0x400e6243: AsyncEventSourceClient::write(char const*, unsigned int) at /Users/steve/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
WARNING Decoded 0x400e6299: AsyncEventSource::send(char const*, char const*, unsigned int, unsigned int) at /Users/steve/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1858
WARNING Decoded 0x400db519: esphome::web_server::WebServer::on_sensor_update(esphome::sensor::Sensor*, float) at /Users/steve/Desktop/co2/co2/src/esphome/components/web_server/web_server.cpp:63
WARNING Decoded 0x4016d999: std::_Function_handler<void (float), esphome::Controller::setup_controller()::{lambda(float)#2}>::_M_invoke(std::_Any_data const&, float&&) at /Users/steve/Desktop/co2/co2/src/esphome/core/controller.cpp:29
 (inlined by) _M_invoke at /Users/steve/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1871
WARNING Decoded 0x400da291: std::function<void (float)>::operator()(float) const at /Users/steve/Desktop/co2/co2/src/esphome/components/sensor/sensor.cpp:85
WARNING Decoded 0x400da2b7: esphome::CallbackManager<void (float)>::call(float) at /Users/steve/Desktop/co2/co2/src/esphome/components/sensor/sensor.cpp:85
 (inlined by) esphome::sensor::Sensor::internal_send_state_to_frontend(float) at /Users/steve/Desktop/co2/co2/src/esphome/components/sensor/sensor.cpp:93
WARNING Decoded 0x400d907a: esphome::sensor::Filter::output(float) at /Users/steve/Desktop/co2/co2/src/esphome/components/sensor/filter.cpp:83
WARNING Decoded 0x400d90a2: esphome::sensor::Filter::input(float) at /Users/steve/Desktop/co2/co2/src/esphome/components/sensor/filter.cpp:83
WARNING Decoded 0x400da304: esphome::sensor::Sensor::publish_state(float) at /Users/steve/Desktop/co2/co2/src/esphome/components/sensor/sensor.cpp:85
WARNING Decoded 0x400d6ce1: esphome::bme680_bsec::BME680BSECComponent::publish_state_(esphome::sensor::Sensor*, float) at /Users/steve/Desktop/co2/co2/src/esphome/components/bme680_bsec/bme680_bsec.cpp:117
WARNING Decoded 0x400d70af: esphome::bme680_bsec::BME680BSECComponent::loop() at /Users/steve/Desktop/co2/co2/src/esphome/components/bme680_bsec/bme680_bsec.cpp:107
WARNING Decoded 0x4016d831: esphome::Component::call_loop() at /Users/steve/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/functional:1754
WARNING Decoded 0x4016d8fd: esphome::Component::call() at /Users/steve/Desktop/co2/co2/src/esphome/core/component.cpp:82
WARNING Decoded 0x400ded82: esphome::Application::loop() at /Users/steve/Desktop/co2/co2/src/esphome/core/application.cpp:72
WARNING Decoded 0x400e1262: loop() at /Users/steve/Desktop/co2/co2/src/main.cpp:381
WARNING Decoded 0x400efbb1: loopTask(void*) at /Users/steve/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:19
WARNING Decoded 0x400896ad: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
[02:51:35]
[02:51:35]Rebooting...
```

closed 2021-05-18 by stale[bot]

marrobHD, 2021-06-29:
The same with bme680_bsec without webserver enabled. Issue with interaction of components?
```
[0;32m[I][logger:170]: Log initialized[0m
[0;35m[C][status_led:014]: Setting up Status LED...[0m
[0;35m[C][ota:366]: There have been 8 suspected unsuccessful boot attempts.[0m
[0;32m[I][app:029]: Running through setup()...[0m
[0;35m[C][uart_esp32:072]: Setting up UART...[0m
[0;35m[C][fastled:010]: Setting up FastLED light...[0m
[0;35m[C][switch.gpio:011]: Setting up GPIO Switch 'Weather Station resistance bias'...[0m
[0;36m[D][switch:025]: 'Weather Station resistance bias' Turning OFF.[0m
[0;36m[D][switch:045]: 'Weather Station resistance bias': Sending state OFF[0m
[0;36m[D][switch:025]: 'Weather Station resistance bias' Turning OFF.[0m
[0;35m[C][light:097]: Setting up light 'Weather Station status light'...[0m
[0;36m[D][light:275]: 'Weather Station status light' Setting:[0m
[0;36m[D][light:288]:   Brightness: 100%[0m
[0;36m[D][light:297]:   Red=100%, Green=100%, Blue=100%[0m
[0;35m[C][bme680_bsec.sensor:016]: Setting up BME680 via BSEC...[0m
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (loopTask) 
Core 1 register dump:
PC      : 0x401284d7  PS      : 0x00060d36  A0      : 0x80124241  A1      : 0x3ffb0060  
A2      : 0x3ffb80c0  A3      : 0x3ffb03a0  A4      : 0x3ffb0284  A5      : 0x00000010  
A6      : 0x00000003  A7      : 0x3ffb02c4  A8      : 0xa0000000  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x10000000  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x7ff00000  A15     : 0x7e37e43c  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x401284d7:0x3ffb0060 0x4012423e:0x3ffb0090 0x40120b51:0x3ffb03a0 0x400ea95e:0x3ffb0460 0x400db479:0x3ffb04d0 0x400db622:0x3ffb0da0 0x400db666:0x3ffb0f90 0x40187159:0x3ffb0fb0 0x40187215:0x3ffb0fd0 0x400e9d71:0x3ffb0ff0 0x400f2d8a:0x3ffb1040 0x40104946:0x3ffb1fb0 0x4008b472:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[0;32m[I][logger:170]: Log initialized[0m
```