esphome/issues #1185: Setting sensor offset (possibly negative) prevents connecting to WiFi

haplm, 2020-04-30:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io (HassOS 3.13, HAC 0.108.9) running on RaspberryPI 4B

**ESP (ESP32/ESP8266, Board/Sonoff):**

[NodeMCU V3 340G Lua WIFI (ESP8266 with 32M Flash)](https://www.banggood.com/NodeMCU-V3-340G-Lua-WIFI-Module-Integration-Of-ESP8266-Extra-Memory-32M-Flash-p-1175347.html?utm_design=41&utm_source=emarsys&utm_medium=Neworder171109&utm_campaign=trigger-emarsys&utm_content=winna&sc_src=email_2675773&sc_eh=c772821e15f48e491&sc_llid=20101026&sc_lid=105229698&sc_uid=PD5TUtQCmO&cur_warehouse=CN)
This board is reported to have similar issue here #952, but that issue is driven by static IP configuration, which is not my case. With that said, there'll be definitely something fishy about the board, as ESP Easy is not able to connect to WiFi at all.

**Affected component:**

https://esphome.io/components/wifi.html
https://esphome.io/components/sensor/dht.html

**Description of problem:**

Although it may sound strange, I have a board that can't connect to wifi when I set a sensor offset to -4.0. It says 'Probe Request Unsuccessful'. If I remove the offset, everything is fine.

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
  - platform: mhz19
    co2:
      name: "Kitchen CO2"
      id: kitchen_co2
    temperature:
      name: "Kitchen Temperature [MS]"
      filters:
        - offset: -4.0
    update_interval: 60s
    id: kitchen_mhz19b
    automatic_baseline_calibration: false

  - platform: dht
    pin: D2
    model: DHT22
    temperature:
      name: "Kitchen Temperature"
      filters:
        - offset: -4.0
    humidity:
      name: "Kitchen Humidity"
    update_interval: 60s

```

**Logs (if applicable):**

```
17:27:40][C][ota:364]: There have been 1 suspected unsuccessful boot attempts.
[17:27:40][I][app:028]: Running through setup()...
[17:27:40][C][uart:111]: Setting up UART bus...
[17:27:40][D][mhz19:075]: MHZ19 Disabling automatic baseline calibration
[17:27:40][C][dht:011]: Setting up DHT...
[17:27:40][C][wifi:033]: Setting up WiFi...
[17:27:40][D][wifi:304]: Starting scan...
[17:27:40][W][mhz19:040]: Invalid preamble from MHZ19!
[17:27:40][D][dht:048]: Got Temperature=28.1°C Humidity=37.3%
[17:27:40][D][sensor:092]: 'Kitchen Temperature': Sending state 24.10000 °C with 1 decimals of accuracy
[17:27:40][D][sensor:092]: 'Kitchen Humidity': Sending state 37.30000 % with 0 decimals of accuracy
[17:27:46][D][wifi:319]: Found networks:
[17:27:46][I][wifi:365]: - 'Zoo' (AC:9E:17:47:29:30) ▂▄▆█
[17:27:46][D][wifi:366]:     Channel: 11
[17:27:46][D][wifi:367]:     RSSI: -41 dB
[17:27:46][D][wifi:369]: - 'UPC3705200' (34:2C:C4:AA:28:98) ▂▄▆█
[17:27:46][D][wifi:369]: - 'DIRECT-3E-HP DeskJet 5200 series' (9E:7B:EF:FA:FF:3E) ▂▄▆█
[17:27:46][D][wifi:369]: - 'UPC Wi-Free' (36:2C:94:AA:28:98) ▂▄▆█
[17:27:46][D][wifi:369]: - 'ZasedackaD9' (E4:BE:ED:79:37:4D) ▂▄▆█
[17:27:46][D][wifi:369]: - 'M24-CCTV' (E4:8D:8C:EA:5D:19) ▂▄▆█
[17:27:46][D][wifi:369]: - 'V2' (88:F7:C7:9E:95:81) ▂▄▆█
[17:27:46][D][wifi:369]: - 'mu-guest' (40:E3:D6:B5:1D:A1) ▂▄▆█
[17:27:46][D][wifi:369]: - 'mu' (40:E3:D6:B5:1D:A0) ▂▄▆█
[17:27:46][D][wifi:369]: - 'GREEN DECOR #2' (74:C1:4F:5E:93:28) ▂▄▆█
[17:27:46][D][wifi:369]: - 'UPC Wi-Free' (3A:43:1D:A4:99:14) ▂▄▆█
[17:27:46][D][wifi:369]: - 'UPC6854865' (38:43:7D:A4:99:14) ▂▄▆█
[17:27:46][D][wifi:369]: - 'UPCF98A277' (AC:22:05:B1:34:D8) ▂▄▆█
[17:27:46][D][wifi:369]: - 'UPC Wi-Free' (AE:22:15:B1:34:D8) ▂▄▆█
[17:27:46][D][wifi:369]: - 'mu' (90:4C:81:F3:D3:00) ▂▄▆█
[17:27:46][D][wifi:369]: - 'UPC Wi-Free' (8A:F7:C7:9E:95:83) ▂▄▆█
[17:27:46][D][wifi:369]: - 'UPC2011651' (34:2C:C4:5D:BC:A8) ▂▄▆█
[17:27:46][D][wifi:369]: - 'UPC3FEB75E' (DC:53:7C:DB:9B:96) ▂▄▆█
[17:27:46][D][wifi:369]: - 'dejvicka30' (5C:F4:AB:0E:8F:CB) ▂▄▆█
[17:27:46][D][wifi:369]: - 'Internet_28' (5C:F4:AB:0F:03:2B) ▂▄▆█
[17:27:46][D][wifi:369]: - 'jatom' (C0:C1:C0:E9:24:4B) ▂▄▆█
[17:27:46][D][wifi:369]: - 'UPC Wi-Free' (DE:53:1C:DB:9B:96) ▂▄▆█
[17:27:46][D][wifi:369]: - 'HUAWEI-B315-02D2' (C4:F0:81:24:02:D2) ▂▄▆█
[17:27:46][D][wifi:369]: - 'BARUSKA-NB_Network' (64:66:B3:E9:F4:7C) ▂▄▆█
[17:27:46][D][wifi:369]: - 'UPC Wi-Free' (56:67:11:9E:0A:FB) ▂▄▆█
[17:27:46][I][wifi:193]: WiFi Connecting to 'Zoo'...
[17:27:52][W][wifi_esp8266:334]: Event: Disconnected ssid='Zoo' reason='Probe Request Unsuccessful'
[17:27:52][W][wifi:451]: Error while connecting to network.
[17:27:52][W][wifi:488]: Restarting WiFi adapter...
[17:27:52][I][wifi:193]: WiFi Connecting to 'Zoo'...
[17:27:58][W][wifi_esp8266:334]: Event: Disconnected ssid='Zoo' reason='Probe Request Unsuccessful'
[17:27:58][W][wifi:451]: Error while connecting to network.
[17:27:58][W][wifi:488]: Restarting WiFi adapter...
[17:28:03][D][wifi:304]: Starting scan...

```

**Additional information and things you've tried:**

popsodav, 2020-05-05:
Have the same issue, but using DS18B20 sensor.
`dallas:
  - pin: D5

sensor:
  - platform: dallas
    index: 0
    name: "Температура в сарае"`

After commenting theese lines, issues is not happening.

popsodav, 2020-05-13:
Also this issue happened when I added AP-mode and LED status.
It seems that the problem with espressif SDK version.
I've changed platform to espressif8266@1.8.0 and everything works fine.

haplm, 2020-05-13:
How can I change the platform to espressif8266@1.8.0? I have ESPHome checked out on my mac, so I can certainly play with some configuration options... Thanks!

haplm, 2020-05-13:
BTW - isn't current version something like 2.5?

popsodav, 2020-05-16:
I did it im YAML file:
`esphome:
  platform: ESP8266
  board: nodemcuv2
  arduino_version: "espressif8266@1.8.0"`

But it seems that this is a not a universal recipe. I have several devices with original NodeMCU board, which also have the same issue in 1.8.0.

closed 2020-09-27 by stale[bot]