esphome/issues #1189: Reset Info: Fatal exception:4 flag:3 (SOFT_WDT)

lpbreault, 2020-05-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io on Ubuntu 18.04.3 LTS
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 d1_mini
**Affected component:**

I don't know precisely.

**Description of problem:**
I have 22 ESP8266 on ESPHome in my house. I removed some unused text sensors yesterday on all of them and proceed to flash them all. Since then, all my esp's are rebooting after few minutes with a watchdog reset. This is really weird and it didn't occurred before. The other thing that I changed was to update HA to 109.0 before doing my modifications.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: brlo04_bathroom_nightlight
  platform: ESP8266
  board: d1_mini
  arduino_version: espressif8266@2.2.3
  includes: []
  esp8266_restore_from_flash: false
  build_path: brlo04_bathroom_nightlight
  libraries: []
  platformio_options: {}
  board_flash_mode: dout
wifi:
  use_address: 172.23.2.199
  ap:
    ssid: Bathroom Nightlight AP
    password: XXXX-REMOVED-XXX
    ap_timeout: 1min
  output_power: 20.0
  domain: .local
  reboot_timeout: 15min
  power_save_mode: NONE
  fast_connect: false
  networks:
  - ssid: XXXX-REMOVED-XXX
    password: XXXX-REMOVED-XXX
    priority: 0.0
captive_portal: {}
api:
  reboot_timeout: 15min
  password: ''
  port: 6053
ota:
  password: ''
  safe_mode: true
  port: 8266
logger:
  level: VERBOSE
  tx_buffer_size: 512
  logs: {}
  baud_rate: 115200
  hardware_uart: UART0
  esp8266_store_log_strings_in_flash: true
debug: {}
binary_sensor:
- platform: status
  name: Bathroom Nightlight Status
  device_class: connectivity
status_led:
  pin:
    number: 2
    inverted: true
    mode: OUTPUT
light:
- platform: neopixelbus
  type: GRBW
[... 477 lines cut ...]
[19:17:32]3ffffe70:  ffffffff 00000000 00000001 3fffbfd0  
[19:17:32]3ffffe80:  00000000 3fffc6fc 00000000 00000010  
[19:17:32]3ffffe90:  3fff0dd0 00119ae0 00119adf 00000030  
[19:17:32]3ffffea0:  00000000 3fff252c 00119ace 00000030  
[19:17:32]3ffffeb0:  00000000 00000000 3fff217c 00000001  
[19:17:32]3ffffec0:  00000000 4bc6a7f0 00119ae0 00119adf  
[19:17:32]3ffffed0:  3fff254c 00000002 3fff20f4 40247b54  
[19:17:32]3ffffee0:  00000000 4bc6a7f0 00119ae0 40220d44  
[19:17:32]3ffffef0:  00000000 00000000 4bc6a7f0 40250579  
[19:17:32]3fffff00:  00000000 4bc6a7f0 00119ae0 00119adf  
[19:17:32]3fffff10:  4010509c 157d7d94 3fff0dd0 00000000  
[19:17:32]3fffff20:  3fff0440 3fff0dd0 3ffe853c 3fff0dd0  
[19:17:32]3fffff30:  00119ae0 00119adf 40220228 3fffefa0  
[19:17:32]3fffff40:  3fff0dd0 00119ae0 00000010 40220803  
[19:17:32]3fffff50:  3fff2550 00000010 3fff0a4c 4020e214  
[19:17:32]3fffff60:  00000000 00000000 3fff23d8 3fff24d4  
[19:17:32]3fffff70:  3fff20f4 3fff2550 feefeffe feefeffe  
[19:17:32]3fffff80:  3fff27f4 3fff24d4 feefeffe 3fff0d74  
[19:17:32]3fffff90:  3fffdad0 00000000 3fff0d44 40210d4c  
[19:17:32]3fffffa0:  3fffdad0 00000000 3fff0d44 402202d8  
[19:17:32]<<<stack<<<

```

**Additional information and things you've tried:**

lpbreault, 2020-05-06:
Ok! I manage to solve the problem for me. I forced compilation with Arduino SDK 2.4.2 and everything came back to normal. When I compiled on the 29 April, ESPHome automatically selected Arduino SDK 2.5.2! But I found that normally, ESPHome 1.14.3 should use 2.4.2? I maybe did not understand correctly that part... but... eh! Everything is stable now.

If you intended to really use 2.5.2, this is gonna have to be addressed because something is making ESPHome crash on devices every 20-ish minutes... and it is really annoying!

Hope every one is having a good day and thank you for all you commitment to this project! I love it.

julianhoare, 2020-06-02:
Many thanks for this, I had the same issue.

Sonoff 4ch Pro R2, ESPHome 1.14.3, HA 110.3.

Standard config for the Sonoff 4ch (from https://esphome.io/devices/sonoff_4ch.html) had no issues.  When adding adc, ntc and resitance components, I had random reboots with the following in the logs:

```
[15:58:30][D][debug:196]: Reset Reason: Software Watchdog
[15:58:30][D][debug:197]: Reset Info: Fatal exception:4 flag:3 (SOFT_WDT) epc1:0x40221477 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000000 depc:0x00000000
```

After forcing arduino_version: 2.4.2, seems stable.

closed 2020-10-14 by stale[bot]