esphome/issues #1190: Setting up SPI bus... fails  - Waveshare 7.5  E Paper

phiwa99, 2020-05-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

https://esphome.io/components/display/waveshare_epaper.html?highlight=waveshare

**Description of problem:**

I try to get a waveshare 7.5in E Paper Display to work with ESPHome (ESPHome 1.14.3) in HassIO.
After compiling the code I get an error caused by the SPI component (I think so).

I updated my ESPHome yesterday from 1.13.6 to 1.14.3 because I had connecting problems with my esp8266 devices
#455 

With Esphome 1.13.6 the waveshare display worked, but I had no connection to my wifi.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: wemos_d1_epaper
  platform: ESP8266
  board: d1_mini
  #arduino_version: 2.5.0
  

  
wifi:
  ssid: "FRITZ!Box 7490"
  password: "xxxxx"

# Enable logging
logger:
  level: VERBOSE

# OTA
ota:
  password: "xxxxx"
# Enable Home Assistant API
api:
  password: "xxxxx"

# Display 640x384 ##############################################################    

display:
  - platform: waveshare_epaper
    cs_pin: D8
    dc_pin: D3
    busy_pin: D2
    reset_pin: D4
    model: 7.50in
    # rotation: 180°
    # full_update_every: 30
    # update_interval: 3600s
    lambda: |-
      it.print(0, 0, id(temp_font), "E Paper Display");
      it.print(330, 200, id(temp_font), "unten");
      /* Linie Mitte horizontal */
      it.line(320, 0, 320, 384);
      /*Linie Mitte waagerecht  */
      it.line(0, 192, 640, 192);      
      
################################################################################
#switch:
#  - platform: gpio
#    pin:
#      number: D4
#      inverted: True
#    name: "LED_EPAPER"
[... 611 lines cut ...]
[09:32:13][C][wifi:297]:   Gateway: 192.168.178.1
[09:32:13][C][wifi:298]:   DNS1: 192.168.178.1
[09:32:13][C][wifi:299]:   DNS2: (IP unset)
[09:32:13][C][logger:175]: Logger:
[09:32:13][C][logger:176]:   Level: VERBOSE
[09:32:13][C][logger:177]:   Log Baud Rate: 115200
[09:32:13][C][logger:178]:   Hardware UART: UART0
[09:32:13][C][ota:029]: Over-The-Air Updates:
[09:32:13][C][ota:030]:   Address: wemos_d1_epaper.local:8266
[09:32:13][C][ota:032]:   Using Password.
[09:32:13][W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 0 restarts
[09:32:13]pm open,type:0 0

```

**Additional information and things you've tried:**

With the ESPHome 1.13.6 the Display worked, Wifi not
I updated ESPHome to 1.14.3. - Wifi worked, SPI not
I also installed ESPHome 1.15.0-dev and tried it.

I changed the code with different options.
`arduino_version: 2.3.0`

If i compile an sample code with the Arduino IDE and upload it to the Wemos D1 mini the display works.

sredfern, 2020-06-24:
Hey @phiwa99 - Do you know if its a v1 or v2 display? A new version was released and is incompatible with the 7.5in. 

See: https://github.com/esphome/issues/issues/1296 

New display: http://waveshare.com/w/upload/6/60/7.5inch_e-Paper_V2_Specification.pdf

phiwa99, 2020-06-24:
Hey
I have the 7.5inch e-Paper HAT (C) (white, black, yellow) Version 1 (640x384 resolution)

It works with the esp32. Just the esp8266 does not work.

closed 2020-12-05 by stale[bot]