esphome/issues #1192: Latest Dev docker container doesn't save the config before upload

poldim, 2020-05-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Latest development docker container

**ESP (ESP32/ESP8266, Board/Sonoff):**

N/A

**Affected component:**

GUI / upload mechanics

**Description of problem:**
Historically, if you made edits to your config, hitting the upload button would first save the config then move to the upload process.
The latest development docker container **does not save** the config before upload.  

**Problem-relevant YAML-configuration entries:**

N/A
**Logs (if applicable):**

N/A

**Additional information and things you've tried:**

jonathanadams, 2020-05-02:
@poldim fix will be coming for this soon.

poldim, 2020-06-13:
@jonathanadams, Is there any hack I can do on my end to make it save? it's bitten me in the ass a dozen times now...old habits die hard

jonathanadams, 2020-07-01:
> @jonathanadams, Is there any hack I can do on my end to make it save? it's bitten me in the ass a dozen times now...old habits die hard

Apologies it has took me a while for a fix - pull request submitted.

This is the line of code that fixes:
https://github.com/esphome/esphome/pull/1107/files#diff-8236425e0fc2a2a7cc7f0fda60e13740R738

If you have access to the sourcecode on your install you can add this line.

closed 2020-07-01 by glmnet