esphome/issues #1193: ESP32 compile crashes home assistant

swifty99, 2020-05-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io 109.1
esphome 1.14.3

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32, ESP32 Camera

**Affected component:**

probably not related

**Description of problem:**
An unchanged yaml can not be compiled anymore. Maybe the issue is on home assistant side.
Anyway, if I try to compile this node, 
this is the last thing I see from home assistant:
`|-- <FS> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.6
|   |-- <AsyncTCP-esphome> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
Compiling /data/3d_supervisor/.pioenvs/3d_supervisor/src/esphome/components/api/api_connection.cpp.o
Compiling /data/3d_supervisor/.pioenvs/3d_supervisor/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/3d_supervisor/.pioenvs/3d_supervisor/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/3d_supervisor/.pioenvs/3d_supervisor/src/esphome/components/api/api_server.cpp.o`

After that I need to reboot Hassio, everything is dead

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: flur_clock
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "Bit_Pumpe"
  password: "123"
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.178.108
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.178.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0  

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Flur Clock Fallback Hotspot"
    password: "123"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

#esp32_ble_beacon:
#  type: iBeacon
#  uuid: '0af406e0-eeee-4685-bbed-e1243c9dd8a5'

esp32_ble_tracker:

binary_sensor:
  - platform: ble_presence
    mac_address: AC:37:43:77:5F:4C
    name: "ESP32 BLE Tracker Google Home Mini"

output:
  - platform: ledc
    pin: GPIO19
    id: my_lever_1

uart:
  tx_pin: 22
  rx_pin: 23
  baud_rate: 115200
  
  
mqtt:
  broker: 192.168.178.52
  username: device
  password: 123
  

          
text_sensor:
  - platform: mqtt_subscribe
    name: "Location position"
    id: clockpos
    topic: flur_clock/postion/set
    on_value:
      then:
        - uart.write: !lambda |-   
            ESP_LOGD("main", "Command %s", x.c_str());
            const char* cStyle = x.c_str();
            
            std::vector<uint8_t> vec(x.begin(), x.end());
            return vec;        

```

**Logs (if applicable):**

```
clean dev compile (same issue)
INFO Reading configuration /config/esphome/3d_supervisor.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/3d_supervisor
Please wait while upgrading PlatformIO...
PlatformIO has been successfully upgraded to 4.1.0!

********************************************************************************
If you like PlatformIO, please:
- follow us on Twitter to stay up-to-date on the latest project news > https://twitter.com/PlatformIO_Org
- star it on GitHub > https://github.com/platformio/platformio
- try PlatformIO IDE for embedded development > https://platformio.org/platformio-ide
- support us with PlatformIO Plus > https://pioplus.com
********************************************************************************

Processing 3d_supervisor (board: esp-wrover-kit; framework: arduino; platform: espressif32@1.11.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Looking for AsyncTCP-esphome library in registry
LibraryManager: Installing id=6798 @ 1.1.1
AsyncTCP-esphome @ 1.1.1 has been successfully installed!
Looking for ESPAsyncWebServer-esphome library in registry
LibraryManager: Installing id=6758 @ 1.2.6
ESPAsyncWebServer-esphome @ 1.2.6 has been successfully installed!
Looking for ESPAsyncTCP-esphome library in registry
LibraryManager: Installing id=6757
ESPAsyncTCP-esphome @ 1.2.2 has been successfully installed!
LibraryManager: Installing id=6798
Dependency Graph
|-- <AsyncTCP-esphome> 1.1.1
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <FS> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.6
|   |-- <AsyncTCP-esphome> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
Compiling /data/3d_supervisor/.pioenvs/3d_supervisor/src/esphome/components/api/api_connection.cpp.o
Compiling /data/3d_supervisor/.pioenvs/3d_supervisor/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/3d_supervisor/.pioenvs/3d_supervisor/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/3d_supervisor/.pioenvs/3d_supervisor/src/esphome/components/api/api_server.cpp.o

```

**Additional information and things you've tried:**

Clean build and compile.
compile with dev Version
Different ESP32 boards and configs

ESP8266 habe no issues!

brandond, 2020-05-02:
This usually means that the build is consuming too much memory and the host becomes unresponsive as things go into swap. How much memory do you have on the host? What other things are you running on it?

swifty99, 2020-05-03:
Good point,

I am running on a Pi3,however the Pi4 with 2GB is ready for deployment. Will check it.
There are a few extensions which might eat up some ressources:
Unifi
Adguard
Samba
Aircast
NodeRed (almost idling)

I temporarily uninstalled Adguard and unifi. Unifi seems to use a LOT of RAM (running on Java). It freed 200MB uninstalling it. 
Compiling ESP32 nodes will need another ~ 200MB. So compiling with less thenn 300MB free RAM maybe is not stable, I had only around 250MB. ESP32 compiling has no issues now.

you can cheack home assistant RAM like this:
'''yaml
sensor:            
  - platform: systemmonitor
    resources:
      - type: memory_free
'''          

Thanks for the help

closed 2020-05-03 by swifty99