esphome/issues #1194: WARNING Disconnected from API: Timeout while waiting for message response!

a74451, 2020-05-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos D1 Mini Pro Wifi ESP8266

**Affected component:**

**Description of problem:**
Good morning, I have this problem (WARNING Disconnected from API), I've read the forum every forum, but I couldn't find anything that could help me.

Despite getting communication with the device, through the WEB and having other indications that I have a connection with it, the API is giving timeout. I'm not using an MQTT configuration.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
substitutions:
  # Modify variables based on your settings
  hostname: 'water_pump'
  ssid: 'MYSSID'
  password: 'MYPASSWORD'

esphome:
  name: $hostname
  platform: ESP8266
  board: d1_mini_pro

wifi:
  ssid: $ssid
  password: $password
  fast_connect: False
  ap:
    ssid: AP ${hostname}

# Enable logging
logger:

# Enable Home Assistant API
api:
  reboot_timeout: 5min 

ota:

web_server:
 port: 80

globals:
   - id: my_global_int
     type: int
     restore_value: no
     initial_value: '0'
     
sensor:
  - platform: wifi_signal
    id: wifi_db
    update_interval: 60s
    
  - platform: template
    name: "WiFi Signal ${hostname}"
    unit_of_measurement: '%'
    accuracy_decimals: 0
    icon: "mdi:wifi"
    lambda: |-
        if (id(wifi_db).state < -92.0)
          return 100.0;
        if (id(wifi_db).state > -21.0)
          return 1.0;
        else
          return round(( -0.0154 * id(wifi_db).state * id(wifi_db).state ) - ( 0.3794 * id(wifi_db).state ) + 98.182 );
    
[... 223 lines cut ...]
INFO Address in use when adding 169.254.147.225 to multicast group, it is expected to happen on some systems
INFO Connecting to water_pump.local:6053 (192.168.0.136)
ERROR Error while reading incoming messages: Error while receiving data: [Errno 9] Bad file descriptor
**WARNING Disconnected from API: Error while receiving data: [Errno 9] Bad file descriptor
WARNING Disconnected from API: Error while writing data: [Errno 32] Broken pipe
WARNING Couldn't connect to API (Timeout while waiting for message response!). Trying to reconnect in 1 seconds**
INFO Address in use when adding 169.254.127.251 to multicast group, it is expected to happen on some systems
INFO Address in use when adding 169.254.147.225 to multicast group, it is expected to happen on some systems
INFO Address in use when adding 169.254.127.251 to multicast group, it is expected to happen on some systems
INFO Address in use when adding 169.254.147.225 to multicast group, it is expected to happen on some systems
INFO Address in use when adding 169.254.127.251 to multicast group, it is expected to happen on some systems
INFO Address in use when adding 169.254.147.225 to multicast group, it is expected to happen on some systems
INFO Connecting to water_pump.local:6053 (192.168.0.136)
INFO Connecting to water_pump.local:6053 (192.168.0.136)
INFO Connecting to water_pump.local:6053 (192.168.0.136)
```

**Additional information and things you've tried:**

Here we can see that we have the indication to be online
[image]
[image]

But we can't see it here
[image]

randybb, 2020-05-04:
Your have pretty bad signal on that location. -72 dBm is really bad for the small pcb antenna... - Try to move it closer or at least position it to get a bit better reception.

a74451, 2020-05-04:
Obrigado pela ajuda, o melhor que consegui foi -63db.

```
[09:52:28][I][app:100]: ESPHome version 1.14.3 compiled on May  3 2020, 19:45:24
[09:52:28][C][status_led:019]: Status LED:
[09:52:28][C][status_led:020]:   Pin: GPIO13 (Mode: OUTPUT, INVERTED)
[09:52:28][C][wifi:415]: WiFi:
[09:52:28][C][wifi:283]:   SSID: [redacted]
[09:52:28][C][wifi:284]:   IP Address: 192.168.0.136
[09:52:28][C][wifi:286]:   BSSID: [redacted]
[09:52:28][C][wifi:287]:   Hostname: 'water_pump'
[09:52:28][C][wifi:291]:   Signal strength: -63 dB ▂▄▆█
[09:52:28][C][wifi:295]:   Channel: 4
[09:52:28][C][wifi:296]:   Subnet: 255.255.255.0
[09:52:28][C][wifi:297]:   Gateway: 192.168.0.1
[09:52:28][C][wifi:298]:   DNS1: 1.0.0.1
[09:52:28][C][wifi:299]:   DNS2: 1.1.1.1
[09:52:28][C][template.sensor:021]: Template Sensor 'WiFi Signal water_pump'
[09:52:28][C][template.sensor:021]:   Unit of Measurement: '%'
[09:52:28][C][template.sensor:021]:   Accuracy Decimals: 0
[09:52:28][C][template.sensor:021]:   Icon: 'mdi:wifi'
[09:52:28][C][template.sensor:022]:   Update Interval: 60.0s
[09:52:28][C][template.text_sensor:020]: Template Sensor 'Pump State'
[09:52:28][C][switch.gpio:042]: GPIO Switch 'Pump Relay'
[09:52:28][C][switch.gpio:043]:   Pin: GPIO5 (Mode: OUTPUT)
[09:52:28][C][switch.gpio:059]:   Restore Mode: Restore (Defaults to OFF)
[09:52:28][C][template.switch:058]: Template Switch 'Pump Remote'
[09:52:28][C][template.switch:059]:   Restore State: NO
[09:52:28][C][template.switch:060]:   Optimistic: NO
[09:52:28][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'S1'
[09:52:28][C][gpio.binary_sensor:016]:   Pin: GPIO0 (Mode: INPUT_PULLUP, INVERTED)
[09:52:28][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'S2'
[09:52:28][C][gpio.binary_sensor:016]:   Pin: GPIO14 (Mode: INPUT_PULLUP, INVERTED)
[09:52:28][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Sw Manual'
[09:52:28][C][gpio.binary_sensor:016]:   Pin: GPIO3 (Mode: INPUT_PULLUP, INVERTED)
[09:52:28][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Sw Timed On - Add Time'
[09:52:28][C][gpio.binary_sensor:016]:   Pin: GPIO2 (Mode: INPUT_PULLUP, INVERTED)
[09:52:28][C][logger:175]: Logger:
[09:52:28][C][logger:176]:   Level: DEBUG
[09:52:28][C][logger:177]:   Log Baud Rate: 115200
[09:52:28][C][logger:178]:   Hardware UART: UART0
[09:52:28][C][web_server:122]: Web Server:
[09:52:28][C][web_server:123]:   Address: water_pump.local:80
[09:52:28][C][ota:029]: Over-The-Air Updates:
[09:52:28][C][ota:030]:   Address: water_pump.local:8266
[09:52:28][C][api:095]: API Server:
[09:52:28][C][api:096]:   Address: water_pump.local:6053
[09:52:28][C][wifi_signal.sensor:009]: WiFi Signal 'wifi_db'
[09:52:28][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dB'
[09:52:28][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[09:52:28][C][wifi_signal.sensor:009]:   Icon: 'mdi:wifi'
[09:52:31][D][text_sensor:015]: 'Pump State': Sending state 'remote'
[09:52:31][D][sensor:092]: 'WiFi Signal water_pump': Sending state 61.00000 % with 0 decimals of accuracy
[09:52:36][D][text_sensor:015]: 'Pump State': Sending state 'remote'
[09:52:36][D][sensor:092]: 'wifi_db': Sending state -63.00000 dB with 0 decimals of accuracy
```

nebula-it, 2020-05-05:
Just to add to this, same thing started happening with all (~10) of my ESPHome devices  recently about 3-4 days ago. I was making sure it was not a networking issues since it started randomly, no ESPHome/HomeAssistant updates preceded this. I've tried recompiling firmware. Upgrading HA to latest version but no dice.
Would be happy to provide any additional info if needed.

a74451, 2020-05-26:
This started to happen very often, after I updated the version of ESPHome (Update Available! 1.13.6 ➡️1.14.3), in the old version it happened eventually, like once a month, and now two to three times a day. The same occurs in other equipment, where the wifi signal is better (-44 dB). What I find strange is that I have connection with the equipment, through the log and the web.

DrToxic, 2020-06-13:
I'm getting similar random disconnects fairly frequently. What stumps me is, one entity of a device will go completely missing! I have a 4-bank smart extension lead, sockets 1-3 go unavailable, socket 4 goes bye-byes! As far as I can see, these corrolate to the log entry:
WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data: Timeout while waiting for API response!
[image]

DrToxic, 2020-06-16:
I've fixed my own issue. Whether this is common to everyone or not, I'm not sure.
My network was unhappy with the way DNS is being handled. Each device now has the following example config under WiFi:

  fast_connect: on
  manual_ip:
    static_ip: 192.168.69.240
    gateway: 192.168.69.1
    subnet: 255.255.255.0
    dns1: 192.168.69.11

h0jeZvgoxFepBQ2C, 2020-07-25:
Looks like the same is happening to me.. No fast_connect used, i really think its somehow related to the 1.14 version of HA.
See also my forum entry: https://community.home-assistant.io/t/esphome-device-says-connected-successfully-is-sending-sensor-data-but-appears-completely-offline-in-home-assistang-no-mqtt/214025

Had the Disconnected error before I set the very_verbose log level, but now its not happening anymore 🙄  But device is still offline.

h0jeZvgoxFepBQ2C, 2020-07-25:
Maybe also related: https://github.com/esphome/issues/issues/1237

ceesvanegmond, 2020-08-17:
Same issue here... Just the message: WARNING Disconnected from API: Timeout while waiting for message response!.

mDNS is working correctly.

kadelidden, 2020-09-09:
Not gunna lie, I've been suffering too. However I have stumbled upon some very strange behaviour that has solved some of my problems. I believe that the Wireless Access Point Standard has something to play in my situation.

**My setup:**
-Ubiquiti Edgerouter X (main router handling dhcp and static)
-Ubiquiti Aircube AC (running dual band)
-Ubiquiti Aircube AC (Running 2.4G band only)
-Unifi UAP-AC-M Outdoor point (Have had both 2.4G band only and Both 2.4G and 5G bands on for testing)
-HassOS 4.12
-ESPHome Dev (have also tested with ESPHome 1.14.5)
-Rpi4
-All devices using API, not MQTT

**ESPHome Devices:**
-10x LinkNode R4 4 channel relays (all switchboard mounted)
-5x AliExpress Cheapie 4channel Relays that use an ESP-01 (as seen here: https://community.home-assistant.io/t/4-channel-relay-esp8266-wifi-board-working-with-esphome/120594)
-2x Sonoff S26 plug in sockets
-4x Sonoff Basic R2's
-1x NodeMCU V2 running only a DHT11 Temp Sensor

**Observations:**
- When any ESPHome device is connected to any of the Access Points running the 802.11ac Standard, it repeatedly disconnects and reconnects from the API. Adding the usual additions to my configs for the devices has not worked for me at all as discussed in the many previous threads.
-As soon as the device is connected to my Aircube ISP which runs the 802.11n standard, they do not disconnect or reconnect continuously. I have scaled back the configurations of the devices connected and no matter what I take away or add, it will remain connected.
-Throughout the whole time (around 1.5 years) the NodeMCU has never had any of the disconnect/reconnect behaviour. Ths device has stock standard config with the only addition being a static IP.
-The aircube ISP has 20Mhz channel width and frequency of 2412 MHz manually selected.
-This behaviour was replicated in both ESPHome Dev and ESPHome 1.14.5

Would be willing to adjust settings for testing if devs want.

I will say this though, I'm not saying this is THE fix for the espressif 8266 API issue, I am saying this worked for me and would encourage anyone who is having this issue to start reading up and trying the various fixes to see what works for you.

If this works for everyone, fantastic!

Cheers from Aus,
Kade

ceesvanegmond, 2020-09-19:
> Not gunna lie, I've been suffering too. However I have stumbled upon some very strange behaviour that has solved some of my problems. I believe that the Wireless Access Point Standard has something to play in my situation.
> 
> **My setup:**
> -Ubiquiti Edgerouter X (main router handling dhcp and static)
> -Ubiquiti Aircube AC (running dual band)
> -Ubiquiti Aircube AC (Running 2.4G band only)
> -Ubiquiti Aircube ISP (2.4G only)
> -Unifi UAP-AC-M Outdoor point (Have had both 2.4G band only and Both 2.4G and 5G bands on for testing)
> -HassOS 4.12
> -ESPHome Dev (have also tested with ESPHome 1.14.5)
> -Rpi4
> -All devices using API, not MQTT
> 
> **ESPHome Devices:**
> -10x LinkNode R4 4 channel relays (all switchboard mounted)
> -5x AliExpress Cheapie 4channel Relays that use an ESP-01 (as seen here: https://community.home-assistant.io/t/4-channel-relay-esp8266-wifi-board-working-with-esphome/120594)
> -2x Sonoff S26 plug in sockets
> -4x Sonoff Basic R2's
> -1x NodeMCU V2 running only a DHT11 Temp Sensor
> 
> **Observations:**
> 
>     * When any ESPHome device is connected to any of the Access Points running the 802.11ac Standard, it repeatedly disconnects and reconnects from the API. Adding the usual additions to my configs for the devices has not worked for me at all as discussed in the many previous threads.
> 
>     * As soon as the device is connected to my Aircube ISP which runs the 802.11n standard, they do not disconnect or reconnect continuously. I have scaled back the configurations of the devices connected and no matter what I take away or add, it will remain connected.
> 
>     * Throughout the whole time (around 1.5 years) the NodeMCU has never had any of the disconnect/reconnect behaviour. Ths device has stock standard config with the only addition being a static IP.
> 
>     * The aircube ISP has 20Mhz channel width and frequency of 2412 MHz manually selected.
> 
>     * This behaviour was replicated in both ESPHome Dev and ESPHome 1.14.5
> 
> 
> Would be willing to adjust settings for testing if devs want.
> 
> I will say this though, I'm not saying this is THE fix for the espressif 8266 API issue, I am saying this worked for me and would encourage anyone who is having this issue to start reading up and trying the various fixes to see what works for you.
> 
> If this works for everyone, fantastic!
> 
> Cheers from Aus,
> Kade
> 
> UPDATE 18/9/2020: tonight I disconnected the dual band WAP that had devices dropping in and out. I also pulled power from HA Pi4 by accident at the same time. All of a sudden, the stable WAP that had no drop outs started dropping devices. Back to the testing 😤

Any more news yet? Back to stable?

kadelidden, 2020-09-20:
> Any more news yet? Back to stable?

None so far. My logbook just keeps on getting longer. Tried power cycling the devices to get them back to the same WAPS as before, no luck with that fix.

Thinking I'll try some individual config changes. Just really weird that a baseline config renders a device communicating perfectly for one person, and another heavily modified device config works for another person. I'm leaning towards it being an access point issue. I just don't see it being an esphome issue, more an API/wifi issue 

Very open to suggestions to fix it though!

user514, 2020-10-09:
Let me add here my experience as I have solved this same error. With an esp32 module, and esphome 1.15.2.

Many disconnects, about one each 20 seconds...
"Error getting initial data for 192.168.1.140: Timeout while waiting for API response!"

  Noticed, it could be related with the access point. One many disconnects, the other pretty solid:

Access point wih 802.ac wifi capability: my esp32 doesn't like that AP and got many disconnects.
Access pont with only 802.bgn: my esp32 likes that and ketp connected all the time.

  Not saying that was the reason, but just to note that the access point have something to do with my problem.

  Ok. And now what really SOLVED MY PROBLEM:

`power_save_mode: none`

As soon as I added that to my wifi component. Uploaded firm. (no fast_connect needed anymore with both access points)

MariusLaurinaitis, 2020-10-14:
> Let me add here my experience as I have solved this same error. With an esp32 module, and esphome 1.15.2.
> 
> Many disconnects, about one each 20 seconds...
> "Error getting initial data for 192.168.1.140: Timeout while waiting for API response!"
> 
> Noticed, it could be related with the access point. One many disconnects, the other pretty solid:
> 
> Access point wih 802.ac wifi capability: my esp32 doesn't like that AP and got many disconnects.
> Access pont with only 802.bgn: my esp32 likes that and ketp connected all the time.
> 
> Not saying that was the reason, but just to note that the access point have something to do with my problem.
> 
> Ok. And now what really SOLVED MY PROBLEM:
> 
> `power_save_mode: none`
> 
> As soon as I added that to my wifi component. Uploaded firm. (no fast_connect needed anymore with both access points)
> 
> Edit: Let me add that the problem with the AC access point was not being far away from it for sure. It is two steps away from it with -48dB wifi strenght, so the signal is was not the problem. The one really far away was the one that was pretty solid from the begining. Now with power_save: none both works great.

To me it's weird that I only have this problem with the Sonoff T1 3gang (UK version). I have 22 Sonoff switches, and only 5 3Gang and they keep disconnecting, as I can't even upgrade them over OTA because ..... Broken pipe.
but why then written ESPhome for Power Save Mode - NONE (least power saving, Default for ESP8266), its not default?

Now I will need to update all my switches and 5 will need to be removed and updated manually.

jul1u5, 2020-10-25:
I was also having problems with "Broken pipe" when updating via OTA. I fixed this by changing the WiFi standard from `802.11b/g/n` to `802.11b/g` in my router.

krambriw, 2021-02-16:
I have this issue as well. Initially everything works just fine but after a while the error appears in the logs. Interesting is the error message that appears just before, "Field number 0 is illegal" in the pyhon code it seems, maybe a clue why it goes wrong. See below. At first the blue LED is just toggling fine and then the traceback happens. As mentioned by others, things seems to continue working fine

[07:32:38][D][switch:045]: 'blue': Sending state ON
[07:32:43][D][switch:025]: 'blue' Turning OFF.
[07:32:43][D][switch:045]: 'blue': Sending state OFF
[07:32:43][D][switch:021]: 'blue' Turning ON.
[07:32:43][D][switch:045]: 'blue': Sending state ON
Exception in thread Thread-4:
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.6/site-packages/esphome/api/client.py", line 391, in run
    self._run_once()
  File "/usr/local/lib/python3.6/site-packages/esphome/api/client.py", line 381, in _run_once
    msg.ParseFromString(raw_msg)
  File "/usr/local/lib/python3.6/site-packages/google/protobuf/message.py", line 199, in ParseFromString
    return self.MergeFromString(serialized)
  File "/usr/local/lib/python3.6/site-packages/google/protobuf/internal/python_message.py", line 1134, in MergeFromString
    if self._InternalParse(serialized, 0, length) != length:
  File "/usr/local/lib/python3.6/site-packages/google/protobuf/internal/python_message.py", line 1184, in InternalParse
    raise message_mod.DecodeError('Field number 0 is illegal.')
google.protobuf.message.DecodeError: Field number 0 is illegal.

WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to 192.168.0.254:6053 (192.168.0.254)
WARNING Couldn't connect to API (Timeout while waiting for message response!). Trying to reconnect in 1 seconds
INFO Connecting to 192.168.0.254:6053 (192.168.0.254)
WARNING Couldn't connect to API (Timeout while waiting for message response!). Trying to reconnect in 1 seconds

krambriw, 2021-02-16:
And then this happened s well. Another reason? "Wrong wire type in tag"
???

[14:23:46][D][switch:025]: 'blue' Turning OFF.
[14:23:46][D][switch:045]: 'blue': Sending state OFF
[14:23:46][D][switch:021]: 'blue' Turning ON.
Exception in thread Thread-4:
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/local/lib/python3.6/site-packages/esphome/api/client.py", line 391, in run
    self._run_once()
  File "/usr/local/lib/python3.6/site-packages/esphome/api/client.py", line 381, in _run_once
    msg.ParseFromString(raw_msg)
  File "/usr/local/lib/python3.6/site-packages/google/protobuf/message.py", line 199, in ParseFromString
    return self.MergeFromString(serialized)
  File "/usr/local/lib/python3.6/site-packages/google/protobuf/internal/python_message.py", line 1134, in MergeFromString
    if self._InternalParse(serialized, 0, length) != length:
  File "/usr/local/lib/python3.6/site-packages/google/protobuf/internal/python_message.py", line 1188, in InternalParse
    buffer, new_pos, wire_type)  # pylint: disable=protected-access
  File "/usr/local/lib/python3.6/site-packages/google/protobuf/internal/decoder.py", line 977, in _DecodeUnknownField
    raise _DecodeError('Wrong wire type in tag.')
google.protobuf.message.DecodeError: Wrong wire type in tag.

WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to 192.168.0.254:6053 (192.168.0.254)
WARNING Couldn't connect to API (Timeout while waiting for message response!). Trying to reconnect in 1 seconds
INFO Connecting to 192.168.0.254:6053 (192.168.0.254)
WARNING Couldn't connect to API (Timeout while waiting for message response!). Trying to reconnect in 1 seconds

silviudc, 2021-02-20:
I have also started to get this all of a sudden after being rock solid for months.
Using an Asus RT-AX88U in access point mode and signal of both devices is around -25db as they are very close.
My set up was just two esp8266 12-Fs each with lux/temp&hum/motion sensors and never disconnected.
Then the other day I have added a third, with lux/temp&hum/motion sensors/co2 sensors and started to get that dreaded warning disconnected from api etc. 
Now I have two sensors getting this warning consistently but the first one is rock solid and hasn't dropped once so it sort of rules out the AP as being the issue.
Updated esphome 1.16.2 today and no change.
Tried fast_connect: true,  power_save_mode: none, reboot_timeout: 0s, static ips, all without success. 
Has anyone tried moving away from api to mqtt to see if it makes any difference?

silviudc, 2021-02-21:
I did some more testing and I think this error might be related due to AP config after all, here is what I have done which has made the connection stable for about 1 day (so far) so maybe it works for others:

- Updated router FW to latest (bought this recently and still had stock FW) 
- Change 20/40 mhz mode to 20 mhz only
- Disabled Wifi 6 mode on 2.4G band
- Manually set the 2.4G channel to 1

prchal, 2021-04-27:
Hi, same issue. All ESP8266s have max uptime around 20000s. So, periodically reset. Not all at once, every in its time. Seems to me something overflow or memory exhausted.
ESP32s are OK.
Would insert image from HomeAssistant. But how?

ilium007, 2021-06-08:
I am seeing this exact behaviour, ESPHone nice running ESP8266 boards reboot every 25mins. 

Mikrotik AP 2.4ghz at 20MHz. 

[image]

ilium007, 2021-06-09:
This is what it looks like with the following set:

```
wifi:
  fast_connect: on
  manual_ip:
    static_ip: 192.168.10.201
    gateway: 192.168.10.1
    subnet: 255.255.255.0
    dns1: 192.168.10.1
  power_save_mode: none
```

ilium007, 2021-06-09:
[image]

ilium007, 2021-06-09:
Hours later, still stable

[image]

ilium007, 2021-06-13:
Is the answer to migrate everything to MQTT? After I changed to status IP's everything has been solid for days.

chemelli74, 2021-06-25:
Seems upgrading to 1.19.2 solved the issue: more than 24h and not a single disconnect event.

Simone

ilium007, 2021-06-25:
Testing now but it hasn't fixed the DHCP client hostname issue :(

ilium007, 2021-06-25:
Broken still in 1.20.0-dev, will try 1.19.2 now.

[image]

chemelli74, 2021-06-25:
> Testing now but it hasn't fixed the DHCP client hostname issue :(

I'm on DHCP and don't experience any issue with DNS resolution.

Simone

ilium007, 2021-06-25:
> > Testing now but it hasn't fixed the DHCP client hostname issue :(
> 
> 
> 
> I'm on DHCP and don't experience any issue with DNS resolution.
> 
> 
> 
> Simone

Hostname isn't passed with initial DHCP request. Known issue.

ilium007, 2021-06-25:
1.19.2 with DHCP still shows constant dropouts 

[image]

ilium007, 2021-06-26:
Back to static IP on 1.19.2 and it's stable again.

[image]

chemelli74, 2021-06-27:
> 1.19.2 with DHCP still shows constant dropouts

My issue was never related to the uptime of the ESP32 module itself, but more about the reported data from BLE sensors.
With 1.19.2 I have zero issues.

I think you should research in your WI-FI network infrastructure some hints.

Simone

tomlut, 2022-02-28:
I have 41 ESPHome modules sprinkled around my house only one has this problem. It has excellent signal strength and other ESPs on the same AP do not have this issue,

I thought it might be the ESP board itself so I replaced it with a known good board. No change, still getting regular drop-outs.

The only thing left to try is to change the power supply. Though I doubt this is the issue as it is a distributed 5V system that other ESP boards run quite happily off of. 

```
esphome:
  name: comms-rack-dht
  platform: ESP32
  board: mhetesp32minikit

wifi:
  ssid: 'WAPRU'
  password: !secret wifi_pwd
  manual_ip:
    static_ip: 10.1.1.72
    gateway: 10.1.1.1
    subnet: 255.255.255.0
  power_save_mode: none
```

```
INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 10.1.1.72 using esphome API
INFO Successfully connected to 10.1.1.72
[12:06:29][I][app:102]: ESPHome version 2022.2.5 compiled on Feb 28 2022, 12:05:36
[12:06:29][C][wifi:491]: WiFi:
[12:06:29][C][wifi:353]:   Local MAC: 34:AB:95:80:AA:D8
[12:06:29][C][wifi:354]:   SSID: [redacted]
[12:06:29][C][wifi:355]:   IP Address: 10.1.1.72
[12:06:29][C][wifi:357]:   BSSID: [redacted]
[12:06:29][C][wifi:358]:   Hostname: 'comms-rack-dht'
[12:06:29][C][wifi:360]:   Signal strength: -51 dB ▂▄▆█
[12:06:29][C][wifi:364]:   Channel: 6
[12:06:29][C][wifi:365]:   Subnet: 255.255.255.0
[12:06:29][C][wifi:366]:   Gateway: 10.1.1.1
[12:06:29][C][wifi:367]:   DNS1: 0.0.0.0
[12:06:29][C][wifi:368]:   DNS2: 0.0.0.0
[12:06:29][C][logger:233]: Logger:
[12:06:29][C][logger:234]:   Level: DEBUG
[12:06:29][C][logger:235]:   Log Baud Rate: 115200
[12:06:29][C][logger:236]:   Hardware UART: UART0
[12:06:29][C][dht:029]:   Temperature 'Comms Rack Temperature'
[12:06:29][C][dht:029]:     Device Class: 'temperature'
[12:06:29][C][dht:029]:     State Class: 'measurement'
[12:06:29][C][dht:029]:     Unit of Measurement: '°C'
[12:06:29][C][dht:029]:     Accuracy Decimals: 1
[12:06:29][C][dht:030]:   Humidity 'Comms Rack Humidity'
[12:06:29][C][dht:030]:     Device Class: 'humidity'
[12:06:29][C][dht:030]:     State Class: 'measurement'
[12:06:29][C][dht:030]:     Unit of Measurement: '%'
[12:06:29][C][dht:030]:     Accuracy Decimals: 0
[12:06:29][C][restart:022]: Restart Switch 'Comms Rack DHT Restart'
[12:06:29][C][restart:022]:   Icon: 'mdi:restart'
[12:06:29][C][mdns:084]: mDNS:
[12:06:29][C][mdns:085]:   Hostname: comms-rack-dht
[12:06:29][C][ota:085]: Over-The-Air Updates:
[12:06:29][C][ota:086]:   Address: 10.1.1.72:3232
[12:06:29][C][ota:089]:   Using Password.
[12:06:29][C][api:138]: API Server:
[12:06:29][C][api:139]:   Address: 10.1.1.72:6053
[12:06:29][C][api:141]:   Using noise encryption: YES
[12:06:29][C][wifi_signal.sensor:009]: WiFi Signal 'Comms Rack DHT WiFi Signal'
[12:06:29][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[12:06:29][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[12:06:29][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[12:06:29][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[... 23 lines cut ...]
[12:11:05][D][dht:048]: Got Temperature=21.0°C Humidity=62.4%
[12:11:20][D][dht:048]: Got Temperature=21.0°C Humidity=62.3%
[12:11:21][I][ota:105]: Boot seems successful, resetting boot loop counter.
[12:11:22][D][esp32.preferences:114]: Saving preferences to flash...
[12:11:36][D][dht:048]: Got Temperature=21.0°C Humidity=62.2%
[12:11:50][D][dht:048]: Got Temperature=21.0°C Humidity=62.2%
[12:12:05][D][dht:048]: Got Temperature=21.0°C Humidity=62.2%
[12:12:21][D][dht:048]: Got Temperature=21.0°C Humidity=62.1%
[12:12:35][D][dht:048]: Got Temperature=21.0°C Humidity=62.1%
[12:12:50][D][dht:048]: Got Temperature=21.0°C Humidity=62.1%
[12:13:05][D][dht:048]: Got Temperature=21.0°C Humidity=62.1%
[12:13:21][D][dht:048]: Got Temperature=21.0°C Humidity=62.1%
INFO 10.1.1.72: Ping timed out!
INFO Disconnected from ESPHome API for 10.1.1.72
WARNING Disconnected from API
INFO Successfully connected to 10.1.1.72
[12:13:35][D][dht:048]: Got Temperature=21.0°C Humidity=62.0%
[12:13:35][D][sensor:125]: 'Comms Rack Temperature': Sending state 20.97334 °C with 1 decimals of accuracy
[12:13:35][D][sensor:125]: 'Comms Rack Humidity': Sending state 62.22667 % with 0 decimals of accuracy
[12:13:50][D][dht:048]: Got Temperature=21.0°C Humidity=62.0%
[12:13:53][D][sensor:125]: 'Comms Rack DHT WiFi Signal': Sending state -51.33333 dBm with 0 decimals of accuracy
[12:14:05][D][dht:048]: Got Temperature=21.1°C Humidity=62.0%
```

```

sheminasalam, 2022-03-28:
any solutions?

tomlut, 2022-03-28:
In my case it may have been interference. I moved the board location (to a location with about the same signal strength of -50dB) and it has been rock solid for weeks.

lorenzo-deluca, 2022-07-04:
I also have this problem.
In my case i know I have a ping loss problem, unfortunately the device is in the garage connected via powerline and the line quality is porr.
It loses a couple of pings every 10-20 minutes but it is the best I can do since from the house to the garage pass at least 4 floors of building.

Is it not possible to increase the time before the timeout is triggered?
It actually works perfectly, I don't lose any state changes and the line is the best I can do.

Thank you!

ardizzonig, 2022-12-07:
I also have randomly ping loss on a Shelly1 flashed with ESP home in a garage connected with powerline. I added a camera connected to the same powerline wifi and I noticed that it was always online with no connection problems.
So with a pinglogger I start to ping the Shelly every second and the connection magically remain stable! I don't know what happen with the shelly and the powerline but I solve pingig it from home assistant every 10 seconds.
Now is rock solid.
I hope this could help!

b3bs, 2022-12-24:
I had an similar behavior. In my case powering the ESP over the 5V Pin instead of the micro USB solved the problem.