esphome/issues #1195: Compiling yaml is causing RPi to freeze

sermayoral, 2020-05-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Raspberry Pi 3 B+ ethernet mode.

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

ESP32, but I think the problem is for every ESP.

**Description of problem:**
Been using ESPhome for a while now and all seems ok until I tried to compile a new .bin file and it just freezes, the RPi attempts to restart but then freezes, I loose access to SSH and access to the unifi add-on so I know its failed, the only way now is to recycle the power and delete the home-assistant_v2.db file as its now corrupt.

**Problem-relevant YAML-configuration entries:**
```yaml
Any yaml

```

**Logs (if applicable):**

```
No log, RPI just freeze

```

**Additional information and things you've tried:**

This issue was open here: #191 

I have reopened the issue because I have a clue about it.

In ESPHome you can change platformio options: https://esphome.io/components/esphome.html#platformio-options

Maybe, if you tell platformio that you are using a raspberry pi 3b, compiler adjust to rpi’s hardware not to causing freeze. Parameters we have to put are described here: https://docs.platformio.org/en/latest/boards/linux_arm/raspberrypi_3b.html

It's possible? Thanks!!

strumf666, 2020-05-08:
I have almost the same problem.
[https://github.com/esphome/issues/issues/1092](url)

abelmatser, 2020-05-22:
See the following [issue ](https://github.com/esphome/issues/issues/1193). It is about not having enough memory free on your system. After turning off some add-ons on my RPi3B it would compile.

strumf666, 2020-05-22:
You are correct, after freeing some ram (to ~450mb) compiling and flashing works.

sermayoral, 2020-05-22:
Thanks @abelmatser. Stop addons before compiling does the trick :-)

closed 2020-10-03 by stale[bot]