esphome/issues #1198: How to filter out "unavailable" from binary sensor?

roc634, 2020-05-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant 0.109.2
ESPHome 1.14.3
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32-nodemcu-32s
**Affected component:**

Binary sensor
https://esphome.io/components/binary_sensor/index.html

**Description of problem:**
I connect a rain sensor to an ESP32 board. I expect ONs and OFFs but I also receive a lot of "unavailable" randomly, which is annoying.  I tried to use lambda,but I don't know how to make it ignore "unavailable".

**Problem-relevant YAML-configuration entries:**
```
binary_sensor:
  - platform: gpio
    pin:
      number: 12
      mode: INPUT_PULLUP
      inverted: True
    name: "rain"
    id: rain_roof
    device_class: moisture
    filters:
      - delayed_on: 1000ms
     - lambda: |-
         if (id(rain_roof).state != 'unavailable') {
           return x;
         } else {
           return {};
         }

```

**Logs (if applicable):**

```
src/main.cpp:281:31: warning: character constant too long for its type
       if (rain_roof->state != 'unavailable') {
                               ^
src/main.cpp: In lambda function:
src/main.cpp:281:28: warning: comparison of constant '1633840229' with boolean expression is always true [-Wbool-compare]
       if (rain_roof->state != 'unavailable') {

```

**Additional information and things you've tried:**
I've tried sensor filter before that, 
```
binary_sensor:
  - platform: gpio
    pin:
      number: 12
      mode: INPUT_PULLUP
      inverted: True
    name: "rain"
    id: rain_roof
    device_class: moisture
    filters:
      - delayed_on: 1000ms
      - filter_out: 'unavailable'
```
but it seems it doesn't apply to binary sensor and I got the log below.
```
      Unable to find filter with the name 'filter_out'.
      filter_out: unavailable [source /root/roof.yaml:89]
```
Could you please help me with that? Thank you so much!!

glmnet, 2020-05-07:
your problem is communication between the ESP and the HA, when they lost communication, HA marks it as unavailable, so it makes no sense to try to see unavailable in ESP side.

You might get quickly answers in HA forums or Discord chat, but this is something you'll have to deal at HA side, or figure out why you are having communication issues.

closed 2020-05-07 by glmnet

roc634, 2020-05-07:
Thanks for your reply, glmnet. I set up an automation in HA to process the ONs and OFFs and ignore the "unavailable". Problem is solved now. 
> your problem is communication between the ESP and the HA, when they lost communication, HA marks it as unavailable, so it makes no sense to try to see unavailable in ESP side.