esphome/issues #1200: Tuya MCU

LukaszP2, 2020-05-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ubuntu server/HA Docker

ESP8266

https://templates.blakadder.com/aofo_4AC+4USB_tuya.html

Strip has 4 outlets, 1 USB group ond power all.

Power All works every time 
First outlet works every time 
2/3/4 after turn on do "MCU restarted" and didnt work 

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: strip
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "SSID"
  password: "PASSWORD"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Strip Fallback Hotspot"
    password: "pkVQ3xINRtIZ"

captive_portal:

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:

# Enable webserver
web_server:
  port: 80

ota:

# My dimmer used the hardware serial port on the alternate pins
uart:
  rx_pin: RX #GPIO03
  tx_pin: TX #GPIO01
  baud_rate: 9600

# Register the Tuya MCU connection
tuya:

light:
  - platform: tuya
    name: Outlet 1
    switch_datapoint: 1
    dimmer_datapoint: 9
  - platform: tuya
    name: Outlet 2
    switch_datapoint: 2
    dimmer_datapoint: 10
  - platform: tuya
    name: Outlet 3
    switch_datapoint: 3
    dimmer_datapoint: 11
  - platform: tuya
    name: Outlet 4
    switch_datapoint: 4
    dimmer_datapoint: 12
  - platform: tuya
    name: USB
    switch_datapoint: 7
    dimmer_datapoint: 15
  - platform: tuya
    name: Strip
    switch_datapoint: 38
    dimmer_datapoint: 99

```

**Logs (if applicable):**

```
[15:16:56][D][light:264]: 'Outlet 2' Setting:
[15:16:56][D][light:273]:   State: ON
[15:16:56][D][light:264]: 'Outlet 2' Setting:
[15:16:56][D][light:277]:   Brightness: 100%
[15:16:57][E][tuya:189]: invalid command (1c) received
[15:16:57][I][tuya:117]: MCU restarted
[15:16:57][D][light:264]: 'Outlet 1' Setting:
[15:16:57][D][light:273]:   State: ON
[15:16:57][D][light:264]: 'Outlet 2' Setting:
[15:16:57][D][light:264]: 'Outlet 3' Setting:
[15:16:57][D][light:264]: 'Outlet 4' Setting:
[15:16:57][D][light:264]: 'USB' Setting:
[15:16:57][D][light:264]: 'Strip' Setting:
[15:16:57][D][light:264]: 'Outlet 1' Setting:
[15:16:57][D][light:277]:   Brightness: 100%
[15:16:57][D][light:264]: 'Outlet 2' Setting:
[15:16:57][D][light:277]:   Brightness: 100%
[15:16:57][D][light:264]: 'Outlet 3' Setting:
[15:16:57][D][light:277]:   Brightness: 100%
[15:16:57][D][light:264]: 'Outlet 4' Setting:
[15:16:57][D][light:277]:   Brightness: 100%
[15:16:57][D][light:264]: 'USB' Setting:
[15:16:57][D][light:277]:   Brightness: 100%
[15:16:57][D][light:264]: 'Strip' Setting:
[15:16:57][D][light:273]:   State: OFF
[15:16:57][D][light:264]: 'Outlet 1' Setting:
[15:16:58][D][light:264]: 'Outlet 1' Setting:
[15:16:58][D][light:277]:   Brightness: 100%
[15:16:58][C][tuya:023]: Tuya:
[15:16:58][C][tuya:032]:   Datapoint 1: switch (value: OFF)
[15:16:58][C][tuya:034]:   Datapoint 9: int value (value: 0)
[15:16:58][C][tuya:032]:   Datapoint 2: switch (value: OFF)
[15:16:58][C][tuya:032]:   Datapoint 3: switch (value: OFF)
[15:16:58][C][tuya:032]:   Datapoint 4: switch (value: OFF)
[15:16:58][C][tuya:032]:   Datapoint 7: switch (value: OFF)
[15:16:58][C][tuya:032]:   Datapoint 38: switch (value: OFF)
[15:16:58][C][tuya:034]:   Datapoint 10: int value (value: 0)
[15:16:58][C][tuya:034]:   Datapoint 11: int value (value: 0)
[15:16:58][C][tuya:034]:   Datapoint 12: int value (value: 0)
[15:16:58][C][tuya:034]:   Datapoint 15: int value (value: 0)
[15:16:58][C][tuya:046]:   Product: '{"p":"gl5fdiv1tc9mkvlp","v":"1.0.0","m":0}'
[15:17:12][E][tuya:189]: invalid command (1c) received

```

**Additional information and things you've tried:**

lpbreault, 2020-05-06:
You are using the light component but your device is switches... The dimmer datapoint you selected are not good as these are the datapoints for delay.

As of today you can't directly use the Tuya integration for your purpose. 

In your case, I would try to issue commands using UART bus...

Maybe something like this (adjust the tx and rx pin!)

```
logger:
  baud_rate: 0

uart:
  baud_rate: 9600
  tx_pin: GPIO1
  rx_pin: GPIO3

switch:
  - platform: template
    name: My plug strip socket 1
    optimistic: true
    turn_on_action:
      - uart.write: [0x55, 0xAA, 0x00, 0x06, 0x00, 0x05, 0x01, 0x01, 0x00, 0x01, 0x01, 0x08]
    turn_off_action:
      - uart.write: [0x55, 0xAA, 0x00, 0x06, 0x00, 0x05, 0x01, 0x01, 0x00, 0x01, 0x00, 0x07]
```

"0x55, 0xAA, 0x00" is the Frame Header Version part
"0x06" is the Command Word to inform MCU we are sending a command
"0x00, 0x05" is the Data Length
"0x01" is the Datapoint Id
"0x01" is the Data Type (in your case a Boolean)
"0x00, 0x01" is the Function Length
"0x01" is the Function Command (01 for ON and 00 for OFF)
"0x08" is the Checksum

I found these infos on [this PDF](https://images.tuyacn.com/smart/aircondition/Guide-to-Interworking-with-the-Tuya-MCU.pdf) (from Tuya CN).

The way to calculate checksum is easy... you sum the bytes after the 0x55 and 0xAA then you remove 1 so if the sum is 0x09 then it's 0x08.

For the other Datapoint Id, you just have to change the Datapoint Id and the checksum ;)

To keep your components updated, you need to create a custom uart component. To do so, read [this](https://esphome.io/custom/uart.html).

You can use the Tuya PDF I linked to decode the strings received from the MCU... normally a string like: 55aa030700050101000101?? should mean the datapoint 1 is ON and 55aa030700050101000100?? should mean the datapoint 1 is OFF

Hope that helps ;)

LukaszP2, 2020-05-06:
big thanks!!  it take time for me ;)

ssieb, 2020-08-06:
There's a Tuya switch component available now, so this can be closed.

closed 2020-08-06 by ssieb

LukaszP2, 2020-08-06:
> There's a Tuya switch component available now, so this can be closed.

where?

ssieb, 2020-08-06:
https://next.esphome.io/components/switch/tuya.html
That's only in dev (and now beta), but there is a release expected soon.