esphome/issues #1202: time.py: server field doesn't allow hostnames

mirko, 2020-05-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

```
time.sntp: [source include/X/common.yaml:16]
  platform: sntp
  id: sntp_time
  servers: 
    
    Invalid domain: ntp.
    - [source include/X/common.yaml:19]
      ntp
```

**Additional information and things you've tried:**
See above.

EDIT: With all due respect, yes, open source developers invest a lot of time (I do myself). But also do those, who narrow and track down bugs, prepare test cases, propose solutions, raise discussions, etc. The enforced template might work for some bugs, but definitely not for all kinds.
Obviously it's your project and decision, but let me just state, that this bug reporting experience definitely increases the threshold for me of reporting furthers.

mirko, 2020-09-16:
The issue is still present - please don't (auto-)close.

mirko, 2021-05-16:
still there

mirko, 2021-09-14:
still there

jesserockz, 2021-09-14:
Apologies that this issue had "gotten lost".

I have made a PR to allow a `cv.hostname` as well for the server list.

jesserockz, 2021-09-14:
As a side note, creating a new issue is now a form with input boxes which makes it harder to break and have things auto closed by the bot.

closed 2021-09-14 by jesserockz