esphome/issues #1205: bang_bang climate issue?

blademckain, 2020-05-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Docker -> Hassio

**ESP (ESP32/ESP8266, Board/Sonoff):**
Esp32

**Affected component:**
bang_bang

**Description of problem:**
If from Home Assistant you set a temperature_high lower than the current temperature, the bang_bang (correctly) sets the action to OFF. But if later you set a temperature_hight higher than the current temperature, the bang bang does not reactivate and remains off

**Problem-relevant YAML-configuration entries:**
```yaml
climate:
  - platform: bang_bang
    sensor: dht22_t
    name: my_bang_climate
    id: my_bang_climate
    visual:
      min_temperature: 12 °C
      max_temperature: 35 °C
      temperature_step: 0.1 °C
    default_target_temperature_low: 27 °C
    default_target_temperature_high: 28 °C
    away_config:
      default_target_temperature_low: 26 °C
      default_target_temperature_high: 27 °C
    heat_action:
      - logger.log: "Heat ON"
      - switch.turn_on: OU7  
    idle_action:
      - logger.log: "Heat OFF"
      - switch.turn_off: OU7

```

**Logs (if applicable):**

```
[09:25:10][D][climate:010]: 'my_bang_climate' - Setting
[09:25:10][D][climate:020]:   Target Temperature Low: 17.70
[09:25:10][D][climate:023]:   Target Temperature High: 27.40
[09:25:10][D][climate:172]: 'my_bang_climate' - Sending state:
[09:25:10][D][climate:175]:   Mode: AUTO
[09:25:10][D][climate:177]:   Action: HEATING
[09:25:10][D][climate:180]:   Current Temperature: 22.92°C
[09:25:10][D][climate:184]:   Target Temperature: Low: 17.70°C High: 27.40°C
[09:25:10][D][climate:189]:   Away: OFF
[09:25:12][D][climate:010]: 'my_bang_climate' - Setting
[09:25:12][D][climate:020]:   Target Temperature Low: 17.70
[09:25:12][D][climate:023]:   Target Temperature High: 20.80
[09:25:12][D][main:683]: Heat OFF
[09:25:12][D][switch:025]: 'OU7' Turning OFF.
[09:25:12][D][switch:045]: 'OU7': Sending state OFF
[09:25:12][D][climate:172]: 'my_bang_climate' - Sending state:
[09:25:12][D][climate:175]:   Mode: AUTO
[09:25:12][D][climate:177]:   Action: OFF
[09:25:12][D][climate:180]:   Current Temperature: 22.92°C
[09:25:12][D][climate:184]:   Target Temperature: Low: 17.70°C High: 20.80°C
[09:25:12][D][climate:189]:   Away: OFF
[09:25:12][D][climate:172]: 'my_bang_climate' - Sending state:
[09:25:12][D][climate:175]:   Mode: AUTO
[09:25:12][D][climate:177]:   Action: OFF
[09:25:12][D][climate:180]:   Current Temperature: 22.92°C
[09:25:12][D][climate:184]:   Target Temperature: Low: 17.70°C High: 20.80°C
[09:25:12][D][climate:189]:   Away: OFF
[09:25:14][D][climate:010]: 'my_bang_climate' - Setting
[09:25:14][D][climate:020]:   Target Temperature Low: 17.70
[09:25:14][D][climate:023]:   Target Temperature High: 29.30
[09:25:14][D][climate:172]: 'my_bang_climate' - Sending state:
[09:25:14][D][climate:175]:   Mode: AUTO
[09:25:14][D][climate:177]:   Action: OFF
[09:25:14][D][climate:180]:   Current Temperature: 22.92°C
[09:25:14][D][climate:184]:   Target Temperature: Low: 17.70°C High: 29.30°C
[09:25:14][D][climate:189]:   Away: OFF

```

**Additional information and things you've tried:**

xheronimo, 2020-05-13:
Hello, 
In automatic mode it does not work until it drops below the minimum temperature it does not start heating. I have corrected that if it is in heat mode or cool mode with this correction
Test file 
https://github.com/esphome/esphome/pull/1005

blademckain, 2020-05-13:
ok thx u
Is it normal that in an esp32 after changing the temperature presets (default and away), after a restart of the esp32 the values ​​are lost and take the default value in the yaml config?

xheronimo, 2020-05-13:
I have not gotten that to test if you save the update, if I have time and try it

blademckain, 2020-05-14:
I tried it and the values ​​are lost.
Would it be possible to set a "restore_mode" as in the [gpio parameters](https://esphome.io/components/switch/gpio.html)?

It would be important that both the values ​​of the current preset and of the other presets possibly changed by the interface were kept

closed 2020-06-17 by glmnet