esphome/issues #1206: Ina3221 wrong power value

sendorm, 2020-05-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Homeasssistant ver 109.1 on Ubuntu 18.04 LTS python 3.7

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 nodemcu v3

**Affected component:**

https://esphome.io/components/sensor/ina3221.html

**Description of problem:**
The power values shown by the integration is 10 times more. I use the default shunt resistance of 0.1ohms. Only when I use a value of 1ohm the power values are correct. The ina board I am using has a shunt value of R100 (0.1ohms).

**Problem-relevant YAML-configuration entries:**
```yaml
  - platform: ina3221
    address: 0x40
    channel_1:
      shunt_resistance: 1 ohm
      power:
        name: "UPS 12V Power"
      bus_voltage:
        name: "UPS 5V Volt"    
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
Changing the shunt value to 1ohm solved the misreading. Showing that the integration calculates wrongly.

Vxider, 2020-06-27:
You can use a filter
```
filters:
  - multiply: 0.1
```

Vxider, 2020-06-27:
After reading the official datasheet, you can divide 8 to get the right results.

```
filters:
  - multiply: 0.125
```

OttoWinter, 2020-07-24:
Fixed in https://github.com/esphome/esphome/pull/1101

closed 2020-07-24 by OttoWinter