esphome/issues #1207: Sensor send eventsource API one

mohammed-jaouad, 2020-05-09:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32):**

**Affected component:**

https://esphome.io/web-api/index.html

**Description of problem:**
The eventsource api from ESPHOME is working well in my sonoff ( it sends state with interval ) while with ESP32 is sends only once ( when I call the <NODE_URL>/events ), I tried with on_value and publish the state but it's not working too.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp32
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "SSID"
  password: "some password"
  
  manual_ip:
    static_ip: 192.168.x.xxx
    gateway: 192.168.x.xxx
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32 Fallback Hotspot"
    password: "K85j1iDSOQl8"

captive_portal:

# Enable logging
logger:
  
#api:

ota:

sensor:
  - platform: dht
    model: DHT11
    pin: GPIO15
    temperature:
      name: "Living Room Temperature"
    humidity:
      name: "Living Room Humidity"
    update_interval: 20s
      
    
  - platform: adc
    pin: GPIO39
    name: "MQ-2 Gas"
    id: mq2_gas
    update_interval: 20s
    filters:
      - multiply: 100
    unit_of_measurement: "%"

web_server:
  port: 80
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
I tried on_value publish and sensor template as well, also component.update but still not working.

danielehrhardt, 2020-05-18:
I have the same Issue.

As you can see at the Log the State is changing but the state event does not trigger
[image]
here you can see i toggle the Gate4 Switch but i do not have State Events there.

mohammed-jaouad, 2020-08-04:
> I have the same Issue.
> 
> As you can see at the Log, the State is changing but the state event does not trigger
> [image]
> here you can see i toggle the Gate4 Switch but i do not have State Events there.

That's exactly the problem...

I tried with MQTT and it's working well, I'm trying to develop a dashboard for my personal project but i'm still stuck here ...

Is there anyone who had the same problem ? 

Thank you very much in advance guys.

glmnet, 2020-08-04:
There have been changes in these components lately. Can you reproduce with ESPHome beta?

danielehrhardt, 2020-08-04:
i will try it out later

closed 2020-12-16 by stale[bot]