esphome/issues #1213: Can't control warm white brightness on RGBW LED

ofalvai, 2020-05-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip, v1.15.0-dev

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 (inside a [H801 LED controller](https://esphome.io/cookbook/h801.html))

**Affected component:**

[RGBW light](https://esphome.io/components/light/rgbw.html)

**Description of problem:**

After updating from the latest stable release to the latest dev, I can't control the warm white LED independently from the main brightness setting (which controls the brightness of the R, G and B leds).

This setting used to set the warm white LED led to 100% brightness while turning off the other LEDs (regardless of the selected color):
[image]

Now every LED is disabled in this state, the warm white LED only turns on if I increase the brightness slider (which increases the brightness of all the LEDs at once)

**Problem-relevant YAML-configuration entries:**
```yaml
output:
  - platform: esp8266_pwm
    pin: 12
    frequency: 1000 Hz
    id: pwm_b
    max_power: 80%
  - platform: esp8266_pwm
    pin: 13
    frequency: 1000 Hz
    id: pwm_g
  - platform: esp8266_pwm
    pin: 15
    frequency: 1000 Hz
    id: pwm_r
  - platform: esp8266_pwm
    pin: 14
    frequency: 1000 Hz
    id: pwm_w

light:
  - platform: rgbw
    name: "Desk LED strip"
    red: pwm_r
    green: pwm_g
    blue: pwm_b
    white: pwm_w

```

**Logs (if applicable):**

This is the log if I first increase the white slider then the brightness:

```
[13:29:32][D][light:264]: 'Desk LED strip' Setting:
[13:29:32][D][light:286]:   Red=100%, Green=100%, Blue=100%
[13:29:32][D][light:289]:   White Value: 24%
[13:29:32][D][light:303]:   Transition Length: 1.0s
[13:29:33][D][light:264]: 'Desk LED strip' Setting:
[13:29:33][D][light:286]:   Red=100%, Green=100%, Blue=100%
[13:29:33][D][light:289]:   White Value: 57%
[13:29:33][D][light:303]:   Transition Length: 1.0s
[13:29:34][D][light:264]: 'Desk LED strip' Setting:
[13:29:34][D][light:286]:   Red=100%, Green=100%, Blue=100%
[13:29:34][D][light:289]:   White Value: 81%
[13:29:34][D][light:303]:   Transition Length: 1.0s
[13:29:34][D][light:264]: 'Desk LED strip' Setting:
[13:29:34][D][light:286]:   Red=100%, Green=100%, Blue=100%
[13:29:34][D][light:289]:   White Value: 100%
[13:29:34][D][light:303]:   Transition Length: 1.0s
[13:29:55][D][light:264]: 'Desk LED strip' Setting:
[13:29:55][D][light:277]:   Brightness: 17%
[13:29:55][D][light:303]:   Transition Length: 1.0s
[13:29:56][D][light:264]: 'Desk LED strip' Setting:
[13:29:56][D][light:277]:   Brightness: 67%
[13:29:56][D][light:303]:   Transition Length: 1.0s

```

**Additional information and things you've tried:**

ofalvai, 2020-05-12:
I think I tracked it down to this change: https://github.com/esphome/esphome/pull/925/files#diff-bfa351b24f2e7a768492e0dec67e4105L188

While I understand the reasoning behind the PR, this behavior makes it impossible to control the warm white level in case of RGBW lights (where the maximum cold white level is R=G=B=255)

pinkeen, 2020-08-17:
Same problem here. I cannot get warm white using rgbww strip because white enables the rgb components too at the same time as ww producing cold white. 

On the other hand the light seems nice with single chip strip so that got me thinking that it would be possible to control temperature with this kind of strip without cw component. I'm willing to implement this if this is something that could be merged into mainline.

ivanovd, 2020-09-21:
Any updates on this? I updated to the latest version of ESPHome and now I cannot independently control the White channel without messing with the brightness of the RGB LEDs.

Thankfully I keep the old bin, however, I cannot make adjustments to the yaml configuration now as it will bring the new behavior back.

peterforeman, 2020-09-25:
I was already wondering why my RGBW (actually, the W) strips did not seem to work anymore. Can we get this fixed? It breaks significant functionality.

speldos, 2020-10-02:
Same here. Ww is not working as before. If i increase ww the rgb tuns to bright white.

CornHead764, 2020-10-10:
Stumbled upon this thread while troubleshooting a similar problem... Does the color_interlock option solve your problem? 

https://esphome.io/components/light/rgbw.html

Here's some copy/paste:

color_interlock (Optional, boolean): When enabled, this will prevent white leds being on at the same time as RGB leds. See Color Interlock for more information. Defaults to false.

Color Interlock
With some LED bulbs, setting the RGB channels to maximum whilst wanting a white light will have a undesired hue affect. Additionally, the brightness command may not work as expected depending upon configuration, leaving users to adjust the white component level separately. For these cases a new configration variable has been added: color_interlock.

Setting this variable to True will turn off RGB leds when white value is above 0 (or if they are to 255,255,255) and turn off white leds if color is not set to 255,255,255. This also allows the brightness parameter to control the intensity of the white leds.

margau, 2020-10-10:
Can you have a look, maybe https://github.com/esphome/esphome/pull/1325 also solves this issued?

MattXcz, 2020-10-13:
I will try it today

closed 2021-02-24 by stale[bot]

SpacebarStar, 2021-04-24:
Hey, is there any update on this? I'm still experiencing exactly the same issue with the Home Assistant add-on version 1.16.2

nickyivyca, 2021-04-26:
I'm still experiencing this as well. With 1.16.2, same behavior as the original post here. https://github.com/esphome/esphome/pull/1325 Did not fix the issue apparently.

MattXcz, 2021-04-26:
Best workaround is to use RGB + W as seperte light.

YSmetana, 2021-05-16:
Same here. From 1.16.2 can not use my light automation properly (controlling W-channel separatelly).

craigueh, 2021-08-08:
This seems to have broken White control on neopixel RBGW lights. I have 3 sets of leds and since updating them to latest version of esphome I'm unable to have white only lighting. The RGB channels are always enabled and can't turn them down when the white channel is turned up. This leaves the colour as kind of blueish.
I tried using color_interlock: false but this won't compile as i presume it's only supported on RGBW platform.

MattXcz, 2021-08-09:
> This seems to have broken White control on neopixel RBGW lights. I have 3 sets of leds and since updating them to latest version of esphome I'm unable to have white only lighting. The RGB channels [...]

You can set it as two separate lights... monochrome white + RGB if that fits you.