esphome/issues #1214: Pulse counter is incremented by 12 despite clear one pulse

abrilevskiy, 2020-05-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

https://esphome.io/components/sensor/pulse_counter.html

**Description of problem:**
Pulse counter is incremented by 12. Pin 12 of esp8266 is connected to Raspberry PI where gpio write 1/0 is used to trigger pulses.  Checked with logic analyser - there is no bouncing (100Ms/s is used to capture data)

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: water_electricity
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: ...
  password: ...
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: ...
    password: ...

captive_portal:

debug:

# Enable logging
logger:
  level: VERBOSE
  logs:
    mqtt.component: ERROR
    mqtt.client: ERROR

# Enable Home Assistant API
api:

ota:

web_server:
  port: 80

  #binary_sensor:
  #  - platform: status
  #  name: "Test"
    #  - platform: gpio
    #pin: 12
    #name: "12 pin"

sensor:
        #  - platform: uptime
        #    name: "Test Uptime Sensor"
        #  - platform: wifi_signal
        #    name: "WiFi Signal Sensor"
        #    update_interval: 10s
    #- platform: pulse_width
    #pin: 04
    #name: "Pulse Width Sensor 4"
    #update_interval: 5s
  - platform: pulse_counter
    pin: 12
    name: "Pulse Counter 12"
    update_interval: 5s
    internal_filter: 1ms

mqtt:
  broker: ...
  username: ...
  password: ...

  #dallas:
  #- pin: GPIO13

```

**Logs (if applicable):**

```
[13:52:52][V][sensor:013]: 'Pulse Counter 12': Received new state 0.000000
[13:52:52][D][sensor:092]: 'Pulse Counter 12': Sending state 0.00000 pulses/min with 2 decimals of accuracy
[13:52:52][V][mqtt:372]: Publish(topic='water_electricity/sensor/pulse_counter_12/state' payload='0.00' retain=1)
[13:52:57][D][pulse_counter:159]: 'Pulse Counter 12': Retrieved counter: 0.00 pulses/min
[13:52:57][V][sensor:013]: 'Pulse Counter 12': Received new state 0.000000
[13:52:57][D][sensor:092]: 'Pulse Counter 12': Sending state 0.00000 pulses/min with 2 decimals of accuracy
[13:52:57][V][mqtt:372]: Publish(topic='water_electricity/sensor/pulse_counter_12/state' payload='0.00' retain=1)
[13:53:02][D][pulse_counter:159]: 'Pulse Counter 12': Retrieved counter: 12.00 pulses/min
[13:53:02][V][sensor:013]: 'Pulse Counter 12': Received new state 12.000000
[13:53:02][D][sensor:092]: 'Pulse Counter 12': Sending state 12.00000 pulses/min with 2 decimals of accuracy
[13:53:02][V][mqtt:372]: Publish(topic='water_electricity/sensor/pulse_counter_12/state' payload='12.00' retain=1)
[13:53:07][D][pulse_counter:159]: 'Pulse Counter 12': Retrieved counter: 0.00 pulses/min
[13:53:07][V][sensor:013]: 'Pulse Counter 12': Received new state 0.000000
[13:53:07][D][sensor:092]: 'Pulse Counter 12': Sending state 0.00000 pulses/min with 2 decimals of accuracy

```

**Additional information and things you've tried:**

kluszczyn, 2020-05-14:
By default `pulse_counter` sensor reports in `pulses/min` - see docs https://next.esphome.io/components/sensor/pulse_counter.html?highlight=pulse#converting-units

abrilevskiy, 2020-05-14:
Thank you! Closing the issue.

closed 2020-05-14 by abrilevskiy