esphome/issues #1215: Enabling ble tracker disconnects wifi

isabellaalstrom, 2020-05-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant (former Hass.io) with core version 0.109.6
**ESP (ESP32/ESP8266, Board/Sonoff):**

m5stack esp32 with camera
**Affected component:**

https://esphome.io/components/esp32_ble_tracker.html
https://esphome.io/components/wifi.html

**Description of problem:**
As soon as I enable the esp32_ble_tracker component the device can no longer connect to wifi. I have attached the entire upload log (connected via usb). If I remove the ble component the device can connect to wifi fine again. I tried with adding scan_parameters after reading some other similar issues regarding ble and ota, but to no avail.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: upstairs_litterbox
  platform: ESP32
  board: m5stack-core-esp32

esp32_camera:
  external_clock:
    pin: GPIO27
    frequency: 20MHz
  i2c_pins:
    sda: GPIO25
    scl: GPIO23
  data_pins: [GPIO17, GPIO35, GPIO34, GPIO5, GPIO39, GPIO18, GPIO36, GPIO19]
  vsync_pin: GPIO22
  href_pin: GPIO26
  pixel_clock_pin: GPIO21
  reset_pin: GPIO15

  name: Upstairs Litterbox

esp32_ble_tracker:
  scan_parameters:
    interval: 512ms
    window: 48ms

wifi:
  ssid: redacted
  password: redacted

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: redacted

ota:
  password: redacted

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
INFO Reading configuration /config/esphome/Upstairs Litterbox.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/upstairs_litterbox
Processing upstairs_litterbox (board: m5stack-core-esp32; framework: arduino; platform: espressif32@1.11.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
[... 192 lines cut ...]
[13:05:17][D][esp32_ble_tracker:477]:   Address Type: RANDOM
[13:05:20][D][esp32_ble_tracker:456]: Found device 52:10:ED:6E:69:FB RSSI=-74
[13:05:20][D][esp32_ble_tracker:477]:   Address Type: RANDOM
[13:05:20][D][esp32_ble_tracker:481]:   TX Power: 2
[13:05:25][E][wifi:312]: Scan timeout!
[13:05:30][D][wifi:304]: Starting scan...
[13:05:30][D][esp-idf:000]: E (36011) wifi: create wifi task: failed to create task
[13:05:30][E][WiFiGeneric.cpp:117] wifiLowLevelInit(): esp_wifi_init 257
[13:06:00][E][wifi:312]: Scan timeout!
[13:06:00][D][esp32_camera:157]: Got Image: len=20591
[13:06:05][D][wifi:304]: Starting scan...
[13:06:05][D][esp-idf:000]: E (71021) wifi: create wifi task: failed to create task
[13:06:05][E][WiFiGeneric.cpp:117] wifiLowLevelInit(): esp_wifi_init 257
[13:06:21][D][xiaomi_ble:158]: Got Xiaomi HHCCJCY01 (C4:7C:8D:60:A5:B3):
[13:06:21][D][xiaomi_ble:173]:   Illuminance: 213lx
[13:06:35][E][wifi:312]: Scan timeout!
[13:06:40][D][wifi:304]: Starting scan...
[13:06:40][D][esp-idf:000]: E (106033) wifi: create wifi task: failed to create task
[13:06:40][E][WiFiGeneric.cpp:117] wifiLowLevelInit(): esp_wifi_init 257
[13:06:40]
[13:06:42][D][xiaomi_ble:158]: Got Xiaomi HHCCJCY01 (C4:7C:8D:60:A5:B3):
[13:06:42][D][xiaomi_ble:170]:   Conductivity: 0µS/cm
```

**Additional information and things you've tried:**

ronytomen, 2020-05-24:
Wifi and Bluetooth are using the same radio and antenna on ESP32.  So, the device has to move between the two, only one can be active at a time.  But also the more the ESP32 is trying to do the more chance you have of missing the window and causing problems with Wifi. 

I wonder if you disable the camera if you no longer have problems?

You can play with scanning intervals for the Bluetooth part, but I just gave up and ended up using an Ethernet enabled ESP32 to be my BLE bridge.

obodnikov, 2020-07-08:
Hi All

It is looks like the issue is not about switching between two radio components but too many memory used by BLE component

See other issues 
https://github.com/esphome/issues/issues/1336
https://github.com/esphome/issues/issues/1308

Also, both WiFi and BLE works just fine for me with other display
```yaml
i2c:
  sda: 5
  scl: 4

display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    address: 0x3C
    rotation: 180
    id: Display
    lambda: |-
      it.strftime(128, 0, id(font_FreeSerif12),TextAlign::TOP_RIGHT , "%d-%m-%Y %H:%M", id(sntp_time).now());

esp32_ble_tracker:
```

closed 2020-11-19 by stale[bot]

koenvervloesem, 2020-12-02:
I discovered the same behavior on an M5Stack Basic Core, with the same `m5stack-core-esp32` board. Wi-Fi worked perfectly, but as soon as I added the `esp32_ble_tracker` component, Wi-Fi failed:

```
[13:53:56][D][esp32_ble_tracker:148]: Starting scan...
[13:53:56][D][binary_sensor:034]: 'connection_status': Sending initial state OFF
[13:53:57][C][wifi:033]: Setting up WiFi...
[13:53:57][D][esp-idf:000]: E (2786) wifi: Expected to init 16 rx buffer, actual is 1
[13:53:57]
[13:53:57][E][WiFiGeneric.cpp:117] wifiLowLevelInit(): esp_wifi_init 257
[13:53:57][W][wifi_esp32:054]: Setting WiFi mode failed!
[13:53:57][D][wifi:324]: Starting scan...
[13:53:57][D][esp-idf:000]: E (2817) wifi: Expected to init 16 rx buffer, actual is 0
[13:53:57]
[13:53:57][E][WiFiGeneric.cpp:117] wifiLowLevelInit(): esp_wifi_init 257
[13:53:57][W][wifi_esp32:054]: Setting WiFi mode failed!
[13:53:57][E][ESPmDNS.cpp:65] begin(): Failed starting MDNS
[13:53:57][E][ESPmDNS.cpp:148] addService(): Failed adding service http.tcp.
[13:53:57]
[13:53:57][E][ESPmDNS.cpp:167] addServiceTxt(): Failed setting service TXT
[13:53:57][D][esp32_ble_tracker:544]: Found device E7:2E:00:B1:38:96 RSSI=-74
[13:53:57][D][esp32_ble_tracker:565]:   Address Type: PUBLIC
```
The BLE tracker works.

iondulgheru, 2021-05-18:
@koenvervloesem , I have the same issue.
Did you find a workaround for this?

koenvervloesem, 2021-05-18:
No, as explained in https://github.com/esphome/issues/issues/1215#issuecomment-655651301 it's a memory issue.