esphome/issues #1216: RC switch and Remote Transmitter problems

robbz23, 2020-05-13:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Wemos D1

Remote Transmitter

https://esphome.io/components/remote_transmitter.html

Right now I control many 433 lights around the house using Home assistant and the rpi_rf platform but I am trying to move away from the raspberry pi platform and I would like a Wemos to take over this function.  So I have been playing with esphome and I can't get it to work.  I tried so many different combinations of types of switch including raw and I can't get esphome to put anything out.  While I was testing I had an Arduino setup running the [rc_switch Arduino example](https://github.com/sui77/rc-switch/blob/master/examples/ReceiveDemo_Advanced/ReceiveDemo_Advanced.ino).  I had no problems reading the remotes if I pushed the buttons but when I tried to send the codes from esphome I received no response at all on the arduino.  So I thought it could be something with my hardware setup, and I tested the same setup but instead I used the [send demo](https://github.com/sui77/rc-switch/tree/master/examples/SendDemo) looping over and over.  That had no problem switching the light on with the same exact hardware setup.  So now I am almost positive it is a problem with esphome or the configuration.  

**Problem-relevant YAML-configuration entries:**
```yaml

esphome:
  name: rotary
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: 
  password: 

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Rotary Fallback Hotspot"
    password: ""

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

remote_transmitter:
  - pin: D5
    id: rf_trans
    carrier_duty_percent: 100%
        
  - platform: template
    name: kitchen_1_on
    turn_on_action:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '000000000000000000010101'
          protocol: 1
          repeat: 
            times: 10
            wait_time: 500ms
          transmitter_id: rf_trans
  - platform: template
    name: kitchen_1_off
    turn_on_action:
       - remote_transmitter.transmit_rc_switch_raw:
          code: '000101010001010101010100'
          protocol: 1
          transmitter_id: rf_trans

  - platform: template
    name: kitchen_2_on
    turn_on_action:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '000101010001010101010100'
          protocol: 1
          repeat: 10   
          transmitter_id: rf_trans
  - platform: template
    name: kitchen_2_off
    turn_on_action:
      - remote_transmitter.transmit_rc_switch_raw:
          code: '000101010001010101010100'
          protocol: 1
          repeat: 10   
          transmitter_id: rf_trans
  
          
          
    
```

**Logs (if applicable):**

```
Sorry I didnt get the logs but I have checked that it is attempting to send the codes to according to the logs

```

I am also using this same D1 as a remote transmitter for ir, but I have tested this with that section completely commented out.

mikalauskas, 2020-05-19:
In my experience, i got it working only after soldering proper antenna (~17cm). I've had the same issues and i think it's a component problem in esphome. Somehow it works very different from other libraries. 

This is my config, if you are curious. Not much difference from yours.

```
esphome:
  name: rf_bridge
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: ""
  password: ""
  domain: ".lan"

logger:

api:
  services:
    - service: rf_bridge_send
      variables:
        code: string
        repeat: int
      then:
        - remote_transmitter.transmit_rc_switch_raw:
            code: !lambda 'return code;'
            protocol: 1
            repeat:
              times: !lambda 'return repeat;'
        - delay: 1s

ota:

remote_transmitter:
  pin:
    number: D6
  carrier_duty_percent: 100%

remote_receiver:
  pin:
    number: D5
    inverted: False
  dump: rc_switch
  tolerance: 50%
  filter: 250us
  idle: 2ms
  buffer_size: 2kb
#  on_rc_switch:
#    then:
#      - sensor.template.publish:
#          id: template_sens
#          state: !lambda 'return x.code;'
#
#    - homeassistant.event:
#        event: esphome.remote_received
#        data:
#          code: !lambda 'return x.code;'
#          protocol: !lambda 'return x.protocol;'

switch:
  - platform: template
    name: "rfoutlet1"
    id: rfoutlet1
    restore_state: true
    optimistic: false
    turn_on_action:
      - delay: 75ms
      - remote_transmitter.transmit_rc_switch_raw:
          code: '000101010001010101010111'
          protocol: 1
[... 181 lines cut ...]
      - remote_transmitter.transmit_rc_switch_raw:
          code: '010001010101000101010100'
          protocol: 1
          repeat: 7
      - switch.template.publish:
          id: rfoutlet6
          state: OFF

binary_sensor:
  - platform: remote_receiver
    name: "sonoff_rf_pir_1"
    device_class: motion
    filters:
      - delayed_off: 10s
    rc_switch_raw:
      code: '111110001001000101101110'
  - platform: remote_receiver
    name: "sonoff_rf_pir_2"
    device_class: motion
    filters:
      - delayed_off: 10s
    rc_switch_raw:
      code: '111110001001000101111110'

```

jaykumar2001, 2020-08-19:
:+1: 

Can someone confirm if you can recieved/transmit across Esphome ==> Arduino running SendDemo or ReceiveDemo from 
  https://github.com/sui77/rc-switch/ 

For me Arduino to Arduino combination works fine, but can't seem to get Esphome + Esp8266 talk to Arduino

jpoles1, 2020-09-19:
@jaykumar2001 I'm in the same situation, using arduino to transmit over 433MHz works just fine but it's not functional using ESPHOME. I've looked at the signal using SDR and the two signals appear completely different as well... very frustrating.

closed 2021-01-31 by stale[bot]