esphome/issues #1218: ESP Home addon does not see the sensors anymore

pr0fetul, 2020-05-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io version 0.109.6 in Docker 
**ESP (ESP32/ESP8266, Board/Sonoff):**
 ESP8266 - 2 boards - same behaviour

**Affected component:**
 The ESP home addon does not see the sensors anymore. This happens on 2 identical sensors. 

**Description of problem:**
The ESP home addon does not see the sensors anymore. I have checked the router - the sensors have static IP addresses. I have unplugged the sensors, replugged them - they show up in router under DHCP with static IP addresses, ESP does not see them. This is the error message from Show Logs:
`
INFO Reading configuration /config/esphome/bedroom_sensor.yaml...
INFO Starting log output from bedroom_sensor.local using esphome API
WARNING Error resolving IP address of bedroom_sensor.local. Is it connected to WiFi?
WARNING (If this error persists, please set a static IP address: https://esphome.io/components/wifi.html#manual-ips)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error resolving IP address: Error resolving address with mDNS: Did not respond. Maybe the device is offline., [Errno -2] Name or service not known). Re-Trying in 1 seconds
`
The sensors worked fine with this version of HA until last night - this morning the communication "died" . 

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: bedroom_sensor
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "xxxx"
  password: "yyyy"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "bedroom_sensor"
    password: "zzzzzz"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

i2c:
  sda: D2
  scl: D1
  scan: True

sensor:
#WiFi Sensor for the NodeMCU
  - platform: wifi_signal
    name: "WiFi Signal Sensor"
    update_interval: 60s
    filters:
      - lambda: |-
          if (x <= -100) {
            return 0;
          } else if (x >= -50) {
            return 100;
          } else {
            return (2 * (x + 100));
          }
    unit_of_measurement: "%"

#UpTime sensor for the NodeMCU
  - platform: uptime
    name: Uptime Sensor
    filters:
      - lambda: |-
          return x / 3600;
    unit_of_measurement: "Hours"

#Light Sensor:
  - platform: bh1750
    name: "Bedroom Luminosity"
    address: 0x23
    update_interval: 60s
    icon: "mdi:brightness-6"

#Temperature Humidity and Pressure Sensor:
  - platform: bme280
    temperature:
      name: "Bedroom Temperature"
      oversampling: 16x
      id: temperature
    pressure:
      name: "Bedroom Pressure"
    humidity:
      name: "Bedroom Humidity"
      oversampling: 16x
      id: humidity
    address: 0x76
    update_interval: 60s

#Air Quality Sensor:
  - platform: ccs811
    eco2:
      name: "Bedroom eCO2 Value"
    tvoc:
      name: "Bedroom Total Volatile Organic Compound"
    address: 0x5A
    update_interval: 60s
    baseline: 0x4481
    temperature: temperature
    humidity: humidity

#Text sensor for the NodeMCU
text_sensor:
  - platform: version
    name: "ESPHome Version"
  - platform: wifi_info
    ssid:
      name: xxxxxx
    

#Reboot Switch for the NodeMCU
switch:
  - platform: restart
    name: "NodeMCU Restart"

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

aptonline, 2020-06-12:
Seeing a similar issue in 0.110.1 and latest ESPHome. I believe it’s an mdns issue of some description.

meichthys, 2020-07-13:
I'm also seeing this in 0.111.4
All of my EspHome sensors are now offline. I did add a network interface to homeassistant, so it's possible that caused an issue for me.

glmnet, 2020-07-13:
It happened to me yesterday that a node got disconnected, never had an issue with it. Looked ok via ping and remote logs. Restarting HA solved the issue, what steps are you taking to bring the node back online?

meichthys, 2020-07-13:
Restarting HA didn’t help for me. I also tried reinstalling The ESPhome addon but that hasn’t worked. I’ve seen some other issues that mention that it is an issue with using a reverse proxy, but even accessing it locally doesn’t seem to help for me. I’ve had this happen once or twice in the past and it seemed to fix itself (maybe it was a router reboot or something network related).

bmbouter, 2020-07-17:
I am also experiencing this issue. I don't think it's related to Hass version as much as the ESPhome itself from 1.14.3 ➡️1.14.5. I believe the significant symptom is that the devices themselves are no longer maintaining their same hostnames or not responding to mDNS somehow. Hass couldn't interfere with this, and here's why I believe that.

My network supports mDNS, I can from a computer on the LAN ping those mDNS names, so that is working fine. What isn't working is pinging these devices showing "offline". These connections would be between my computer <-> Wifi <-> switch <-> Wifi <-> esphome_device. Hass wouldn't be involved.

Interestingly, hass established connections with these devices and I can still control them! So really what isn't working is the mDNS part (as described here by others also). To restate what is working, if I connect to the logs port for example by IP address directly the device is actually online and working. If I try to connect to the device by mDNS it does not...

I've been using this setup for 12ish months rock solid. About 10 days ago I updated from 1.14.3 ➡️1.14.5, rebuilt my code for all devices and pushed (big mistake). Now all of them show offline. They come online for a few minutes when I restart them. Something between 1.14.3 and 1.14.5. in the client code has regressed this. I use 8266 deices (mix of nodemcu's and sonoffs).

meichthys, 2020-07-17:
^This. 
@bmbouter great summary. 
One thing to add: I've noticed that if you create a device with a static IP address it will still show as online - further supporting the idea that it seems like an mDNS issue.

h0jeZvgoxFepBQ2C, 2020-07-25:
I think I also have this problem: https://community.home-assistant.io/t/esphome-device-says-connected-successfully-is-sending-sensor-data-but-appears-completely-offline-in-home-assistang-no-mqtt/214025

bmbouter, 2020-07-26:
> I think I also have this problem: https://community.home-assistant.io/t/esphome-device-says-connected-successfully-is-sending-sensor-data-but-appears-completely-offline-in-home-assistang-no-mqtt/214025

The symptoms you describe are different. You can push firmware, but with mDNS not working I cannot. Also I cannot show the logs because of mDNS issue, but it sounds like you can.

Perhaps they are the same, but the symptoms sound different from the description.

PedroLamas, 2020-07-26:
Seeing this issue myself, only noticed yesterday after I rebuilt and re-deployed one of my nodes... after that mDNS for that node (ESP8266) stopped showing up...

Tokn59, 2020-08-02:
Same problem here but I was able to add new sensors through configuration - ESPHome - and then adding the node with it's IP address .... No automatic discovery ...

OttoWinter, 2020-08-06:
Not sure what this is, and haven't been able to reproduce on my testing RPi 3b+ Hass.io arm64v8 system.

Might be related to the supervisor now having its own mDNS listener. The fact that this appears to be semi-random makes that less likely though.

PedroLamas, 2020-08-06:
@OttoWinter I for one don't think this has got anything to do with Hass.io integration!

Using my Network Analyzer app in my Android phone, I can't see any mDNS announcement from my ESP8266 node, yet I do see for everything else on the network (both wired and wireless devices appear fine)

h0jeZvgoxFepBQ2C, 2020-08-06:
I also don't have a supervisor afaik (native linux installation).

PedroLamas, 2020-08-06:
Actually, now that you mention that, I too am not using Hass.io Supervisor! My install is everything under docker, attached to the host network.

Here's my node config just in case:

```yaml
esphome:
  name: testnode
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: !secret wifi_domain
  fast_connect: true
  reboot_timeout: 3min

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Testnode Fallback Hotspot"
    password: !secret wifi_ap_password

sensor:
  - platform: pulse_counter
    pin: GPIO12
    unit_of_measurement: 'kW'
    name: 'Power Meter'
    count_mode:
      rising_edge: DISABLE
      falling_edge: INCREMENT
    internal_filter: 1000us
    filters:
      - multiply: 0.075
    id: my_power
  - platform: total_daily_energy
    name: "Total Daily Energy"
    power_id: my_power

time:
  - platform: sntp
    id: my_time

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: !secret api_password

ota:
  password: !secret ota_password
```

bmbouter, 2020-08-06:
My testing was between my laptop and the esp8266 device itself. My ping tests using the mDNS name do no work. In that test, hass.io is just some other random client on the network. This brings me to the conclusion that this has nothing to do with hass.io or hass.

dkliban, 2020-08-12:
@bmbouter  I was able to push an update to my device using ESPHome 1.14.4. I then restarted the device and was still able to continue pinging it from my laptop.

meichthys, 2020-08-20:
Seems to be working for me - although i can't pinpoint why it started working again.

h0jeZvgoxFepBQ2C, 2020-08-20:
Can't confirm it, I updated already to the latest versions days ago and I still have the same problems :(

meichthys, 2020-08-20:
@h0jeZvgoxFepBQ2C  What version of Homeassistant and what version of the HomAssistant Supervisor are you on?

h0jeZvgoxFepBQ2C, 2020-08-20:
Updated now Home assistant again, still doesn't work. Running now with Esphome 1.15.0b3 and Home assistant 0.114.3

meichthys, 2020-09-10:
@pr0fetul i think we can close this. Can you confirm if things are working for you on the latest version of Homeassistant?

pr0fetul, 2020-09-10:
Hi, yes, on my side it work fine now.

Thank you,
Vlad Chiriloiu

bmbouter, 2020-10-07:
Aaaaand it's back. I'm using esphome version `1.15.2`. Here's what I experienced:

1. I have a sensor that was working with version `1.14.4`
2. I make no changes and click "upload" to upgrade it to `1.15.2`
3. The upload completes successfully and the logs are read back after the ESP reboots
4. esphome portal now shows it as offline even though I can ping it via it's IP address
5. Also my laptop (on the same wifi as the ESP w/ no firewall rules) cannot ping it by it's sensor_name.local hostname

bmbouter, 2020-10-07:
So this is interesting. Upon further testing, when the device is first powered on its mDNS does work. Both esphome's UI and another computer using `ping node_name.local` also works. Then after about 30-45 minutes, the node stops responding to mDNS but it is still online. I know this because I am reading the log output during both the working and non-working period, the logs show regular output, with no distinguishable change. Also while the node stops responding to mDNS it is still pingable by its IP.

Rebooting the node puts it back into the same pattern. It works ... time passes ... then it doesn't.

Also, I just flashed a new ESP with esphome `1.15.2` and it is showing the same pattern. Specifically it works initially, but then after some time mDNS stops responding.

bmbouter, 2020-10-07:
I monitored the logs locally on this device and I'm seeing some strange stuff. The device itself seems to be legitimately going offline for some reason. It boots up, runs normally for a long time, then eventually resets with this error message:

```
[15:42:36][E][api:084]: No client connected to API. Rebooting...
[15:42:36][I][app:132]: Forcing a reboot...
[15:42:36][W][wifi_esp8266:395]: Event: Disconnected ssid='1805' bssid=F0:79:59:75:B3:6A reason='Association Leave'
[15:42:36]
[15:42:36] ets Jan  8 2013,rst cause:2, boot mode:(3,6)
[15:42:36]
[15:42:36]load 0x4010f000, len 3584, room 16 
[15:42:36]tail 0
[15:42:36]chksum 0xb0
[15:42:36]csum 0xb0
[15:42:36]v2843a5ac
[15:42:36]~ld
[15:42:36][I][logger:166]: Log initialized
[15:42:36][C][ota:366]: There have been 0 suspected unsuccessful boot attempts.
[15:42:36][I][app:029]: Running through setup()...
[15:42:36][D][binary_sensor:034]: 'Upstairs Bathroom Motion': Sending initial state OFF
[15:42:36][C][dht:011]: Setting up DHT...
[15:42:36][C][wifi:033]: Setting up WiFi...
[15:42:36][D][wifi:324]: Starting scan...
```

sparky56, 2020-10-24:
I have had this problem for weeks.  Reload, restart, delete, reload, restart..... nothing helps.
below is my info from ESP home.   The only thing I see is in the logs, my sensors are working.  
Just not in my dashboard

Add-on version: 1.15.3
 You are running the latest version of this add-on.
 System: HassOS 4.15  (armv7 / raspberrypi3)
 Home Assistant Core: 0.116.4
 Home Assistant Supervisor: 249

bmbouter, 2021-02-22:
What I think is really going on here is:

1. A device has been online for a while, longer than its DHCP lease time
2. I push a new esphome config to it. It accepts it and restarts
3. The device gets a new IP address
4. The container still has a cached copy of the mDNS resolution to the old IP.
5. The esphome UI shows the device as offline until the mDNS record ages out.

So I think a) things are mostly working and b) the issue is actually in the esphome code with the UI, not the end device.

Trilis29, 2021-03-11:
I have the same problem. Tried to ping mdns nodes from RPI4 (Hassio on docker is ther) all ok. Nor mdns nor static ip works on home assistant. Even esphome does not show red/green stripe upon. There should be problem in home-assistant or esphome. 

To reproduce I have installed on another device docker esphome and it works flawlessly.

556duckvader, 2021-04-02:
I am having the issue with my ESPHome devices not coming back online after restarting Home Assistant.  I can go into to the ESPHome Web UI and look at the log for the devices and they show up and appear to be working just fine and I can ping them too.  But they all show up as unavailable.

Theoreet, 2021-04-06:
Exactly the same problem as above. Some Action ( LSC) devices worked fine before, but now they where unavailable. 
Ping: working fine
Web server: working fine 
Debug log accessible via Webserver  and/of ESPHome addon : reporting fine (status /uptime/ state etc.)  

Running ESPHome 1.16.2, Home Assistant 2021.3.4 ( core-2021.3.4, supervisor-2021.03.9 ) 

Is tried several things, like re-upload the config via OTA, removed the enitities, cleaned the ESPHome folder, restarted serveral times, but nothing helps.
After i removed the devices, it's not availble anymore on the configuration -> integration anymore, and the sensors are "undetected" and not available on the Developer Tools --> Current entities .

So what can i do / try next ?

elaiel, 2021-05-17:
I have a very similar issue, not sure if it is exacly the same. At some point Home Assistant (2021.4.x) started showing my sensor readings from esphome (v1.19.0) based nodemcu's with bme280 as offline. The devices are working, the firmware is up-to-date, and data is send (confirmed via esp-home).
If I delete the integrations for the nodemcu's and restart homeassistant (v2021.5.x), the nodemcus are auto discovered, can be integrated and work as designed (sensor data is available). But if I restart homeassistant, no sensor data is available anymore, so I have to delete the integrations again, restart homeassistant, re-add them so they will work till the next restart.

bmbouter, 2021-09-14:
It's still an issue AFAICT.

ErikRust, 2021-09-20:
Still an issue, particularly on poe esp32

my-wheel, 2021-12-09:
I started to get the same problem when switching from a pip-based installation to a Docker-based installation.
It seems that the IP-address can not be resolve when the hostname (here: "bedroom_sensor") is extended by ".local"
(here: "bedroom_sensor.local").
I found a workaround that might help you until the issue is resolved.
Taking pr0fetul's example, you just need to add a single line with **use_address** in the wifi section:

esphome:
  name: bedroom_sensor
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "xxxx"
  password: "yyyy"
  use_address: "bedroom_sensor"

This essentially forces the system to use the original name without the .local extension.

truglodite, 2021-12-30:
Not sure if it's even related, but I was dealing with a similar issue on one of my esphome 8266 devices. My numerous other esphome devices are working fine, but one of them kept showing up as "unavailable" in HA after restarting the HA. Also the device still seemed to be working as expected despite all the HA entities showing unavailable (logs show correct data in the esphome webui).

I tried many things, including 'hacking' storage files to try and get the device working. What finally worked for me was removing "encryption" from the device. Now the device is working just as reliably as it was in previous years. I do have several other devices working fine with encryption, but the device in question also has a lot more going on (perhaps 8266 ram maxed out and encryption failing). On a side note, the device in question was also in a spot with poor wifi (~80db). Not sure if that has something to do with encryption not 'reconnecting' like it should.

cjj25, 2022-01-11:
Change "status_use_ping" to "true" - this will use ICMP pings instead of mDNS to check its status