esphome/issues #1220: APDS9960 stops reading proximity above 16%

HFsi, 2020-05-14:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP01S

**Affected component:**

https://esphome.io/components/sensor/apds9960.html

**Description of problem:**
When the proximity sensor reachs 16%, it stops reading and logging

**Problem-relevant YAML-configuration entries:**
```yaml

esphome:
  name: o1s
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "xxx"
  password: "xxxxxxxxxx"

logger:

api:

ota:

i2c:
  sda: GPIO2
  scl: TX
  
apds9960:
  address: 0x39
  update_interval: 1s
  
sensor:
  - platform: apds9960
    type: PROXIMITY
    name: "APDS9960 Clear Channel"
    
binary_sensor:
  - platform: apds9960
    direction: UP
    name: "APDS9960 Up"
    
  - platform: apds9960
    direction: DOWN
    name: "APDS9960 Down"
    
  - platform: apds9960
    direction: LEFT
    name: "APDS9960 Left"
    
  - platform: apds9960
    direction: RIGHT
    name: "APDS9960 Right"
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
(in video)
```

**Additional information and things you've tried:**

I made a video that gives a good idea of the issue, I think

https://www.youtube.com/watch?v=-Ok8f71xhGk

rradar, 2020-05-15:
@HFsi in your video you make quite big movements. Did you keep in mind that the operating range for this sensor is only 10-20cm? Can you attach a `very_verbose` logging and tell how far/close you are to the sensor?

closed 2020-09-26 by stale[bot]

jackking444, 2021-03-28:
I have the same problem