esphome/issues #1221: I2C Failure:  Received NACK on transmit of address 0x76

deftdawg, 2020-05-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 / NodeMCU 4MB Flash

**Affected component:**

I2C Bus (BME280)

**Description of problem:**
I have an NodeMCU ESP8266 (4MB) running ESPHome, it is connected to a BME280 sensor on I2C at address 0x76.  The hardware and configuration work, however after some period of time often between 24h to 140h,  ESPHome will report errors in the I2C bus and sensor values will no longer be read. 

There are a few problems: 
1. Obviously it would be great if this problem didn't happen at all or if the I2C bus reset itself and recovered on its own.
2. A reboot does *not* fix the problem, power must be interrupted to the ESP8266 before the I2C bus detects the sensor and reports values again.   This sucks because I can not compensate for this issue by rebooting the ESP.
3. Don't know of a way to detect the issue within Home Assistant other than to alert if values stagnate for 3 polling periods or something like that.   Does HA have way to read the logs to see NACKs?

**Problem-relevant YAML-configuration entries:**
```yaml
i2c:
  sda: D6
  scl: D5
  scan: True
  id: bus_a

sensor:
  # Uptime sensor.
  - platform: uptime
    name: ESP Uptime
    filters:
    - lambda: return x / 3600;
    unit_of_measurement: "h"

  - platform: bme280
    temperature:
      name: "BME280 Temperature"
      id: bme280_temperature
    pressure:
      name: "BME280 Pressure"
      id: bme280_pressure
    humidity:
      name: "BME280 Relative Humidity"
      id: bme280_humidity
    address: 0x76
    update_interval: 30s
```

**Logs (if applicable):**

Log from the device of it occurring:
```
[D][sensor:092]: 'ESP Uptime': Sending state 24.86042 h with 0 decimals of accuracy
[D][bme280.sensor:197]: Got temperature=21.4°C pressure=995.7hPa humidity=40.0%
[D][sensor:092]: 'BME280 Temperature': Sending state 21.35000 °C with 1 decimals of accuracy
[D][sensor:092]: 'BME280 Pressure': Sending state 995.73346 hPa with 1 decimals of accuracy
[D][sensor:092]: 'BME280 Relative Humidity': Sending state 39.97852 % with 1 decimals of accuracy
[D][bme280.sensor:197]: Got temperature=21.4°C pressure=995.8hPa humidity=41.0%
[D][sensor:092]: 'BME280 Temperature': Sending state 21.35000 °C with 1 decimals of accuracy
[D][sensor:092]: 'BME280 Pressure': Sending state 995.75269 hPa with 1 decimals of accuracy
[D][sensor:092]: 'BME280 Relative Humidity': Sending state 40.99609 % with 1 decimals of accuracy
[D][sensor:092]: 'ESP Uptime': Sending state 24.87708 h with 0 decimals of accuracy
[D][bme280.sensor:197]: Got temperature=21.4°C pressure=995.8hPa humidity=44.7%
[D][sensor:092]: 'BME280 Temperature': Sending state 21.36000 °C with 1 decimals of accuracy
[D][sensor:092]: 'BME280 Pressure': Sending state 995.76263 hPa with 1 decimals of accuracy
[D][sensor:092]: 'BME280 Relative Humidity': Sending state 44.67383 % with 1 decimals of accuracy
[W][i2c:070]: Received NACK on transmit of address 0x76
[D][sensor:092]: 'ESP Uptime': Sending state 24.89375 h with 0 decimals of accuracy
[W][i2c:070]: Received NACK on transmit of address 0x76
[W][i2c:070]: Received NACK on transmit of address 0x76
[D][sensor:092]: 'ESP Uptime': Sending state 24.91042 h with 0 decimals of accuracy
[W][i2c:070]: Received NACK on transmit of address 0x76
[W][i2c:070]: Received NACK on transmit of address 0x76
[D][sensor:092]: 'ESP Uptime': Sending state 24.92709 h with 0 decimals of accuracy
[W][i2c:070]: Received NACK on transmit of address 0x76
```

**Additional information and things you've tried:**
Regrettably, I don't really know much about the low-level guts of the I2C bus, though there is seems to be discussions about this @ https://github.com/esp8266/Arduino/issues/1025 and for the BME280 @ https://github.com/finitespace/BME280/pull/78 

I may see if there is a way I can chain another ESP to act as a power reset to this one or something like that, so I could at least reliably remotely reset the I2C.

glmnet, 2020-05-15:
It’d be great if you can test another bme module and double check wires are good

Number 2 suggest there is a module side issue there.

deftdawg, 2020-05-16:
@glmnet I don't think it's an issue with the BME280 or the wiring as they are both seated on a breadboard (see pic).  I've replaced the sensor with a spare...    That's caused me to notice a different likely related issue with RH being returned as `nan`.  When I replaced the BME with new one the RH when to `nan` until I rebooted; I just checked on a different room w/ identical hardware and it has been been displaying `nan` for RH for some time now, in both case Temp and Pressure are still being reported. 

I think it most likely there's a problem with either the BME/ESPHome software stack or an issue with the ESP8266's board design.  The `nan` also makes an appearance with a DHT22 in #239 which points more towards ESPHome or the ESP8266. 

[image]

glmnet, 2020-05-16:
If it requieres a full power cycle  to recover It, that is the bmp crashed or went to sleep or something. 
Try cycling power to the bmp module alone and check if comms resume without restarting the esp. 
it should be easy on the bread board
I have a few bmp280 and they do ok besides Temp accuracy is awful. I believe there are many counterfeit of these.

deftdawg, 2020-07-10:
Following up on this, still an intermittent problem with my various ESP8266s running 1.14.5.  Pulling power from the sensor does get it to send readings again when the ESP rediscovers in... 

I'm currently using a statistics sensor to look for loss of variance, change, etc and notifying to go manually reset the device... 

```yaml
- platform: statistics
  entity_id: sensor.upstairs_hallway_temperature
  name: Hallway Temperature for past 2 hours
  sampling_size: 120
  max_age:
    minutes: 120
```
```jinja2
{{ state_attr('sensor.hallway_temperature_for_past_2_hours', 'count') > 20 
and state_attr('sensor.hallway_temperature_for_past_2_hours', 'variance') == 0 
and state_attr('sensor.hallway_temperature_for_past_2_hours', 'change') == 0 }}
```

I wonder if it would be possible to use a Digital pin set high to serve as VCC to the sensor.   That way either the ESP could power up the sensor to take a reading...  I feel like that would be a more reliable way to use these sensors, unless they need some time to warm up or something (in which case could leave the Digitial VCC pin high all the time except when sensor stops reporting new values and then drop it low to reset for a second).

closed 2020-12-05 by stale[bot]

pxFIN, 2021-08-15:
> @glmnet I don't think it's an issue with the BME280 or the wiring as they are both seated on a breadboard (see pic). I've replaced the sensor with a spare... That's caused me to notice a different [...]

Maybe this helps other people in similar situation and although this is an old issue i bumped into similar challenge.. BME280 not found from I2C bus after reset or NACK after some time of normal operation.

I noticed from the picture BME280 is 6 pin version(I2C + SPI bus supported) not 4 pin which means there are extra SDO and CSB pins available.. i cant see CSB connected to anything so that might be the issue (CSB Chip Select needs to be connected to voltage input if I2C is used, down is SPI is used).

deftdawg, 2021-08-15:
@pxFIN thanks for leaving a comment, its definitely something for me to look at. I didn't realize CSB was needed for anything! 😄

glmnet, 2021-08-15:
If this is the case it would be worth mentioning this on docs. Thanks

amishv, 2021-09-22:
The issue seems to be in  ESPHome. If I flash an Arduino based firmware the same setup starts reporting the correct temperature and pressure on BMP280, while I get signature error on ESPHome or it stops reporting temp & pressure after a few readings, if it gets the temp and pressure on power cycle.

pxFIN, 2021-09-22:
I've been playing recently more with bme280 + bme680 and from my experiences one thing to try in Esphome is lowering I2C bus frequency. Default is 50kHz.. Try 10kHz if it helps.. helped me

amishv, 2021-09-22:
Affter connecting CSB to VCC,  I am able to consistently  use 200KHz. One device is connected to the utility power directly and switches on or off with the utility power, still works flawlessly with CSB connected to VCC and speed = 200KHz. 
I did not experience such issue when using the ESP Arduino library.