esphome/issues #1222: Home Assistant Text Sensor break OTA flash

blademckain, 2020-05-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32

**Affected component:**
[Home Assistant Text Sensor](https://esphome.io/components/text_sensor/homeassistant.html?highlight=home%20assistant)

**Description of problem:**
after adding a Home Assistant Text Sensor to my yaml, sending the new firmware via OTA, the ESP32 stops reporting the error:

`
[09:42:45][D][ota:072]: Starting OTA Update from 192.168.12.241...
[09:42:45][V][ota:097]: OTA features is 0x00
[09:42:45][V][ota:171]: OTA size is 898672 bytes
[09:42:45][D][Updater.cpp:136] begin(): OTA Partition: app0
[09:42:45][V][ota:219]: Update: Binary MD5 is d8d8d1492959c64a0a4be59f42877455
[09:42:45][D][ota:243]: OTA in progress: 0.1%
[09:42:46][D][ota:243]: OTA in progress: 7.8%
[09:42:47][D][ota:243]: OTA in progress: 15.6%
[09:42:48][D][ota:243]: OTA in progress: 23.7%
[09:42:49][D][ota:243]: OTA in progress: 31.5%
[09:42:50][D][ota:243]: OTA in progress: 39.7%
[09:42:51][D][ota:243]: OTA in progress: 47.5%
[09:42:52][D][ota:243]: OTA in progress: 55.2%
[09:42:53][D][ota:243]: OTA in progress: 63.0%
[09:42:54][D][ota:243]: OTA in progress: 70.7%
[09:42:56][D][ota:243]: OTA in progress: 78.9%
[09:42:57][D][ota:243]: OTA in progress: 86.6%
[09:42:58][D][ota:243]: OTA in progress: 94.4%
[09:42:59][I][ota:267]: OTA update finished!
[09:42:59][I][app:137]: Rebooting safely...
[09:42:59][V][esp-idf:000]: I (444348) wifi: state: run -> init (0)
[09:42:59][V][esp-idf:000]: W (444352) wifi: hmac is stop, discard
`

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: f_test2
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: "myssd"
  password: "mypasw"
  power_save_mode: none
  # Optional manual IP
  manual_ip:
    static_ip: 192.168.12.9
    gateway: 192.168.12.254
    subnet: 255.255.255.0
    dns1: 8.8.8.8
    dns2: 8.8.4.4
  reboot_timeout: 1440min

# Enable logging
logger:
  level: VERBOSE
  
# Enable Home Assistant API
api:

ota:
  safe_mode: True

web_server:
  port: 80

time:
  - platform: homeassistant
    timezone: Europe/Rome
    id: homeassistant_time

text_sensor:
  - platform: homeassistant
    name: "HVAC mode"
    entity_id: input_select.hvac
    id: imp_HVAC_mode

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

glmnet, 2020-05-15:
Please review your title and description. 

 it’s confusing for me. Thanks

blademckain, 2020-05-17:
modified
okay so?

glmnet, 2020-05-17:
Ok now i get it. I’ll try to reproduce it

closed 2020-09-28 by stale[bot]