esphome/issues #1223: GPS Time is working now

deltazerorsan, 2020-05-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESPHome version 1.14.3 over Home Assistant 0.109.6

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP-WROOM-32 (various)
**Affected component:**

gps

**Description of problem:**
PROBLEM FIXED in #576 (#725)

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  devicename: test_esphome
  uper_devicename: test_esphome
  device_ip: 192.168.1.30

esphome:
  name: $devicename
  platform: ESP32
  board: esp-wrover-kit

wifi:
  networks:
    - ssid: !secret wifi_ssid
      password: !secret wifi_pass
    - ssid: !secret wifi2_ssid
      password: !secret wifi_pass
    - ssid: !secret wifi3_ssid
      password: !secret wifi_pass
  manual_ip:
    static_ip: $device_ip
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: !secret dns1
    dns2: !secret dns2

# Enable logging
logger: 
  level: INFO

# Enable Home Assistant API
api:
  password: !secret api_pass
  reboot_timeout: 0s

ota:
  password: !secret ota_pass
  
uart:
  rx_pin: 4
  baud_rate: 9600

# Declare GPS module
gps:

# GPS as time source
time:
  - platform: gps
    id: gps_time
    on_time:
      - seconds: '*'
        then:
          - lambda: 'ESP_LOGI("gps", "%02d:%02d:%02d", id(gps_time).now().hour, id(gps_time).now().minute, id(gps_time).now().second);'

```

**Logs (if applicable):**

```
20:45:56.555 -> [0;32m[I][app:100]: ESPHome version 1.14.3 compiled on May 15 2020, 20:44:44[0m
20:45:58.694 -> [0;32m[I][gps:239]: 20:45:59[0m
20:45:59.693 -> [0;32m[I][gps:239]: 20:46:00[0m
20:46:00.695 -> [0;32m[I][gps:239]: 20:46:01[0m
20:46:01.701 -> [0;32m[I][gps:239]: 20:46:02[0m
20:46:02.704 -> [0;32m[I][gps:239]: 20:46:03[0m
20:46:03.707 -> [0;32m[I][gps:239]: 20:46:04[0m
20:46:04.708 -> [0;32m[I][gps:239]: 20:46:05[0m
20:46:05.680 -> [0;32m[I][gps:239]: 20:46:06[0m
20:46:06.687 -> [0;32m[I][gps:239]: 20:46:07[0m
20:46:07.689 -> [0;32m[I][gps:239]: 20:46:08[0m
20:46:08.669 -> [0;32m[I][gps:239]: 20:46:09[0m
20:46:09.669 -> [0;32m[I][gps:239]: 20:46:10[0m
20:46:10.677 -> [0;32m[I][gps:239]: 20:46:11[0m
20:46:11.681 -> [0;32m[I][gps:239]: 20:46:12[0m
20:46:12.686 -> [0;32m[I][gps:239]: 20:46:13[0m
20:46:13.688 -> [0;32m[I][gps:239]: 20:46:14[0m

```

**Additional information and things you've tried:**

rradar, 2020-05-18:
> **Description of problem:**
> PROBLEM FIXED in #576 (#725)

Why open a issue then? :thinking:

deltazerorsan, 2020-05-18:
> > **Description of problem:**
> > PROBLEM FIXED in #576 (#725)
> 
> Why open a issue then? 🤔

Hi,

The doc related to the GPS component still has a Warning and ask for some help to test the component.

I've tested it with a ublock 6 and put here the yaml and the results.

In the issues #576 (#725) there is a mixup, a lack of info and no tests.

rradar, 2020-05-19:
> The doc related to the GPS component still has a Warning and ask for some help to test the component.

:+1: That's what is it about - now I understand! You could make a [pull request for the gps page](https://github.com/esphome/esphome-docs/blob/current/components/gps.rst) if you want. Just take the box out :wink:

Meanwhile I will search my sensor box to find a gps module to test and confirm your findings too :clock11:

closed 2020-10-07 by stale[bot]