esphome/issues #1225: [W][esp32_camera:151]: Got invalid frame from camera!

bereska, 2020-05-16:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

docker esphome on mac / docker homeassistant on rpi
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32-cam AI Thinker clone
**Affected component:**

**Description of problem:**
after successfully flashing esp32-cam I get this error:
[W][esp32_camera:151]: Got invalid frame from camera!
https://gist.github.com/bereska/3a004a409fb84f024ad9daeada465e49

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

rradar, 2020-05-18:
Your home assistant version is very old btw... 0.110 is around the corner :house: 

Try to detach the lens carefully from the board - maybe clean the contacts if you see something (humidity easily harms it) and reattach it again.

bereska, 2020-05-18:
@rradar thank you. yes, indeed, it's pretty old. Need to update. Too many gateways, platforms and sensors with automation, you know).
you are right. It's a hardware problem. The other esp32cam works just fine.
cheers

rradar, 2020-05-19:
@bereska perfect. Time to close the issue then :wink:

CarlosGS, 2020-09-04:
Hi @bereska, I'm having the same problem. Two brand new camera modules tested with the same esp32.

One camera works OK (left), the other does not (right) _"Got invalid frame from camera!"._

[image]

I only have two cameras so far, so I'm not sure if the differences that I'm observing are relevant:
- The camera that works OK has a _slightly_ larger lens and brown-ish reflection.
- The camera that doesn't work has a green-ish reflection.

Could you please confirm if you can observe these differences in your modules?

bereska, 2020-09-04:
@CarlosGS hi, it's all hardware related. I got the other one working by cleaning the connector pins with alcohol and inserting it tightly in the slot

closed 2021-01-17 by stale[bot]