esphome/issues #1228: High power consumption in deep sleep

jagjordi, 2020-05-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
pip

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32 doit devkit v1

**Affected component:**
https://esphome.io/components/deep_sleep.html

**Description of problem:**
The deep sleep component causes the power consumption to be very high, arround 20 mA, the datasheet claims arround 0.1 mA, and even taking into account the consumption of the UART chip and power LED etc. 20 mA is too high.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp32test
  platform: ESP32
  board: esp32doit-devkit-v1

wifi:
  ssid: ""
  password: ""

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32Test Fallback Hotspot"
    password: ""

binary_sensor:
  - platform: gpio
    pin:
      number: 32
      mode: INPUT_PULLUP
    id: pollo
    name: 'Living room: street window'
    device_class: window

deep_sleep:
  run_duration: 10s
  sleep_duration: 10min
  #wakeup_pin: 32

captive_portal:

# Enable logging
logger:
  level: NONE

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

```

**Logs (if applicable):**

```
[12:52:49]  8 2016 00:22:57
[12:52:49]
[12:52:49]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[12:52:50]configsip: 0, SPIWP:0xee
[12:52:50]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[12:52:50]mode:DIO, clock div:2
[12:52:50]load:0x3fff0018,len:4
[12:52:50]load:0x3fff001c,len:1044
[12:52:50]load:0x40078000,len:8896
[12:52:50]load:0x40080400,len:5828
[12:52:50]entry 0x400806ac
[12:52:50][I][logger:156]: Log initialized
[12:52:50][C][ota:364]: There have been 2 suspected unsuccessful boot attempts.
[12:52:50][I][app:028]: Running through setup()...
[12:52:50][D][binary_sensor:034]: 'Living room: street window': Sending initial state ON
[12:52:50][C][wifi:033]: Setting up WiFi...
[12:52:50][D][wifi:304]: Starting scan...
[12:52:52][D][wifi:319]: Found networks:
[12:52:52][I][wifi:365]: - 'MochiLAN' (5C:6A:80:57:7B:CC) ▂▄▆█
[12:52:52][D][wifi:366]:     Channel: 12
[12:52:52][D][wifi:367]:     RSSI: -49 dB
[12:52:52][D][wifi:369]: - 'Corona 2.4' (00:16:A6:21:2D:84) ▂▄▆█
[12:52:52][D][wifi:369]: - 'ASUS_repeter' (2C:FD:A1:2C:92:00) ▂▄▆█
[12:52:52][D][wifi:369]: - 'Hizli.ve.sifreli' (A4:91:B1:4D:CC:85) ▂▄▆█
[12:52:52][I][wifi:193]: WiFi Connecting to 'MochiLAN'...
[12:52:53][I][wifi:423]: WiFi Connected!
[12:52:53][C][wifi:283]:   SSID: 'MochiLAN'
[12:52:53][C][wifi:284]:   IP Address: 192.168.1.95
[12:52:53][C][wifi:286]:   BSSID: 5C:6A:80:57:7B:CC
[12:52:53][C][wifi:287]:   Hostname: 'esp32test'
[12:52:53][C][wifi:291]:   Signal strength: -47 dB ▂▄▆█
[12:52:53][C][wifi:295]:   Channel: 12
[12:52:53][C][wifi:296]:   Subnet: 255.255.255.0
[12:52:53][C][wifi:297]:   Gateway: 192.168.1.1
[12:52:53][C][wifi:298]:   DNS1: 192.168.1.1
[12:52:53][C][wifi:299]:   DNS2: 0.0.0.0
[12:52:53][D][wifi:432]: Disabling AP...
[12:52:53][C][ota:029]: Over-The-Air Updates:
[12:52:53][C][ota:030]:   Address: esp32test.local:3232
[12:52:53][C][ota:032]:   Using Password.
[12:52:53][W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 8 restarts
[12:52:53][C][api:022]: Setting up Home Assistant API server...
[12:52:53][C][deep_sleep:013]: Setting up Deep Sleep...
[12:52:53][I][app:058]: setup() finished successfully!
[12:52:53][I][app:100]: ESPHome version 1.14.3 compiled on May 17 2020, 12:48:55
[12:52:53][C][wifi:415]: WiFi:
[12:52:53][C][wifi:283]:   SSID: 'MochiLAN'
[12:52:53][C][wifi:284]:   IP Address: 192.168.1.95
[12:52:53][C][wifi:286]:   BSSID: 5C:6A:80:57:7B:CC
[12:52:53][C][wifi:287]:   Hostname: 'esp32test'
[12:52:53][C][wifi:291]:   Signal strength: -47 dB ▂▄▆█
[12:52:53][C][wifi:295]:   Channel: 12
[12:52:53][C][wifi:296]:   Subnet: 255.255.255.0
[12:52:53][C][wifi:297]:   Gateway: 192.168.1.1
[12:52:53][C][wifi:298]:   DNS1: 192.168.1.1
[12:52:53][C][wifi:299]:   DNS2: 0.0.0.0
[12:52:53][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Living room: street window'
[12:52:53][C][gpio.binary_sensor:015]:   Device Class: 'window'
[12:52:53][C][gpio.binary_sensor:016]:   Pin: GPIO32 (Mode: INPUT_PULLUP)
[12:52:53][C][logger:175]: Logger:
[12:52:53][C][logger:176]:   Level: DEBUG
[12:52:53][C][logger:177]:   Log Baud Rate: 115200
[12:52:53][C][logger:178]:   Hardware UART: UART0
[12:52:53][C][captive_portal:169]: Captive Portal:
[12:52:53][C][ota:029]: Over-The-Air Updates:
[12:52:53][C][ota:030]:   Address: esp32test.local:3232
[12:52:53][C][ota:032]:   Using Password.
[12:52:53][W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 8 restarts
[12:52:53][C][api:095]: API Server:
[12:52:53][C][api:096]:   Address: esp32test.local:6053
[12:52:53][C][deep_sleep:020]: Setting up Deep Sleep...
[12:52:54][C][deep_sleep:023]:   Sleep Duration: 600000 ms
[12:52:54][C][deep_sleep:026]:   Run Duration: 10000 ms
[12:52:58][D][api.connection:583]: Client 'Home Assistant 0.105.5 (192.168.1.237)' connected successfully!
[12:53:03][I][deep_sleep:067]: Beginning Deep Sleep

```

jagjordi, 2020-05-17:
#634 asked a similar question but was never resolved

rradar, 2020-05-18:
> The deep sleep component causes the power consumption to be very high, arround 20 mA, the datasheet claims arround 0.1 mA

Which datasheet exactly? The one for your board or for the soc? Do you mind to post a link?

Most of them are show false values anyway. Many boards are badly designed and even if it's well done the power consumption typically even varies a lot depending if you power it with usb or via the onboard battery for example.

> 20 mA is too high.

It would be also interesting **how you measure** the consumption. For low currents you need special equipment to get valid results.

A [interesting video](https://www.youtube.com/watch?v=-769_YIeGmI) about some esp32 boards and current draws in different modes/power sources, a [excel sheet with the results](https://docs.google.com/spreadsheets/d/1Mu-bNwpnkiNUiM7f2dx8-gPnIAFMibsC2hMlWhIHbPQ/edit#gid=0) is also available.

And from your linked comment their is a [interesting option related to the wake up mode](https://github.com/esphome/issues/issues/634#issuecomment-526764541) from a user you could try meanwhile and report back.

rradar, 2020-05-19:
And also very interesting: https://olimex.wordpress.com/2020/05/18/new-open-source-hardware-board-esp32-s2-devkit-lipo-breaks-our-internal-record-for-esp32-low-power-consumption/

> The new ESP32-S2-DevKit-Lipo uses ultra low power power supply circuit which makes **current consumption during sleep only 6uA**  (4uA of them are due to the battery measurement resistor divider)

rzolau, 2020-07-14:
Same issue here: the esp32 dev board from Ezsbc (known for very low current draw in deep sleep).

Test 1:

Simple code implemented directly in platformio: bootup, go into deep sleep, wake on GPIO2 -- measured current draw 12 uA.

Test 2:

The same board with the following esphome config:

```
esphome:
  name: XXX
  platform: ESP32
  board: esp32dev

wifi:
  ssid: "XXX"
  password: "XXX"
  fast_connect: true
  manual_ip:
    static_ip: XXX
    gateway: XXX
    subnet: XXX

ota:
  password: "XXX"

mqtt:
  broker: XXX
  discovery: false
  topic_prefix: XXX

deep_sleep:
  id: deep_sleep_entry
  run_duration: 30s
  sleep_duration: 120s
```

Everything works as expected, we enter the deep sleep and the measured current draw is again 12 uA.

Test 3:

We make the following change to the deep sleep entry in Test 2 (this is the only change we make):

```
deep_sleep:
  id: deep_sleep_entry
  run_duration: 30s
  wakeup_pin:
    number: 2
    mode: INPUT
    inverted: false
```

Now when we enter the deep sleep, the measured current draw is 1.3 mA (100x more then should be).
Disabling WiFi, MQTT, OTA changes nothing. The only way to get to 12 uA is not to use wakeup_pin.

It seems that in this case something is not powered down correctly, not able to trace what...

closed 2020-11-25 by stale[bot]