esphome/issues #1229: bang bang climate loses values ​​sent by home assistant

blademckain, 2020-05-17:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32

**Affected component:**
[Bang bang climate controller
](https://esphome.io/components/climate/bang_bang.html)

**Description of problem:**
The default_target_temperature_low and default_target_temperature_high settings, modified through the home assistant's lovelace gui are lost when the esp32 is restarted

**Problem-relevant YAML-configuration entries:**
```yaml
climate:
  - platform: bang_bang
    sensor: dht22_t
    name: my_bang_climate
    id: my_bang_climate
    visual:
      min_temperature: 12 °C
      max_temperature: 35 °C
      temperature_step: 0.1 °C
    default_target_temperature_low: 20 °C
    default_target_temperature_high: 22 °C
    away_config:
      default_target_temperature_low: 16 °C
      default_target_temperature_high: 17 °C
    heat_action:
      - logger.log: "Heat ON"
      - switch.turn_on: OU7  
    idle_action:
      - logger.log: "Idle ON"
      - switch.turn_off: OU7

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
Each time the esp32 is restarted, the bang bang start with the configurations taken in the .yaml file and loses any changes made through the home assistant gui

blademckain, 2020-05-22:
only I find this problem with esp32?
I found [another topic with the same problem](https://github.com/esphome/issues/issues/588) but used esp8266 and I think it solved using esp8266_restore_from_flash
But this config I don't think works on esp32

blademckain, 2020-05-25:
I set the log to VERY_VERBOSE
After changing the default_target_temperature_high from the homeassistant gui interface,
 in the logs of esphome I read
`
[09:10:26][VV][preferences:051]: SAVE 9: 0=0x00000001 1=0x666641A0 (Type=2577382936, CRC=0x928D8065)
`

But if I restart the esp32 this restarts with the default_target_temperature_high set in the .yaml

blademckain, 2020-05-26:
more deep into the problem
I have inserted some printf

The problems seems that the bang bang controller correctly take values ​​from memory with function:
`
// restore set points
  auto restore = this->restore_state_();
  if (restore.has_value()) {
	    ESP_LOGVV("main", "RESTORE FROM MEMORY");
    restore->to_call(this).perform();
`
but immediately afterwards is called the function:

`void BangBangClimate::control(const climate::ClimateCall &call) {
  if (call.get_mode().has_value())
    this->mode = *call.get_mode();
  if (call.get_target_temperature_low().has_value())
    this->target_temperature_low = *call.get_target_temperature_low();
  if (call.get_target_temperature_high().has_value())
    this->target_temperature_high = *call.get_target_temperature_high();
  if (call.get_away().has_value())
    this->change_away_(*call.get_away());

  this->compute_state_();
   ESP_LOGVV("main", "BANG BANG control");
  this->publish_state();
}`

which restores the values ​​from the yaml file

Entire log:
`
[09:31:53][VV][preferences:038]: LOAD 10: valid=YES, 0=0x999A0001 1=0x00004151 (Type=2577382936, CRC=0xD1606898)
[09:31:53][VV][main:152]: TEMP LOW:13.100000
[09:31:53][VV][main:153]: TEMP HIGH:22.000000
[09:31:53][VV][main:022]: RESTORE FROM MEMORY
[09:31:53][VV][main:235]: CLIMATE TO_CALL
[09:31:53][VV][main:240]: TEMP LOW:13.100000
[09:31:53][VV][main:241]: TEMP HIGH:22.000000
[09:31:53][D][climate:010]: 'my_bang_climate' - Setting
[09:31:53][D][climate:014]:   Mode: AUTO
[09:31:53][D][climate:020]:   Target Temperature Low: 13.10
[09:31:53][D][climate:023]:   Target Temperature High: 22.00
[09:31:53][D][climate:026]:   Away Mode: OFF
[09:31:53][VV][main:042]: BANG BANG control
[09:31:53][D][climate:180]: 'my_bang_climate' - Sending state:
[09:31:53][D][climate:183]:   Mode: AUTO
[09:31:53][D][climate:185]:   Action: OFF
[09:31:53][D][climate:188]:   Current Temperature: nan°C
[09:31:53][D][climate:192]:   Target Temperature: Low: 20.00°C High: 22.00°C
[09:31:53][D][climate:197]:   Away: OFF
[09:31:53][VV][main:158]: CALL to SAVE_STATE
[09:31:53][VV][main:166]: TEMP LOW:20.000000
[09:31:53][VV][main:167]: TEMP HIGH:22.000000
[09:31:53][VV][preferences:051]: SAVE 10: 0=0x00000001 1=0x000041A0 (Type=2577382936, CRC=0xC4CBD08C)
`

blademckain, 2020-05-26:
Ok the problem is in the function change_away_
that ALWAYS takes the values ​​from the conf and not from the memory!!!

`
void BangBangClimate::change_away_(bool away) {
  if (!away) {
    this->target_temperature_low = this->normal_config_.default_temperature_low;
    this->target_temperature_high = this->normal_config_.default_temperature_high;
  } else {
    this->target_temperature_low = this->away_config_.default_temperature_low;
    this->target_temperature_high = this->away_config_.default_temperature_high;
  }
  this->away = away;
}
`

kbx81, 2020-05-26:
@blademckain could you test this branch and confirm if it is fixed for you? most of the control code was already replaced in this PR and I just worked on some updates to the `setup()` function (which I believe is responsible for this task at start-up). I will try to test on my setup later today, as well.

blademckain, 2020-05-31:
yes, your branch fix the issue

closed 2020-06-17 by glmnet