esphome/issues #1232: http_request post always sends the same url

ricreis394, 2020-05-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker
**ESP (ESP32/ESP8266, Board/Sonoff):**

```yaml
platform: ESP8266
board: esp01_1m
```
**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
http_request:
  useragent: esphome/device

binary_sensor:
  - platform: gpio
    id: switch_1
    pin:
      number: GPIO12
      mode: INPUT_PULLUP
      inverted: True
    on_state:
      - http_request.post:
          url: http://10.10.130.174/light/livingroom_led_sanca/toggle
      
  - platform: gpio
    id: switch_2
    pin:
      number: GPIO14
      mode: INPUT_PULLUP
      inverted: True
    on_state:
      - http_request.post:
          url: http://10.10.130.175/light/suite_led_sanca/toggle
```

**Logs (if applicable):**

```
[21:59:03][D][binary_sensor:036]: 'switch_1': Sending state OFF
[21:59:03][D][http_request:060]: HTTP Request completed; URL: http://10.10.130.174/light/livingroom_led_sanca/toggle; Code: 200
[21:59:05][D][binary_sensor:036]: 'switch_1': Sending state ON
[21:59:05][D][http_request:060]: HTTP Request completed; URL: http://10.10.130.174/light/livingroom_led_sanca/toggle; Code: 200
[21:59:06][D][binary_sensor:036]: 'switch_2': Sending state ON
[21:59:06][D][http_request:060]: HTTP Request completed; URL: http://10.10.130.175/light/suite_led_sanca/toggle; Code: 200
[21:59:08][D][binary_sensor:036]: 'switch_2': Sending state OFF
[21:59:08][D][http_request:060]: HTTP Request completed; URL: http://10.10.130.175/light/suite_led_sanca/toggle; Code: 200
[21:59:08][D][binary_sensor:036]: 'switch_1': Sending state OFF
[21:59:08][D][http_request:060]: HTTP Request completed; URL: http://10.10.130.174/light/livingroom_led_sanca/toggle; Code: 200
[21:59:10][D][binary_sensor:036]: 'switch_2': Sending state ON
[21:59:10][D][http_request:060]: HTTP Request completed; URL: http://10.10.130.175/light/suite_led_sanca/toggle; Code: 200

```

**Additional information and things you've tried:**

Logs are correct but in fact the post command isn't sending to the correspondig url.
In this case, the binary sensor always toggles the same light which is the **suite_led_sanca**

redlegoman, 2020-06-03:
This happens to me too. It seems to use the url composed the first time it is called then never update

```
http_request:
  useragent: esphome/device
  timeout: 10s

# Individual sensors
sensor:
  - platform: dallas
    force_update: true
    index: 0
    name: "Temperature Roshin"
    id: temp1
    on_value:
      then:
        - http_request.get:
            url: !lambda |-
              std::string payload;
              payload.append("http://heating/temps/update.php?key=test1&tempc=");
              char pubtemp[10];
              dtostrf(x,2,1,pubtemp);
              payload.append(pubtemp);
              return payload;
```

And the log in esphome looks like it is calling the URL with a different temperature value each time:

```
[15:45:27][D][dallas.sensor:177]: 'Temperature Roshin': Got Temperature=27.2°C
[15:45:27][D][http_request:060]: HTTP Request completed; URL: http://heating/temps/update.php?key=test1&tempc=27.2; Code: 200
[15:45:27][V][sensor:013]: 'Temperature Roshin': Received new state 27.250000
[15:45:27][D][sensor:092]: 'Temperature Roshin': Sending state 27.25000 °C with 1 decimals of accuracy
[15:46:27][D][dallas.sensor:177]: 'Temperature Roshin': Got Temperature=32.7°C
[15:46:27][D][http_request:060]: HTTP Request completed; URL: http://heating/temps/update.php?key=test1&tempc=32.7; Code: 200
[15:46:27][V][sensor:013]: 'Temperature Roshin': Received new state 32.687500
[15:46:27][D][sensor:092]: 'Temperature Roshin': Sending state 32.68750 °C with 1 decimals of accuracy
```

But here are the logs from Apache, showing the URL being called with the same temperature everytime:

```
10.10.10.220 - - [03/Jun/2020:15:44:27 +0100] "GET /temps/update.php?key=test1&tempc=26.9 HTTP/1.1" 200 246 "-" "esphome/device"
10.10.10.220 - - [03/Jun/2020:15:45:27 +0100] "GET /temps/update.php?key=test1&tempc=26.9 HTTP/1.1" 200 246 "-" "esphome/device"
10.10.10.220 - - [03/Jun/2020:15:46:27 +0100] "GET /temps/update.php?key=test1&tempc=26.9 HTTP/1.1" 200 246 "-" "esphome/device"

```

marianomd, 2020-07-09:
Same here. Subsequent requests go all to the first URL.

glmnet, 2020-07-11:
CC @Anonym-tsk

closed 2020-12-03 by jesserockz