esphome/issues #1234: xQueueCreateCountingSemaphore with esp32_ble_tracker

marcoCasamento, 2020-05-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** PIP (same issue on HASS.IO)

**ESP (ESP32/ESP8266, Board/Sonoff):** ESP32 esp32dev
Board bought from AZDelivery, [datasheet](https://cdn.shopify.com/s/files/1/1509/1638/files/ESP_-_32_NodeMCU_Developmentboard_Datenblatt_a3bf98d8-6a53-4d26-8f1c-c61b1c82af39.pdf?76837)

**Affected component:** [espr32_ble_tracker](https://esphome.io/components/esp32_ble_tracker.html)

**Description of problem:**
when the esp32_ble_tracker is enabled, the esp32 reset very often, apparently when doing a bluetooth scan. It does scan succesful a couple of time (random, can't find a pattern), then it crashes

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: lucigiorno
  platform: ESP32
  board: esp32dev

wifi:
  ssid: "MyNetwork"
  password: "My Secure Pwd"
  # fast_connect: true
  manual_ip:
    static_ip: 192.168.1.33
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 192.168.1.205
  ap:
    ssid: "My Fallback Hotspot"
    password: "My other password"

captive_portal:

logger:
  level: DEBUG
api:
  password: "and password"

ota:
  password: "and password"

esp32_ble_tracker:

```

**Logs (if applicable):**

```
[10:09:11][D][xiaomi_ble:164]:   Humidity: 58.8%
[10:09:25]/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:706 (xQueueCreateCountingSemaphore)- assert failed!
[10:09:37]abort() was called at PC 0x4008e0c7 on core 0
[10:09:37]
[10:09:37]Backtrace: 0x40092094:0x3ffe8530 0x400922c5:0x3ffe8550 0x4008e0c7:0x3ffe8570 0x40119e8d:0x3ffe8590 0x40119eda:0x3ffe85b0 0x400dbe00:0x3ffe85d0 0x4008453e:0x3ffe8610 0x400f9ba1:0x3ffe8660 0x4013f1b1:0x3ffe86b0 0x40144229:0x3ffe86f0 0x40149512:0x3ffe8710 0x4013299b:0x3ffe8730 0x4008e7dd:0x3ffe8760
[33mWARNING Found stack trace! Trying to decode it[0m
[10:09:37]
[10:09:37]Rebooting...
[10:09:37]ets Jun  8 2016 00:22:57
[10:09:37]
[10:09:37]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[10:09:37]configsip: 0, SPIWP:0xee
[10:09:37]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[10:09:37]mode:DIO, clock div:2
[10:09:37]load:0x3fff0018,len:4
[10:09:37]load:0x3fff001c,len:1044
[10:09:37]load:0x40078000,len:8896
[10:09:37]load:0x40080400,len:5828
[10:09:37]entry 0x400806ac
[10:09:37][I][logger:156]: Log initialized

```

**Additional information and things you've tried:**

I'm not sure about the right board to use, since I can't find the AZDelivery one within those listed by esphome, so I choose esp32dev, I've however tried other, including nodemcu-32s

sachadr, 2020-05-20:
Hi @marcoCasamento, thanks for posting this. I'm having the same issue with boards: nodemcu-32s & mhetesp32devkit . I am certain about my boards. Using esphome version 1.14.3.

marcoCasamento, 2020-05-20:
ok, @sachadr that exclude some issues on my side about board selection.
Just to exclude any hardware issue, I've just flashed another AZDelivery board exactly the same as the other one, brand new out of his box, but it restarts as well.

I'll try to repurpose an ESP32 from DOIT (or some clone, i'm not sure, it came from AliExpress) just to test with another board

sachadr, 2020-05-20:
Alright, I think I solved it (at least for me)...

I started running into this problem when I decided to set up a new wifi access point specifically for my smart home devices. Reason was to keep the normal wifi less congested.

I've been testing the different wifi modes which my old router supports (11b, 11g & 11n). I started with "n", then "g" and finally "b". For the last few hours since using "b", the problem didn't come back. I thought let me change back to "n" again to make sure it is the case. Immediately it started producing the error and rebooting often.

So, for me while using the esp32_ble_tracker component, seems that I can make the device happy by ensuring that my wifi (which I use only for my IOT devices) is making use of 11b wifi mode. I tested this on the nodemcu-32s board. I actually want to have this working on another board, a mhetesp32devkit, so I'll test how it works and share the result later (likely in the next few days).

@marcoCasamento , if you can configure different wifi modes, try using 11b and see if this helps you too.

marcoCasamento, 2020-05-20:
Thanks for sharing.
Apparently I can't limit a SSID to  certain radio modes, not anymore at least, but I'll look for support at ubiquiti.
In the meantime I've flashed a esp32doit-devkit-v1 and it's way more stable, but still crashes once in an hour or so, same error on xQueueCreateCountingSemaphore
I'm wondering if it possible to change the radio mode from the client perspective, giving that 802.11b is way enough for my needs

sachadr, 2020-05-20:
So I found a moment to get it running on the other board. Running perfectly, no reboots after a couple hours, and sending through data for my xiaomi_lywsdcgq device.

marcoCasamento, 2020-05-22:
Hi @sachadr tinkering with the .cpp files I've managed to force the client to choose protocol "B", and both the boards (the az-delivery one and the doit one) are running for about three hours without a single crash. 

Thanks for your input, the problem is solved, despite it requires to hack with the .cpp driver. 
Maybe I can file a PR if the author think it is interesting AND if would ever be able to put one together.

Just for sharing what I did, I've added

`esp_wifi_set_protocol(WIFI_IF_STA, WIFI_PROTOCOL_11B);`

in esphome\components\wifi\wifi_component_esp32.cpp at line 170, just before of calling 

`esp_wifi_get_config(WIFI_IF_STA, &current_conf);`

So, we can say that at least some boards doesn't support wifi ng AND ble_tracker.

sachadr, 2020-05-22:
Most welcome @marcoCasamento. Glad you managed to find a way via the code. Thanks for sharing the solution as well, I'm impressed that you are able to write such code. I would be very curious as to whether they would merge it. Please drop me a msg with a link to the MR, if you do, so I can follow.

Zixim, 2020-05-31:
@marcoCasamento 
I'm having trouble applying your patch, every time I make the edit, the file gets reset.
Is this the correct path for a Hassio install ?
`\config\esphome\<devicename>\src\esphome\components\wifi\`

buxtronix, 2020-06-22:
I think forcing 11B is simply hiding the real problem. I have been digging into this lately - it seems that the Esphome log code itself might be problematic as various bug reports (eg issue #837) show it crashing in the same place, which is when assigning the static object FORMAT_COPY in core/log.cpp.
This is dubious already, as multiple FreeRTOS threads may be trying to log, however this static variable is not thread-safe.

The fact that it is always triggered from MDNS code is pretty odd also.

I've been experimenting with replacing this with a malloc()ed variable instead, and result are quite promising. If it's consistently fixed, I may submit a PR.

marcoCasamento, 2020-06-22:
@buxtronix 

> This is dubious already, as multiple FreeRTOS threads may be trying to log, however this static variable is not thread-safe.

This really makes sense. When I was testing it, I clearly remember that adding other components (that MCU commands two relays for a curtain and 6 other lights) greatly raised the odds of crashing, and honestly, it was probably my impression, that it did so while logging the result of the BLE scan or the readings from BLE sensors.

I've to add that I've not been able to make more than 24hr on a row of uptime, it randomly keep crashing, and I've recently removed the esp32_ble_tracker from that node.
I have another spare board at hands, if you believe I can help in testing or so, just drop a line

reopening

buxtronix, 2020-06-22:
@marcoCasamento 

I submitted a PR, feel free to try it out yourself, I'm yet to have a crash today since implementing, and heap memory looks pretty stable:

https://github.com/esphome/esphome/pull/1098

glmnet, 2020-06-23:
see also: https://github.com/espressif/arduino-esp32/issues/3332

buxtronix, 2020-06-24:
I spent a bit of time today looking further into this, and think I have the cause narrowed a lot. It's likely the same as #837.

The crash seems almost always happen in the MDNS code:

WARNING Decoded 0x400f5eed: _udp_recv at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mdns/mdns_networking.c:123 (discriminator 2)

I decided to tcpdump MDNS traffic on my lan at the time of the crash, and each time it happened, there was a significant burst of requests coming from the ESPHome service within my HomeAssistant server (sorry I dont have a capture). In my case I have 17 ESPHome devices configured, plus a bunch of other MDNS devices (eg smart speakers, etc).

I captured and replayed the packets, and could fairly reliably (70% of the time) trigger a crash each time i replayed. At times it didnt crash I noticed a significant drop in free Heap memory (i log it a couple of times per second).

So my theory here:

- The ESP32 is very memory constrained when ESPHome has BLE running. Typically only 40kb free normally.
- The ESPHome server bursts out MDNS discovery periodically (all at once, no smearing).
- All these requests (plus from other devices) hit the ESP32 at around the same time.
- If they all hit with the right conditions (# packets + other RAM events in the device) the device runs out of memory.
- The UDP receive code fails memory allocation, and calls a function to log this:

https://github.com/espressif/esp-idf/blob/master/components/mdns/mdns_networking.c#L121

- This ends up in the ESPHome logger, which crashes and fails to alloc memory when logging from the IDF:

https://github.com/esphome/esphome/blob/dev/esphome/core/log.cpp#L60

- This code seems to only be called from the IDF logging, Esphome uses a different path without the static alloc. This log
call is typically the first time it's been called during runtime, so the FORMAT_COPY static string has not yet been allocated,
so it gets allocated.

- Trying to allocate memory when logging a memory exhaustion problem = Boom!

The solutions for this would be one/more of:

- Run fewer ESPHome devices to minimise requests from the server.
- Smear MDNS requests from the ESPHome server. Other clients may still burst though.
- Reduce memory allocation elsewhere in the ESP32 code, but BT+Wifi stacks seems to always consume the most memory.
- Not do anything that will attempt to alloc memory within the logging code (or at least fail gracefully).

I raised esphome/esphome#1098 which addresses the last, but the other changes need consideration also.

VaticanUK, 2020-06-29:
see also: https://github.com/espressif/arduino-esp32/issues/2162

Also suggests it's a memory issue

closed 2020-11-10 by stale[bot]