esphome/issues #1235: SSD1306 OLED Display brightness doesn't work?

freekode, 2020-05-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

wemos d1 mini

**Affected component:**

https://next.esphome.io/components/display/ssd1306.html

**Description of problem:**
I'm trying to control brightness for the display, it doesn't change. Tried two displays, just with different screen size.
Neither brightness from display configuration nor my template switch don't work.
Tried to call `setup()` in template switch, no changes.
Overall I would like to switch off display to reduce power consumption, if there is another way todo it?

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: test_unit
  platform: ESP8266
  board: nodemcuv2

i2c:

font:
  - file: 'Arial.ttf'
    id: font1
    size: 14

display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x32"
    brightness: 10%
    id: screen
    lambda: |-
      it.print(0, 0, id(font1), "Hello World!");

switch:
  - platform: template
    name: "Backlights"
    assumed_state: true
    optimistic: true
    turn_on_action:
      then:
        - lambda: |-
            id(screen).set_brightness(1);
            id(screen).fill(COLOR_ON);
    turn_off_action:
      then:
        - lambda: |-
            id(screen).set_brightness(0);
            id(screen).fill(COLOR_OFF);
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

bestlong, 2020-05-25:
https://esphome.io/components/display/ssd1306.html

kbx81, 2020-06-05:
Pretty sure I see what the problem is. I've got one of these displays on order. I can probably fix it sometime next week.

closed 2020-08-23 by glmnet