esphome/issues #1237: ESPHOME disconnects from Home assistant constantly

jazzmonger, 2020-05-22:
Im having the exact same issue here. At first I thought it was my new AP I installed yesterday, but all my other 2.4ghz devices are staying connected for hours. I've tried all different 802.11 standards to see if it was WiFi protocol related, but it's not (2.4ghz a/b/g/n all do the same thing).

my d1_mini_pro and nodemcuv2 connect fine and then about every 3 minutes they both disconnect and then immediately reconnect from the AP. I tried the loop "fix" @thedishking recommended to no avail.

This all seems very recent. It was rock solid until a week or so ago.
I was on the beta channel until this AM.
I'm now on 0.110.1

Update Available 🎉
Operating System 3.13
You are currently running version 4.6

I'm going to OS 3.13 to see if that helps.
UPDATE: This did NOT help.  Going back to .107 as suggested

Very frustrating!!

Jeff

An example log disconnecting
```
[08:46:20][C][ota:029]: Over-The-Air Updates:
[08:46:20][C][ota:030]:   Address: 192.168.1.145:8266
[08:46:20][C][api:095]: API Server:
[08:46:20][C][api:096]:   Address: 192.168.1.145:6053
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to 192.168.1.145:6053 (192.168.1.145)
WARNING Couldn't connect to API (Error connecting to 192.168.1.145: timed out). Trying to reconnect in 1 seconds
INFO Connecting to 192.168.1.145:6053 (192.168.1.145)
INFO Successfully connected to 192.168.1.145
[08:46:55][D][api.connection:602]: Client 'Home Assistant 0.110.1 (192.168.1.250)' connected successfully!
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to 192.168.1.145:6053 (192.168.1.145)
WARNING Couldn't connect to API (Error connecting to 192.168.1.145: timed out). Trying to reconnect in 1 seconds
INFO Connecting to 192.168.1.145:6053 (192.168.1.145)
INFO Successfully connected to 192.168.1.145
[08:47:18][D][api.connection:602]: Client 'Home Assistant 0.110.1 (192.168.1.250)' connected successfully!
[08:47:19][D][api:067]: Disconnecting Home Assistant 0.110.1 (192.168.1.250)
[08:47:24][D][api:067]: Disconnecting Home Assistant 0.110.1 (192.168.1.250)
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to 192.168.1.145:6053 (192.168.1.145)
WARNING Couldn't connect to API (Error connecting to 192.168.1.145: timed out). Trying to reconnect in 1 seconds
INFO Connecting to 192.168.1.145:6053 (192.168.1.145)
WARNING Couldn't connect to API (Error connecting to 192.168.1.145: timed out). Trying to reconnect in 1 seconds
INFO Connecting to 192.168.1.145:6053 (192.168.1.145)
WARNING Couldn't connect to API (Error connecting to 192.168.1.145: [Errno 113] No route to host). Trying to reconnect in 2 seconds
INFO Connecting to 192.168.1.145:6053 (192.168.1.145)
WARNING Couldn't connect to API (Error connecting to 192.168.1.145: [Errno 113] No route to host). Trying to reconnect in 3 seconds
INFO Connecting to 192.168.1.145:6053 (192.168.1.145)
INFO Successfully connected to 192.168.1.145
[08:48:18][D][api.connection:602]: Client 'Home Assistant 0.110.1 (192.168.1.250)' connected successfully!
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to 192.168.1.145:6053 (192.168.1.145)
WARNING Couldn't connect to API (Error connecting to 192.168.1.145: timed out). Trying to reconnect in 1 seconds
INFO Connecting to 192.168.1.145:6053 (192.168.1.145)
WARNING Couldn't connect to API (Error connecting to 192.168.1.145: timed out). Trying to reconnect in 1 seconds
INFO Connecting to 192.168.1.145:6053 (192.168.1.145)
WARNING Couldn't connect to API (Error connecting to 192.168.1.145: [Errno 113] No route to host). Trying to reconnect in 2 seconds
INFO Connecting to 192.168.1.145:6053 (192.168.1.145)
WARNING Couldn't connect to API (Error connecting to 192.168.1.145: [Errno 113] No route to host). Trying to reconnect in 3 seconds
INFO Connecting to 192.168.1.145:6053 (192.168.1.145)
WARNING Couldn't connect to API (Error connecting to 192.168.1.145: [Errno 113] No route to host). Trying to reconnect in 5 seconds
INFO Connecting to 192.168.1.145:6053 (192.168.1.145)
INFO Successfully connected to 192.168.1.145
[08:49:25][D][api.connection:602]: Client 'Home Assistant 0.110.1 (192.168.1.250)' connected successfully!
[08:49:32][D][api:067]: Disconnecting Home Assistant 0.110.1 (192.168.1.250)
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to 192.168.1.145:6053 (192.168.1.145)
WARNING Couldn't connect to API (Error connecting to 192.168.1.145: timed out). Trying to reconnect in 1 seconds
INFO Connecting to 192.168.1.145:6053 (192.168.1.145)
INFO Successfully connected to 192.168.1.145
[08:50:10][D][api:067]: Disconnecting Home Assistant 0.110.1 (192.168.1.250)
[08:51:07][D][sensor:092]: 'Coop ESP8266 Upensor': Sending state 1.27334 hours with 0 decimals of accuracy
```

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
HASSIO Home Assistant 0.110.1

arch | armv7l
-- | --
chassis | embedded
dev | false
docker | true
hassio | true
host_os | HassOS 3.13
installation_type | Home Assistant
os_name | Linux
os_version | 4.19.114-v7
python_version | 3.7.7
supervisor | 222
timezone | America/Los_Angeles
version | 0.110.1
virtualenv | false

**ESP (ESP32/ESP8266, Board/Sonoff):**
NODEMCU, D1 Mini PRO

**Affected component:**

**Description of problem:**
Disconnections every 3 minutess or so

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: gatecontroller
  platform: ESP8266
  board: d1_mini_pro

  on_loop:
    then:
    - lambda: delay(5);
wifi:
  networks:
  - ssid:  "xx"
    password: "xxx"
  - ssid: "x"
    password: "xxx"

#  ssid: "xx Wifi"
#  password: "xxx"

  power_save_mode: none
  # ...
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.1.146
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 192.168.1.250
    dns2: 8.8.8.8
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Gate ESP Fallback hotspot"
    password: "Wine4You"

captive_portal:
# Enable Home Assistant API
ota:
# Enable logging
logger:
# Enable Home Assistant API
api:

#GPIO00 = D3
#GPIO01 = XX
#GPIO02 = D4
#GPIO04 = D2
#GPIO05 = D1
#GPIO12 = D6
#GPIO13 = D7
#GPIO14 = D5
#GPIO16 = D0
#-----------------------------
switch:
  - platform: gpio
    pin: 5 #GPIO5, D1, pin 2
    inverted: no
    id: relay
    restore_mode: ALWAYS_OFF
  - platform: template
    name: "Gate Remote"
    icon: "mdi:gate"
    turn_on_action:
    - switch.turn_on: relay
    - delay: 1000ms
    - switch.turn_off: relay

#-----------------------------    
    
binary_sensor:
  - platform: gpio
    name: Gate Open SW
    pin:
      number: 0 #GPIO0, D3
      inverted: True
      mode: INPUT_PULLUP

sensor:
  - platform: uptime
    name: Gate Controller Uptime Sensor
    filters:
      - lambda: return x / 3600.0;
    unit_of_measurement: "hours"
    accuracy_decimals: 2
    update_interval: 600s #10 min

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

changing WiFi to 802.11a, b, g, n individually has no effect as does changing to a new Ubiquity Loco M2 AP.  All other 2.4ghz devices work just fine and stay connected for hours.

jazzmonger, 2020-05-22:
More info.  Watching 2 device logs indicates that the ESP's disconnect and reconnect at the same times.  This would indeed imply that that the problem is on the HOME ASSISTANT side, not the ESP side or in the Wifi chain:

COOP NODE MCU ESP
```
[11:16:36][I][app:101]: ESPHome version 1.15.0-dev compiled on Apr 13 2020, 08:28:28
[11:16:36][C][wifi:415]: WiFi:
[11:16:36][C][wifi:283]:   SSID: 'xx'[redacted]
[11:16:36][C][wifi:284]:   IP Address: 192.168.1.145
[11:16:36][C][wifi:286]:   BSSID: 00:0E:3B:81:01:04[redacted]
[11:16:36][C][wifi:287]:   Hostname: 'coop'
[11:16:36][C][wifi:291]:   Signal strength: -61 dB ▂▄▆█
[11:16:36][C][wifi:295]:   Channel: 8
[11:16:36][C][wifi:296]:   Subnet: 255.255.255.0
[11:16:36][C][wifi:297]:   Gateway: 192.168.1.1
[11:16:36][C][wifi:298]:   DNS1: 192.168.1.250
[11:16:36][C][wifi:299]:   DNS2: (IP unset)
[11:16:36][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Coop Bottom Button'
[11:16:36][C][gpio.binary_sensor:016]:   Pin: GPIO0 (Mode: INPUT_PULLUP, INVERTED)
[11:16:36][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Coop Top Button'
[11:16:36][C][gpio.binary_sensor:016]:   Pin: GPIO2 (Mode: INPUT_PULLUP, INVERTED)
[11:16:36][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Coop Outside Reed SW'
[11:16:36][C][gpio.binary_sensor:015]:   Device Class: 'door'
[11:16:36][C][gpio.binary_sensor:016]:   Pin: GPIO4 (Mode: INPUT_PULLUP)
[11:16:36][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Coop Inside Reed SW'
[11:16:36][C][gpio.binary_sensor:015]:   Device Class: 'door'
[11:16:36][C][gpio.binary_sensor:016]:   Pin: GPIO5 (Mode: INPUT_PULLUP)
[11:16:36][C][output.switch:009]: Output Switch 'Coop Door Inside'
[11:16:36][C][output.switch:009]: Output Switch 'Coop Door Outside'
[11:16:36](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[11:16:36](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[11:16:36](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[11:16:36][C][ota:029]: Over-The-Air Updates:
[11:16:36](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[11:16:36](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[11:16:36](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to 192.168.1.145:6053 (192.168.1.145)
WARNING Couldn't connect to API (Error connecting to 192.168.1.145: timed out). Trying to reconnect in 1 seconds
INFO Connecting to 192.168.1.145:6053 (192.168.1.145)
INFO Successfully connected to 192.168.1.145
[11:17:03][D][api.connection:602]: Client 'Home Assistant 0.110.1 (192.168.1.250)' connected successfully!
[11:17:06][D][api:067]: Disconnecting Home Assistant 0.110.1 (192.168.1.250)
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to 192.168.1.145:6053 (192.168.1.145)
WARNING Couldn't connect to API (Error connecting to 192.168.1.145: timed out). Trying to reconnect in 1 seconds
INFO Connecting to 192.168.1.145:6053 (192.168.1.145)
INFO Successfully connected to 192.168.1.145
[11:17:39][D][api.connection:602]: Client 'Home Assistant 0.110.1 (192.168.1.250)' connected successfully!
[11:17:50][D][api:067]: Disconnecting Home Assistant 0.110.1 (192.168.1.250)
[11:17:56][D][sensor:092]: 'Coop ESP8266 Upensor': Sending state 0.99762 hours with 0 decimals of accuracy
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to 192.168.1.145:6053 (192.168.1.145)
WARNING Couldn't connect to API (Error connecting to 192.168.1.145: timed out). Trying to reconnect in 1 seconds
INFO Connecting to 192.168.1.145:6053 (192.168.1.145)
WARNING Couldn't connect to API (Error connecting to 192.168.1.145: timed out). Trying to reconnect in 1 seconds
INFO Connecting to 192.168.1.145:6053 (192.168.1.145)
WARNING Couldn't connect to API (Error connecting to 192.168.1.145: timed out). Trying to reconnect in 2 seconds
INFO Connecting to 192.168.1.145:6053 (192.168.1.145)
WARNING Couldn't connect to API (Error connecting to 192.168.1.145: [Errno 113] No route to host). Trying to reconnect in 3 seconds
INFO Connecting to 192.168.1.145:6053 (192.168.1.145)
INFO Successfully connected to 192.168.1.145
[11:18:54][D][api:067]: Disconnecting Home Assistant 0.110.1 (192.168.1.250)
```

GATE D1 Mini ESP
```
INFO Successfully connected to 192.168.1.146
[11:16:34][I][app:101]: ESPHome version 1.15.0-dev compiled on May 22 2020, 10:28:20
[11:16:34][C][wifi:415]: WiFi:
[11:16:34][C][wifi:283]:   SSID: 'xx'[redacted]
[... 33 lines cut ...]
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to 192.168.1.146:6053 (192.168.1.146)
WARNING Couldn't connect to API (Error connecting to 192.168.1.146: timed out). Trying to reconnect in 1 seconds
INFO Connecting to 192.168.1.146:6053 (192.168.1.146)
INFO Successfully connected to 192.168.1.146
[11:17:06][D][api:067]: Disconnecting Home Assistant 0.110.1 (192.168.1.250)
[11:17:10][D][api:067]: Disconnecting Home Assistant 0.110.1 (192.168.1.250)
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to 192.168.1.146:6053 (192.168.1.146)
WARNING Couldn't connect to API (Error connecting to 192.168.1.146: timed out). Trying to reconnect in 1 seconds
[... cut ...]
WARNING Couldn't connect to API (Error connecting to 192.168.1.146: timed out). Trying to reconnect in 1 seconds
INFO Connecting to 192.168.1.146:6053 (192.168.1.146)
WARNING Couldn't connect to API (Error connecting to 192.168.1.146: timed out). Trying to reconnect in 2 seconds
INFO Connecting to 192.168.1.146:6053 (192.168.1.146)
INFO Successfully connected to 192.168.1.146
[11:18:53][D][api:067]: Disconnecting Home Assistant 0.110.1 (192.168.1.250)
[11:19:02][D][api.connection:599]: Client 'Home Assistant 0.110.1 (192.168.1.250)' connected successfully!
```

jazzmonger, 2020-05-22:
More info from the home assistant log file:

```
2020-05-22 11:31:02 INFO (MainThread) [homeassistant.components.esphome] Can't connect to ESPHome API for 192.168.1.155: Error connecting to ('192.168.1.155', 6053): [Errno 113] Connect call failed ('192.168.1.155', 6053)
2020-05-22 11:31:02 INFO (MainThread) [homeassistant.components.esphome] Trying to reconnect in 60 seconds
2020-05-22 11:31:35 WARNING (MainThread) [homeassistant.util.async_] Detected I/O inside the event loop. This is causing stability issues. Please report issue to the custom component author for sonoff doing I/O at custom_components/sonoff/__init__.py, line 374: ), headers=self._headers)
2020-05-22 11:31:37 WARNING (SyncWorker_12) [amcrest.http] <None:None> Trying again due to error: ConnectionError(MaxRetryError("HTTPConnectionPool(host='192.168.1.87', port=80): Max retries exceeded with url: /cgi-bin/magicBox.cgi?action=getMachineName (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at ADDRESS>: Failed to establish a new connection: [Errno 113] Host is unreachable'))"))
2020-05-22 11:31:40 INFO (MainThread) [aioesphomeapi.connection] 192.168.1.145: Ping Failed!
2020-05-22 11:31:40 INFO (MainThread) [homeassistant.components.esphome] Disconnected from ESPHome API for 192.168.1.145
2020-05-22 11:31:40 INFO (MainThread) [aioesphomeapi.connection] 192.168.1.145: Error while reading incoming messages: Error while receiving data: 0 bytes read on a total of 1 expected bytes
2020-05-22 11:31:53 INFO (MainThread) [aioesphomeapi.connection] 192.168.1.146: Ping Failed!
2020-05-22 11:31:54 INFO (MainThread) [homeassistant.components.esphome] Disconnected from ESPHome API for 192.168.1.146
2020-05-22 11:31:54 INFO (MainThread) [aioesphomeapi.connection] 192.168.1.146: Error while reading incoming messages: Error while receiving data: 0 bytes read on a total of 1 expected bytes
2020-05-22 11:31:54 ERROR (MainThread) [aioharmony.hubconnector] 192.168.1.122: Connection timed out for hub 10785974
2020-05-22 11:31:56 INFO (MainThread) [homeassistant.components.esphome] Successfully connected to 192.168.1.145
2020-05-22 11:31:57 INFO (MainThread) [homeassistant.components.esphome] Successfully connected to 192.168.1.146
2020-05-22 11:32:05 INFO (MainThread) [homeassistant.components.esphome] Can't connect to ESPHome API for 192.168.1.155: Error connecting to ('192.168.1.155', 6053): [Errno 113] Connect call failed ('192.168.1.155', 6053)
2020-05-22 11:32:05 INFO (MainThread) [homeassistant.components.esphome] Trying to reconnect in 60 seconds
2020-05-22 11:32:08 ERROR (MainThread) [homeassistant.components.ipp] Error fetching ipp data: Invalid response from API: Error occurred while communicating with IPP server.
2020-05-22 11:32:08 WARNING (MainThread) [homeassistant.config_entries] Config entry for ipp not ready yet. Retrying in 80 seconds.
2020-05-22 11:32:12 ERROR (MainThread) [aioharmony.hubconnector] 192.168.1.103: Exception on post: Cannot connect to host 192.168.1.103:8088 ssl:None [Connect call failed ('192.168.1.103', 8088)]
2020-05-22 11:32:12 ERROR (MainThread) [aioharmony.hubconnector] 192.168.1.103: Unable to retrieve HUB id
2020-05-22 11:32:12 WARNING (MainThread) [homeassistant.components.harmony.remote] Master Bedroom: Unable to connect to HUB.
2020-05-22 11:32:12 WARNING (MainThread) [homeassistant.config_entries] Config entry for harmony not ready yet. Retrying in 80 seconds.
```

jazzmonger, 2020-05-22:
OK - I just had a major discovery.  On a hunch from looking at 
```[aioesphomeapi.connection] 192.168.1.145: Ping Failed!``` 
I set up a ping to each ESP from my Mac.  Each of them stops responding at exactly the same time.  They also disconnect from my access point at pretty much exactly the same time.  so, this would point to the ESPs themselves, NOT HOME ASSISTANT !!!

Firemogle, 2020-05-22:
`INFO Reading configuration /config/esphome/kitchen_leak.yaml...
WARNING The esphomeyaml section has been renamed to esphome in 1.11.0. Please replace 'esphomeyaml:' in your configuration with 'esphome:'.
INFO Starting log output from kitchen_leak.local using esphome API
WARNING Exception occurred:
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/esphome/zeroconf.py", line 486, in handle_read
    data, (addr, port) = socket_.recvfrom(_MAX_MSG_ABSOLUTE)
OSError: [Errno 9] Bad file descriptor
INFO Connecting to kitchen_leak.local:6053 (192.168.2.113)
INFO Successfully connected to kitchen_leak.local
[15:13:30][I][app:101]: ESPHome version 1.15.0-dev compiled on May 19 2020, 12:56:51
[15:13:30][C][wifi:415]: WiFi:
[15:13:30][C][wifi:283]:   SSID: [redacted]
[15:13:30][C][wifi:284]:   IP Address: 192.168.2.113
[15:13:30][C][wifi:286]:   BSSID: [redacted]
[15:13:30][C][wifi:287]:   Hostname: 'kitchen_leak'
[15:13:30][C][wifi:291]:   Signal strength: -76 dB ▂▄▆█
[15:13:30][C][wifi:295]:   Channel: 11
[15:13:30][C][wifi:296]:   Subnet: 255.255.255.0
[15:13:30][C][wifi:297]:   Gateway: 192.168.2.1
[15:13:30][C][wifi:298]:   DNS1: 192.168.2.1
[15:13:30][C][wifi:299]:   DNS2: (IP unset)
[15:13:30][C][template.binary_sensor:018]: Template Binary Sensor 'kitchen_leak'
[15:13:30][C][template.binary_sensor:018]:   Device Class: 'moisture'
[15:13:30][C][logger:175]: Logger:
[15:13:30][C][logger:176]:   Level: DEBUG
[15:13:30][C][logger:177]:   Log Baud Rate: 115200
[15:13:30](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[15:13:30](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[15:13:30](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[15:13:30](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[15:13:30](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[15:13:30](Message skipped because it was too big to fit in TCP buffer - This is only cosmetic)
[15:13:30][C][adc:051]:   Update Interval: 60.0s
[15:13:30][C][ota:029]: Over-The-Air Updates:
[15:13:30][C][ota:030]:   Address: kitchen_leak.local:8266
[15:13:30][C][ota:032]:   Using Password.
[15:13:30][C][api:095]: API Server:
[15:13:30][C][api:096]:   Address: kitchen_leak.local:6053
[15:13:50][D][adc:056]: 'kitchen_leak_raw': Got voltage=0.00V
[15:13:51][D][sensor:092]: 'kitchen_leak_raw': Sending state 0.00645 V with 2 decimals of accuracy
[15:14:51][D][adc:056]: 'kitchen_leak_raw': Got voltage=0.00V
[15:14:51][D][sensor:092]: 'kitchen_leak_raw': Sending state 0.00645 V with 2 decimals of accuracy
[15:15:50][D][adc:056]: 'kitchen_leak_raw': Got voltage=0.00V
[15:15:50][D][sensor:092]: 'kitchen_leak_raw': Sending state 0.00645 V with 2 decimals of accuracy
[15:16:50][D][adc:056]: 'kitchen_leak_raw': Got voltage=0.00V
[15:16:52][D][sensor:092]: 'kitchen_leak_raw': Sending state 0.00967 V with 2 decimals of accuracy
[15:17:50][D][adc:056]: 'kitchen_leak_raw': Got voltage=0.00V
[15:17:50][D][sensor:092]: 'kitchen_leak_raw': Sending state 0.00967 V with 2 decimals of accuracy
WARNING Disconnected from API: Timeout while waiting for message response!
INFO Connecting to kitchen_leak.local:6053 (192.168.2.113)
WARNING Couldn't connect to API (Error connecting to 192.168.2.113: timed out). Trying to reconnect in 1 seconds
INFO Connecting to kitchen_leak.local:6053 (192.168.2.113)
INFO Successfully connected to kitchen_leak.local
[15:19:07][D][api.connection:599]: Client 'Home Assistant 0.109.6 (192.168.2.2)' connected successfully!
[15:19:50][D][adc:056]: 'kitchen_leak_raw': Got voltage=0.00V
[15:19:51][D][sensor:092]: 'kitchen_leak_raw': Sending state 0.00967 V with 2 decimals of accuracy
[15:20:51][D][adc:056]: 'kitchen_leak_raw': Got voltage=0.00V
[15:20:53][D][sensor:092]: 'kitchen_leak_raw': Sending state 0.00967 V with 2 decimals of accuracy`

`INFO Reading configuration /config/esphome/mini_aquarium.yaml...
INFO Starting log output from 192.168.2.112 using esphome API
INFO Connecting to 192.168.2.112:6053 (192.168.2.112)
INFO Successfully connected to 192.168.2.112
[15:14:41][I][app:101]: ESPHome version 1.15.0-dev compiled on May 18 2020, 12:48:48
[15:14:41][C][wifi:415]: WiFi:
[15:14:41][C][wifi:283]:   SSID: [redacted]
[15:14:41][C][wifi:284]:   IP Address: 192.168.2.112
[15:14:41][C][wifi:286]:   BSSID: [redacted]
[15:14:41][C][wifi:287]:   Hostname: 'mini_aquarium'
[... 181 lines cut ...]
[15:18:06][D][sensor:092]: 'MiniTank Pressure': Sending state 984.15594 P with 1 decimals of accuracy
[15:18:06][D][sensor:092]: 'MiniTank Humidity': Sending state 36.20703 % with 1 decimals of accuracy
[15:18:13][D][adc:056]: 'bed_prob_v': Got voltage=1.50V
[15:18:13][D][sensor:092]: 'bed_prob_v': Sending state 1.50286 V with 2 decimals of accuracy
[15:18:13][D][resistance:037]: 'bed_probe_r' - Resistance 8362.5Ω
[15:18:13][D][sensor:092]: 'bed_probe_r': Sending state 8362.48145 Ω with 1 decimals of accuracy
[15:18:13][D][ntc:026]: 'bed_probe_c' - Temperature: 29.0°C
[15:18:13][D][sensor:092]: 'bed_probe_c': Sending state 28.97750 °C with 1 decimals of accuracy
ERROR Error while reading incoming messages: Error while receiving data: [Errno 104] Connection reset by peer
WARNING Disconnected from API: Error while receiving data: [Errno 104] Connection reset by peer
INFO Connecting to 192.168.2.112:6053 (192.168.2.112)
[... cut ...]
[15:19:06][D][bme280.sensor:197]: Got temperature=31.8°C pressure=984.2hPa humidity=36.3%`

Adding a small log of my disconnect, but the narrative on this issue so far sounds similar to mine, this is the output from 2 devices, one with only one sensor and one with several IO and they disconnect around the same time.

I noticed this on all the esp devices shortly after I ran several upgrades, router (pfsense) was moved to a new machine, home assistant to a new VM and updated.  I have lots of variables but found all of my esps are doing similar as described.

n8hjp, 2020-05-23:
Here is my addition to the list.   I have 5 wemos D1s that are dropping in and out.     I hooked one up the console on one and am getting this error:

jazzmonger, 2020-05-23:
SOLVED (I think***)

Ok, this is an esoteric 802.11 wifi related issue. I'm not sure if it's at the esp chip level, the firmware chip driver for the WiFi chip on the esp or the esphome code our brilliant young programmer created for these cool little esp8266 based chips.

Here's the workaround:

On the Access Point that all your esp's are connecting to (or all your AP's)  change the channel's bandwidth to be FIXED at 20mhz, not AUTO.  Restart you AP. I use channel 8, but I have not verified that other channels work or don't work yet.  I will do more testing to try and isolate the issue further after I post this. 

The esps connection algorithm apparently uses 40mhz as its default connection scheme and this is where the problem lies. 

I've tried this on a nodemcu AND D1 mini and it "fixes" the problem.  Unfortunately some APs have buggy 802.11 standards implemented in them (cheap Chinese crap often comes with brain dead programmers trying to implement complex standards) and try as I might, the $500 Hawking HOW17ACM I just bought is absolute garbage. The esps as well as other devices like a Mac and an iPad also have problems on this AP.

My trusty Ubiquity Loco M2's are rock solid. But, the esps need the 20mhz channel bandwidth setting to stay connected.  Otherwise they cycle connect/disconnect.

If some of you can try this fix and report back here, this might give the ESPHOME devs some clues about where to look. There is some documented weirdness on these devices when they are underclocked, so this might be a place to look:

https://hackaday.com/2019/01/04/underclocking-the-esp8266-leads-to-wifi-weirdness/

Jeff

Firemogle, 2020-05-27:
I wanted to update that I recently started having these issues and the wifi fix did not work for me.

MaximumSU, 2020-05-29:
> On the Access Point that all your esp's are connecting to (or all your AP's) change the channel's bandwidth to be FIXED at 20mhz, not AUTO.

I not confirm this also. I tried to fix the WiFi channel width to 20MHz - no difference. I have 3 types of routers (4 pieces total). All routers are dual band (2.4GHz, and 5GHz), 3 routers of 2 types (2 kinds of Zyxel) connected as routed MESH network and last one of another type (TPLINK) stay alone. Also i have 2 types of boards with ESP8266 and ESP8285 chips. 
1. I use for almost all devices the 1.13.6 firmware with Arduino dev. For this firmware, if it connected to the one router - no disconnects at all. But, if same device connected occasenly to another router within the MESH network (to another type of router) - it starting disconnect constantly. 
2. For the one device i use 1.15 dev  and Arduino dev. This device connected to the 3-rd type of router outside the MESH network, and it also no disconnects at all. BUT - if same device connected to the one of ANY routers from MESH network - it start's disconnecting constantly. If this device is flashed by 1.13.6 firmware - the same as punkt 1.
So, i can't find any pattern in such behavior. But it is not random also. If it not works reliable in some combination of firmware+router - it never works, if some combination works - it always works.

daffster, 2020-06-03:
I've applied the 20MHz channel change, and am still having these issues across multiple devices.
Wemos D1 Mini, ESP-12E bare, NodeMCU and bulbs based on ESP8285 are all disconnecting.

This only seems to impact devices that utilise esp8266_pwm though, and only while those outputs are NOT set to 100% (255).

laca75tn, 2020-06-05:
I am having the same issues.  Node MCUs using esp8266_pwm.  I have three blinds with servos.  They have been working perfectly for months, and in the last week or two they started having erratic behavior.  WiFi signal strength is around 60, as I said they were working flawlessly for months and now all three are disconnecting several times each hour.

mikalauskas, 2020-06-08:
it's not related wifi issue on router side for sure, i have the same problem but can still ping esp, also my wifi is 20mhz on default. and esphome ... logs monitoring disconnects but still can ping. 

my dirty workaround is to restart esp every day early morning

```
time:
  - platform: sntp
    id: sntp_time
    timezone: Asia/Yekaterinburg
    on_time:
      - seconds: 0
        minutes: 30
        hours: 4
        then:
          - switch.turn_on: restart1

switch:
  - platform: restart
    name: "rf_bridge restart"
    id: restart1
    internal: true
```

back0nline, 2020-07-22:
I'm also having this problem for months.

At first i taught that was related with the Aruba AP Firmware but after 3 updates nothing has changed.
What i know is that all my esp loses the connection with the AP, and if i restart the AP all esp are able to reconnect again.
It's nothing related with the AP because all my other devices are working fine only my esphome devices are not able work.

On the AP I forced all my connected devices to reconnect after 24h and it did not solve the issue.

daffster, 2020-08-09:
My initial comment states that this only impacted devices with esp8266_pwm, but I was wrong.

[garage-esphome.log](https://github.com/esphome/issues/files/5048555/garage-esphome.log)
I've attached a log of a one of my esp8266 wemos d1 mini's that shows the full startup, a crash and a reboot.

CarlosGS, 2020-10-28:
[image]

I observe instantaneous re-connections (see gray strips), **exclusively when enabling** `power_save_mode: light`.

Also note #1532 power save light/high may be swapped at the moment. I will test the "modem" mode and write back.

CarlosGS, 2020-10-28:
In any case, would it be possible to have some tolerance against re-connections? A couple of seconds of allowance seem more reasonable rather than logging every instantaneous drop.

glenn20, 2021-01-27:
After belatedly finding the sollution from @jazzmonger (set wifi chanel width to 20MHz  - not 40MHz), I confirm this instantly fixed my connection issues with all my ESP32 esphome devices. Before the change, all my esphome sensors toggled to "unavailable" every minute or two for a few seconds up to a minute or so. This has plagued and frustrated me for months.

My access point is a TP-Link Archer C2600 running openwrt 19.04.

The difference is easily diagnosed with "ping":

wifi channel width set to 40MHz:

PING downstairs.lan (192.168.40.10): 56 data bytes
64 bytes from 192.168.40.10: seq=3 ttl=255 time=4212.492 ms
64 bytes from 192.168.40.10: seq=4 ttl=255 time=3214.421 ms
64 bytes from 192.168.40.10: seq=5 ttl=255 time=2215.980 ms
64 bytes from 192.168.40.10: seq=6 ttl=255 time=1217.056 ms
64 bytes from 192.168.40.10: seq=7 ttl=255 time=216.737 ms
64 bytes from 192.168.40.10: seq=11 ttl=255 time=4178.138 ms
64 bytes from 192.168.40.10: seq=12 ttl=255 time=3177.930 ms
64 bytes from 192.168.40.10: seq=13 ttl=255 time=2179.113 ms
64 bytes from 192.168.40.10: seq=14 ttl=255 time=1179.397 ms
64 bytes from 192.168.40.10: seq=15 ttl=255 time=179.243 ms
64 bytes from 192.168.40.10: seq=17 ttl=255 time=4570.686 ms
64 bytes from 192.168.40.10: seq=18 ttl=255 time=3571.966 ms
64 bytes from 192.168.40.10: seq=19 ttl=255 time=2572.417 ms
64 bytes from 192.168.40.10: seq=20 ttl=255 time=1573.015 ms
64 bytes from 192.168.40.10: seq=21 ttl=255 time=574.027 ms

wifi channel set to 20MHz:

PING downstairs.lan (192.168.40.10): 56 data bytes
64 bytes from 192.168.40.10: seq=0 ttl=255 time=194.041 ms
64 bytes from 192.168.40.10: seq=1 ttl=255 time=218.359 ms
64 bytes from 192.168.40.10: seq=2 ttl=255 time=36.877 ms
64 bytes from 192.168.40.10: seq=3 ttl=255 time=59.691 ms
64 bytes from 192.168.40.10: seq=4 ttl=255 time=83.883 ms
64 bytes from 192.168.40.10: seq=5 ttl=255 time=108.578 ms
64 bytes from 192.168.40.10: seq=6 ttl=255 time=128.894 ms
64 bytes from 192.168.40.10: seq=7 ttl=255 time=153.089 ms
64 bytes from 192.168.40.10: seq=8 ttl=255 time=174.360 ms
64 bytes from 192.168.40.10: seq=9 ttl=255 time=197.648 ms
64 bytes from 192.168.40.10: seq=10 ttl=255 time=16.564 ms
64 bytes from 192.168.40.10: seq=11 ttl=255 time=40.342 ms
64 bytes from 192.168.40.10: seq=12 ttl=255 time=65.580 ms
64 bytes from 192.168.40.10: seq=13 ttl=255 time=88.499 ms
64 bytes from 192.168.40.10: seq=14 ttl=255 time=118.784 ms
64 bytes from 192.168.40.10: seq=15 ttl=255 time=138.178 ms
64 bytes from 192.168.40.10: seq=16 ttl=255 time=156.239 ms
64 bytes from 192.168.40.10: seq=17 ttl=255 time=240.215 ms
64 bytes from 192.168.40.10: seq=18 ttl=255 time=202.772 ms

glenn20, 2021-01-27:
> After belatedly finding the sollution from @jazzmonger (set wifi chanel width to 20MHz - not 40MHz), I confirm this instantly fixed my connection issues with all my ESP32 esphome devices. Before the [...]

Ok - I got ahead of myself. Turns out it is not changing the channel width to 20MHz, it seems to just be restarting the interface which resolves the issue - for a time - but it returns after a few minutes up to over 10 minutes ( I just got a good stretch of 10 mins of perfection when I first tried it). So it goes...

closed 2021-06-11 by stale[bot]

chemelli74, 2021-06-22:
Seems the "solution" provided is just a workaround.
Any better and reliable approach ?

Simone

XavitoHA19, 2021-06-23:
On your router make sure that you are using 20MHz width and a fixed channel.  Upgrade to the latest version of EspHome.  That fixed all my issues

chemelli74, 2021-06-23:
> On your router make sure that you are using 20MHz width and a fixed channel. Upgrade to the latest version of EspHome. That fixed all my issues

I got it that it fixed issues for you, but is more a workaround than a real solution ;-)

back0nline, 2021-06-27:
I dont have the 20mhz enable and i dont have any problems for months but I don't know what solved it

glenn20, 2021-09-28:
Hopefully this will be useful to people. After over a year of continuous disconnects/reconnects and false dawns with various work-arounds I have had no problems with disconnects for over 8 months now.

I found the combination of two changes fixed this for me:

1. Set `arduino_version: latest` under the `esphome:` section.
2. Set `power_save_mode: none` under the `wifi:` section.

I also set a static IP address, but I don't believe that has any effect.

It may well be that the first fix is no longer necessary, but I haven't backed it out yet to test that.

Good luck.

chrispontiga, 2022-01-23:
@glenn20 Thank you.  You saved my life.  Was almost done installing some monochromatic lights under my cabinets and in the process of buttoning things up.  Then noticed they were not responding properly to HA controls timely, and pings were super slow and all over the map.  They were just working earlier.  Tore through the project box again and troubleshooted all the soldered connections.  Was ready to just build another one when I found your suggestion.  Wasted so much time with that stupid power save mode.  I had updated to the latest ESPHome release 2022.1.1 which enabled this setting by default, just before installing under my cabinets.  Once I fixed that setting, it was all good and working.  Wasted an extra three hours because of this stupid setting.

theres, 2022-01-24:
For me only one thing helped - change of AP. In my case it was change to UniFi, but I've heard a lot good about Mikrotik as well.
Before: restart every 5-20min
After: uptime 2 weeks (since I've change the AP)

CarlosGS, 2022-01-24:
> Wasted so much time with that stupid power save mode. I had updated to the latest ESPHome release 2022.1.1 which enabled this setting by default

@chrispontiga The defaults have not been changed, though. They are "modem sleep" for ESP32 and "none" for ESP8266.

Don't blame power_save_mode. Note in ESP8266 [it is simply mislabeled!](https://github.com/esphome/issues/issues/1532) People using `power_save_mode: light` are in fact using the highest setting which throttles the CPU and causes problems. Afterwards, they don't bother trying the "high" setting.
But in fact the best setting to use in ESP8266 is `power_save_mode: high`, see https://github.com/esphome/issues/issues/1532#issuecomment-718669014 for details & measurements. That setting matches the ESP32 default (modem sleep) which really reduces heating.

If you also set `arduino_version: latest`, then that's what fixed the problem and is a good clue to find the cause :smile:
- First, please confirm you are using ESP8266 and weren't manually setting `power_save_mode: light` previous to the changes, as that would have been the problem.
- Then, when you have the time: Could you please re-compile with only the `arduino_version` change, and share the compile logs with and without the option? This way we could know the exact Arduino versions and check for differences in their repository. Thanks!

chrispontiga, 2022-01-24:
@CarlosGS Currently on an ESP32.  The only thing that was added to resolve the connectivity/responsiveness issue was power `power_save_mode: none.`  I did not add the `arduino_version` line.  This is why I blame the power_save_mode.  

It worked fine before the update to 2022.1.1.  It was hardly working after the update.

atv2016, 2022-04-27:
I tried to set arduino_version but if i set it under esp_home it says move it to the board specific section. When i add it there it doesn't seem to like it either.

Does anyone have any code examples with latest ESPhome where to put this statement?

ADeadPixel, 2022-06-24:
> I tried to set arduino_version but if i set it under esp_home it says move it to the board specific section. When i add it there it doesn't seem to like it either.
> 
> Does anyone have any code examples with latest ESPhome where to put this statement?

I'm experiencing the same issue. I cannot set the Ardunio_Version in ESPHome 2022.6.2. I'm attempting the WiFi power mode change right now to see if this fixes my devices that keep disconnecting.

MikeMakes, 2022-07-09:
I found the same problem and I observed that the esp8266 nodemcu v1 would reconnect by his own on certain terms. In my case it would disconnect after 5min exactly and never come back unless;
-Reset the esp
-Reset the AP
-Some action occurred in the esp (for example, a toggle in a gpio switch scripted at some after a delay)
Because of the later I supossed it was some kind of power saving mode so I include power_save_mode: none and it got fixed. Looking for more info (to see if this was intentional) I ended up here. I will try to change the AP config and itself to see if it changes anything without changing the power save mode.