esphome/issues #1240: Climate Fujitsu_general don't work for me

wisesokol, 2020-05-23:
Home Assistent with hassio 107.6 
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Tuya s08 (Avatto)
**ESP (ESP32/ESP8266, Board/Sonoff):**

TYWE3S
**Affected component:**

https://next.esphome.io/components/climate/ir_climate.html

**Description of problem:**
Unfortunately, my Fujitsu air conditioner (Model ASY7RSCCW) with the AR-DB5 remote does not respond to the existing component. Can you help me?

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: ir
  platform: ESP8266
  board: esp_wroom_02

wifi:
  ssid: "1"
  password: "1"

logger:

api:
  password: "1"

ota:
  password: "1"

status_led:
  pin: GPIO4

remote_receiver:
  pin: 
    number: GPIO5
    inverted: true
  dump: all

remote_transmitter:
  pin: 
    number: GPIO14
    inverted: false
  carrier_duty_percent: 50%
  id: "s08_transmitter"

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO13
      inverted: true
    internal: true
    name: "s08_button"
    id: "s08_button"

climate:
  - platform: fujitsu_general
    name: "s06_erisson_ac"
    transmitter_id: "s08_transmitter"
    id: "s08_erisson_ac"
    visual:
      min_temperature: 16 °C
      max_temperature: 31 °C
      temperature_step: 1 °C
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

glmnet, 2020-05-24:
are you sure your hardware is ok? i.e. IR led brightness ok? You can check e.g. controlling another IR stuff

wisesokol, 2020-05-24:
Hardware is ok. My tv LG work fine with esphome software. Leds blinks, when send command

wisesokol, 2020-05-30:
I found that if I send a command SWING
`  - platform: template
    name: "AC"
    turn_on_action:
      remote_transmitter.transmit_panasonic:
        address: 0x28C6  
        command: 0x000808B6`
it is work fine.
But other command don't work. 
`climate:
  - platform: fujitsu_general`
don't work.
I found a description of the commands on the arduino website
`    SET: 
    0x28C6000808 36

    SWING:
    0x28C6000808 B6

    TEMPERATURE: (27-26-25 °C)
    0x28C6000808 3F **100C0D4 000000000 78**
    0x28C6000808 3F 100C054 000000000 74
    0x28C6000808 3F 100C094 000000000 7C

    MASTER (Here it is in 'Fan' mode):
    0x28C6000808 3F 100C05C 000000000 B4`
0x28C6- adress
000808 B6 - command
Whith work fine.
In logs i see only 
`[remote.panasonic:068]: Received Panasonic: address=0x28C6, command=0x0008083F`- turn on on original remote
[remote.panasonic:068]: Received Panasonic: address=0x28C6, command=0x00080840 - turn off
i don't see other code such as     0x28C6000808 3F **100C0D4 000000000 78**

wisesokol, 2020-05-31:
On Tasmota firmware in logs i see next:
```
18:46:57 MQT: tele/tasmota_0788F4/RESULT = {"IrReceived":{"Protocol":"FUJITSU_AC","Bits":120,"Data":"0x0x1463001010FC08307101040000005A","Repeat":0,"IRHVAC":{"Vendor":"FUJITSU_AC","Model":2,"Power":"On","Mode":"Cool","Celsius":"On","Temp":23,"FanSpeed":"Min","SwingV":"Off","SwingH":"Off","Quiet":"On","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
18:47:00 MQT: tele/tasmota_0788F4/RESULT = {"IrReceived":{"Protocol":"FUJITSU_AC","Bits":48,"Data":"0x0x146300101002","Repeat":0,"IRHVAC":{"Vendor":"FUJITSU_AC","Model":2,"Power":"Off","Mode":"Off","Celsius":"On","Temp":16,"FanSpeed":"Auto","SwingV":"Off","SwingH":"Off","Quiet":"Off","Turbo":"Off","Econo":"Off","Light":"Off","Filter":"Off","Clean":"Off","Beep":"Off","Sleep":-1}}}
```
First line- Power on on remote
Second- Power OFF

I do not like Tasmota firmware, I want to use ESPhome. Help.

closed 2020-12-05 by stale[bot]

werdnum, 2021-03-11:
I'm also experiencing this issue. I'm using a Lolin Wemos D1 Mini, hooked up to a TSOP4138 IR receiver module.

I set 'inverted' on the pin because esphome told me to ("starts high"), and I get similar codes to above.

Here's the config I used to get raw codes (swap "raw" for "all" to get the config I used to get "Panasonic" codes:

```
esphome:
  name: ac_remote_1
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "AC Remote Fallback Hotspot"
    password: !secret wifi_password

captive_portal:

# Enable logging
logger:

mqtt:
  broker: 192.168.124.200
  port: 32323
  will_message:
  birth_message:

ota:
  password: !secret ota_password

remote_transmitter:
  pin: D0
  carrier_duty_percent: 50%

remote_receiver:
  pin:
    number: D1
    inverted: true
  dump: raw
  idle: 1s
  filter: 1us
  tolerance: 50%

climate:
  - platform: fujitsu_general
    name: "Living Room AC"
```

Panasonic codes:
```
[00:26:48][D][remote.panasonic:068]: Received Panasonic: address=0x28C6, command=0x00080840
[00:26:51][D][remote.panasonic:068]: Received Panasonic: address=0x28C6, command=0x0008087F
```

0x00080840 appears to mean "off" and 0x0008087F appears to mean "on" with no further details.

Raw codes received switching device on and off, changing temp a few times:
```
[00:31:54][D][remote.raw:028]: Received Raw: 3348, -1541, 482, -333, 478, -342, 473, -1149, 486, -333, 474, -1149, 485, -332, 476, -338, 477, -334, 481, -1149, 478, -1149, 482, -333, 478, -339, 476, -334, 482, -1150, 476, -1149, 482, -333, 478, -338, 478, -333, 482, -333, 482, -334,
[00:31:54][D][remote.raw:028]:   477, -338, 478, -333, 482, -334, 477, -338, 478, -333, 482, -334, 482, -333, 478, -333, 482, -1149, 478, -338, 478, -333, 482, -333, 478, -338, 479, -333, 481, -333, 482, -334, 477, -1149, 482, -334, 477, -338, 478, -333, 482, -334, 482, -1144, 482,
[00:31:54][D][remote.raw:028]:   -1149, 478, -1149, 482, -1149, 478, -1148, 482, -1145, 482, -1149, 477, -1149, 482, -334, 482, -333, 478, -1148, 482, -334, 478, -337, 478, -333, 483, -333, 482, -333, 478, -340, 475, -333, 482, -334, 478, -1148, 482, -1149, 478, -338, 477, -334, 481,
[00:31:54][D][remote.raw:028]:   -1149, 478, -333, 482, -334, 482, -333, 478, -338, 478, -1148, 482, -334, 477, -334, 482, -1148, 478, -338, 477, -334, 482, -333, 482, -334, 477, -334, 483, -332, 486, -330, 477, -338, 478, -1148, 482, -334, 477, -341, 475, -333, 482, -333, 482, -334,
[00:31:54][D][remote.raw:028]:   477, -334, 482, -333, 482, -334, 477, -338, 478, -333, 482, -333, 482, -334, 477, -334, 482, -333, 482, -334, 477, -338, 480, -331, 482, -334, 486, -332, 475, -333, 482, -334, 482, -333, 478, -338, 477, -334, 482, -333, 478, -338, 477, -334, 484, -331,
[00:31:54][D][remote.raw:041]:   482, -334, 477, -340, 476, -333, 482, -333, 478, -338, 478, -333, 482, -333, 482, -1145, 482, -333, 482, -334, 482, -333, 480, -333, 480, -1150, 477, -1148, 482, -334, 478, -337, 478, -333, 485, -1146, 478
[... 3 lines cut ...]
[00:32:04][D][remote.raw:028]: Received Raw: 3345, -1540, 481, -333, 478, -338, 477, -1149, 482, -334, 477, -1149, 482, -333, 482, -334, 478, -337, 480, -1147, 477, -1149, 482, -333, 482, -334, 477, -338, 478, -1149, 477, -1149, 482, -334, 482, -333, 478, -337, 478, -333, 482, -334,
[00:32:04][D][remote.raw:028]:   477, -338, 478, -337, 478, -333, 482, -334, 477, -339, 477, -333, 482, -334, 481, -334, 477, -1149, 482, -334, 486, -333, 473, -338, 478, -333, 482, -333, 482, -334, 478, -337, 478, -1148, 478, -338, 477, -334, 482, -334, 480, -335, 478, -1148, 482,
[00:32:04][D][remote.raw:028]:   -1149, 478, -1148, 482, -1149, 478, -1149, 481, -1145, 482, -1149, 477, -1149, 482, -333, 482, -334, 477, -1149, 482, -334, 477, -338, 478, -337, 478, -334, 483, -332, 477, -338, 478, -333, 482, -334, 482, -1144, 482, -1149, 481, -334, 478, -333, 482,
[00:32:04][D][remote.raw:028]:   -1149, 478, -338, 477, -334, 481, -334, 477, -338, 478, -1149, 482, -333, 478, -338, 477, -1149, 482, -333, 478, -338, 477, -334, 482, -333, 478, -337, 478, -333, 482, -334, 482, -333, 478, -1149, 482, -333, 482, -333, 478, -338, 477, -334, 485, -330,
[00:32:04][D][remote.raw:028]:   478, -338, 477, -334, 482, -333, 482, -338, 473, -338, 477, -334, 482, -333, 478, -338, 477, -334, 482, -333, 482, -334, 477, -338, 478, -333, 482, -333, 479, -337, 477, -334, 482, -333, 482, -334, 477, -338, 477, -334, 482, -333, 482, -334, 477, -338,
[00:32:04][D][remote.raw:041]:   478, -333, 482, -334, 480, -335, 477, -334, 482, -337, 478, -334, 477, -338, 478, -1149, 477, -338, 478, -337, 478, -333, 482, -334, 478, -1148, 482, -1149, 478, -338, 477, -334, 482, -333, 478, -1149, 482
[00:32:08][D][remote.raw:028]: Received Raw: 3345, -1539, 482, -333, 482, -334, 477, -1149, 482, -333, 482, -1149, 478, -333, 482, -334, 477, -338, 478, -1149, 482, -1148, 478, -338, 477, -334, 482, -333, 478, -1148, 482, -1149, 478, -337, 479, -333, 481, -334, 477, -339, 477, -337,
[00:32:08][D][remote.raw:028]:   478, -334, 482, -333, 478, -337, 478, -334, 482, -333, 478, -337, 478, -334, 482, -333, 482, -1145, 481, -334, 484, -331, 478, -338, 477, -339, 477, -333, 482, -333, 478, -338, 477, -1151, 476, -338, 477, -338, 478, -333, 482, -333, 478, -1149, 482,
[... cut ...]
[00:32:15][D][remote.raw:028]:   482, -329, 482, -333, 482, -334, 482, -329, 486, -333, 478, -334, 482, -333, 482, -329, 482, -333, 482, -334, 482, -329, 486, -329, 484, -333, 481, -333, 482, -333, 478, -334, 482, -333, 482, -333, 482, -329, 482, -334, 482, -333, 482, -329, 486, -330,
[00:32:15][D][remote.raw:041]:   481, -334, 482, -333, 482, -329, 486, -332, 480, -333, 482, -333, 483, -329, 482, -1145, 486, -329, 482, -334, 482, -1147, 479, -334, 482, -1144, 487, -1144, 482, -334, 482, -1144, 482, -1145, 482, -333, 482
```