esphome/issues #1244: Ethernet Module documentation infers only 3 connections are required.

marrold, 2020-05-24:
Hi, first of all thanks for creating and maintaining ESPHome!

The Ethernet Documentation infers that only 3 connections are required between the ESP32 and the LAN8720 module, but after checking the schematics for the wESP32 and ESP32-POE boards it's apparent that many more connections are required and they aren't documented.

Is it possible to update the documentation to include all the required connections ( And ideally a guide to connecting the generic LAN8720 modules ) or put a warning saying its only supported on specific boards?

Thanks

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 Dev Module

**Affected component:**

Ethernet

**Description of problem:**

The Ethernet Documentation infers that only 3 connections are required between the ESP32 and the LAN8720 module, but after checking the schematics for the wESP32 and ESP32-POE boards it's apparent that many more connections are required and they aren't documented.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

glmnet, 2020-05-25:
> it's apparent that many more connections are required

Did you actually check this or are just guessing our docs are wrong?

glmnet, 2020-09-22:
Ahh you're right, since this issue was opened I did lot's of test with the Ethernet stuff and indeed many connections are needed. Using ethernet interfaces on custom boards are so tricky due to high frequency signals mostly, so this is why there are not many commercially available break out boards of ethernet interfaces. 
So it's ok for ESPHome to target boards which are already pre built, if anybody comes with a new board the configuration can be added to the docs, no problem.

Just a clarification that the missing required wiring is not user configurable, i.e. you need certain wires set up but you cannot choose where to connect them.

If somebody wants to contribute the required wiring, it will be welcomed.

closed 2021-02-03 by stale[bot]