esphome/issues #1245: Logger UART0_SWAP not working when set uart

bestlong, 2020-05-25:
I need use UART0, and i set Logger hardware_uart UART0_SWAP and log message output GPIO15.
then i add set uart (id: uart_2, baud_rate: 115200, tx_pin: GPIO01, rx_pin: GPIO03).
do not know why log message output back to GPIO1 ?

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266, (nodemcuv2, d1_mini_pro)

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**

log message output GPIO15

```yaml
esphome:
  name: wemos_107_212
  platform: ESP8266
  board: d1_mini_pro

wifi:
  ssid: "my_ssid"
  password: "my_pw"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "wemos_107_212 Fallback Hotspot"
    password: "xxx"

captive_portal:

# Enable logging
logger:
  hardware_uart: UART0_SWAP

# Enable Home Assistant API
api:

ota:

status_led:
  pin:
    number: D4
    inverted: true

sensor:
  - platform: wifi_signal
    name: wifi
    update_interval: 5s
```

Add uart config log message output GPIO1, UART0_SWAP failure

```yaml
esphome:
  name: wemos_107_212
  platform: ESP8266
  board: d1_mini_pro

wifi:
  ssid: "my_ssid"
  password: "my_pw"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "wemos_107_212 Fallback Hotspot"
    password: "xxx"

captive_portal:

# Enable logging
logger:
  hardware_uart: UART0_SWAP

uart:
  id: uart_2
  baud_rate: 115200
  tx_pin: GPIO1
  rx_pin: GPIO3

# Enable Home Assistant API
api:

ota:

status_led:
  pin:
    number: D4
    inverted: true

sensor:
  - platform: wifi_signal
    name: wifi
    update_interval: 5s
```

**Logs (if applicable):**

Only set UART0_SWAP

```
[11:39:57][C][logger:175]: Logger:
[11:39:57][C][logger:176]:   Level: DEBUG
[11:39:57][C][logger:177]:   Log Baud Rate: 115200
[11:39:57][C][logger:178]:   Hardware UART: UART0_SWAP
```

set UART0_SWAP and uart

```
[11:38:00][C][uart:140]: UART Bus:
[11:38:00][C][uart:142]:   TX Pin: GPIO1
[11:38:00][C][uart:145]:   RX Pin: GPIO3
[11:38:00][C][uart:147]:   Baud Rate: 115200 baud
[11:38:00][C][uart:148]:   Stop bits: 1
[11:38:00][C][uart:150]:   Using hardware serial interface.
[11:38:00][W][uart:378]:   You're using the same serial port for logging and the UART component. Please disable logging over the serial port by setting logger->baud_rate to 0.
[11:38:00][C][logger:175]: Logger:
[11:38:00][C][logger:176]:   Level: DEBUG
[11:38:00][C][logger:177]:   Log Baud Rate: 115200
[11:38:00][C][logger:178]:   Hardware UART: UART0_SWAP
```

**Additional information and things you've tried:**

glmnet, 2020-05-25:
was this working before for you? I just pushed a uart commit to dev that might be related to this, are you running dev?

bestlong, 2020-05-25:
my esphome cli install by pip install esphome

``` shell
$ pip list
Package          Version
---------------- ----------
bottle           0.12.18
certifi          2020.4.5.1
chardet          3.0.4
click            7.1.2
colorama         0.4.3
colorlog         4.0.2
ecdsa            0.15
esphome          1.14.3
esptool          2.7
idna             2.9
ifaddr           0.1.6
paho-mqtt        1.4.0
Pillow           7.1.2
pip              20.1.1
platformio       4.0.3
protobuf         3.10.0
pyaes            1.6.1
pyserial         3.4
pytz             2019.3
PyYAML           5.1.2
requests         2.23.0
semantic-version 2.8.5
setuptools       40.6.2
six              1.15.0
tabulate         0.8.7
tornado          5.1.1
tzlocal          2.0.0
urllib3          1.25.9
voluptuous       0.11.7
```

I need stable UART and ESP8266 only UART0 have RX/TX both.

I just think that UART0_SWAP released GPIO3 / GPIO1 should be usable.

bestlong, 2020-05-25:
@glmnet Thank you for your reminder.

I will try using the second port UART1 for the logger.

closed 2020-06-04 by glmnet