esphome/issues #1248: ESP32 output.platform: ledc - connecting problems to WiFi

xsaren11, 2020-05-25:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

srottschaefer, 2020-06-01:
I have the same problem. I am using the ledc output to switch some LED lights and the wifi connection is very unstable after flashing new code. I have to reboot the ECU several times until it finally stays connected to my home network. It looks exactly the same with the empty SSID and all zeros for BSSID and the "WiFi Connected!" messages and instant disconnects.

MasterTim17, 2020-06-04:
I have the same problem!
With esp32 dev, and ttgo esp32

the verbose logs say i got wifi event 4 and 7 (STA Connected, STA Got IP), but the wifi scan is restarting direct after this. and after the attempt to reconnect i got the "Association Leave" error

`[21:02:23][I][wifi:193]: WiFi Connecting to 'myFRITZ!xxx'...
[21:02:23][V][wifi:195]: Connection Params:
[21:02:23][V][wifi:196]:   SSID: 'myFRITZ!xxx'
[21:02:23][V][wifi:199]:   BSSID: E8:DF:70:1C:xx:xx
[21:02:23][V][wifi:203]:   Password: 'xxxxxxx'
[21:02:23][V][wifi:205]:   Channel: 11
[21:02:23][V][wifi:215]:   Using DHCP IP
[21:02:23][V][wifi:217]:   Hidden: NO
[21:02:23][V][esp-idf:000]: I (67802) wifi: state: run -> init (0)
[21:02:23][V][esp-idf:000]: I (67808) wifi: pm stop, total sleep time: 3543045 us / 6959568 us
[21:02:23]
[21:02:23][V][esp-idf:000]: I (67816) wifi: n:11 0, o:11 2, ap:255 255, sta:11 2, prof:1
[21:02:23][I][wifi:423]: WiFi Connected!
[21:02:23][V][esp-idf:000]: W (67828) wifi: Haven't to connect to a suitable AP now!
[21:02:23][V][esp-idf:000]: W (67835) wifi: Haven't to connect to a suitable AP now!
[21:02:23][C][wifi:283]:   SSID: ''[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[21:02:23][W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 8 - ASSOC_LEAVE
[21:02:23][W][wifi_esp32:336]: Event: Disconnected ssid='myFRITZ!xxx' bssid=E8:DF:70:1C:xx:xx reason='Association Leave'
[21:02:23]
[21:02:23][C][wifi:284]:   IP Address: 0.0.0.0
[21:02:23][C][wifi:286]:   BSSID: 00:00:00:00:00:00
[21:02:23][C][wifi:287]:   Hostname: 'sensor_test'
[21:02:23][V][esp-idf:000]: W (67884) wifi: Haven't to connect to a suitable AP now!
[21:02:23][C][wifi:291]:   Signal strength: 0 dB ▂▄▆█
[21:02:23][V][wifi:293]:   Priority: -10.0
[21:02:23][C][wifi:295]:   Channel: 11
[21:02:23][C][wifi:296]:   Subnet: 0.0.0.0
[21:02:23][C][wifi:297]:   Gateway: 0.0.0.0
[21:02:23][C][wifi:298]:   DNS1: 192.168.178.1
[21:02:23][C][wifi:299]:   DNS2: 0.0.0.0
[21:02:23][W][wifi:100]: WiFi Connection lost... Reconnecting...
[21:02:23][V][esp-idf:000]: I (68027) wifi: n:11 2, o:11 0, ap:255 255, sta:11 2, prof:1
[21:02:23][V][esp-idf:000]: I (68029) wifi: state: init -> auth (b0)
[21:02:23][V][esp-idf:000]: I (68036) wifi: state: auth -> assoc (0)
[21:02:23][V][esp-idf:000]: I (68121) wifi: state: assoc -> run (10)
[21:02:23][V][esp-idf:000]: I (68153) wifi: connected with myFRITZ!xxx, channel 11, 40D
[21:02:23][V][esp-idf:000]: I (68154) wifi: pm start, type: 1
[21:02:23]
[21:02:23][D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
[21:02:23][V][wifi_esp32:324]: Event: Connected ssid='myFRITZ!xxx' bssid=E8:DF:70:1C:xx:xx channel=11, authmode=WPA2 PSK
[21:02:23][D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[21:02:23][D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.178.22, MASK: 255.255.255.0, GW: 192.168.178.1
[21:02:23][V][wifi_esp32:349]: Event: Got IP static_ip=192.168.178.22 gateway=192.168.178.1
[21:02:28][D][wifi:304]: Starting scan...
[21:02:30][D][WiFiGeneric.cpp:337] _eventCallback(): Event: 1 - SCAN_DONE
[21:02:30][V][wifi_esp32:307]: Event: WiFi Scan Done status=0 number=3 scan_id=138
[21:02:30][D][wifi:319]: Found networks:
[21:02:30][I][wifi:365]: - 'myFRITZ!xxx' (E8:DF:70:1C:xx:xx) ▂▄▆█
[21:02:30][D][wifi:366]:     Channel: 11
[21:02:30][D][wifi:367]:     RSSI: -72 dB
[21:02:30][D][wi
[21:02:30][I][wifi:193]: WiFi Connecting to 'myFRITZ!xxx'...
[21:02:30][V][wifi:195]: Connection Params:
[21:02:30][V][wifi:196]:   SSID: 'myFRITZ!xxx'
[21:02:30][V][wifi:199]:   BSSID: E8:DF:70:1C:xx:xx
[21:02:30][V][wifi:203]:   Password: 'xxxxxxx'
[21:02:30][V][wifi:205]:   Channel: 11
[21:02:30][V][wifi:215]:   Using DHCP IP
[21:02:30][V][wifi:217]:   Hidden: NO
[21:02:30][V][esp-idf:000]: I (75025) wifi: state: run -> init (0)
[21:02:30][V][esp-idf:000]: I (75031) wifi: pm stop, total sleep time: 3592852 us / 6875355 us
[21:02:30]
[21:02:30][V][esp-idf:000]: I (75039) wifi: n:11 0, o:11 2, ap:255 255, sta:11 2, prof:1
[21:02:30][I][wifi:423]: WiFi Connected!
[21:02:30][V][esp-idf:000]: W (75051) wifi: Haven't to connect to a suitable AP now!
[... 32 lines cut ...]
[21:02:37][D][wifi:319]: Found networks:
[21:02:37][I][wifi:365]: - 'myFRITZ!xxx' (E8:DF:70:1C:xx:xx) ▂▄▆█
[21:02:37][D][wifi:366]:     Channel: 11
[21:02:37][D][wifi:367]:     RSSI: -73 dB
[21:02:37][D][wifi:369]: - '' (FE:65:DE:0C:1D:34) ▂▄▆█
[21:02:37][D][wif
[21:02:37][I][wifi:193]: WiFi Connecting to 'myFRITZ!xxx'...
[21:02:37][V][wifi:195]: Connection Params:
[21:02:37][V][wifi:196]:   SSID: 'myFRITZ!xxx'
[21:02:37][V][wifi:199]:   BSSID: E8:DF:70:1C:xx:xx
[21:02:37][V][wifi:203]:   Password: 'xxxxxxx'
[21:02:38][V][wifi:205]:   Channel: 11
[21:02:38][V][wifi:215]:   Using DHCP IP
[21:02:38][V][wifi:217]:   Hidden: NO
[21:02:38][V][esp-idf:000]: I (82258) wifi: state: run -> init (0)
[21:02:38][V][esp-idf:000]: I (82264) wifi: pm stop, total sleep time: 3491024 us / 6955211 us
[21:02:38]
[21:02:38][V][esp-idf:000]: I (82272) wifi: n:11 0, o:11 2, ap:255 255, sta:11 2, prof:1
[21:02:38][I][wifi:423]: WiFi Connected!
[21:02:38][V][esp-idf:000]: W (82284) wifi: Haven't to connect to a suitable AP now!
[21:02:38][V][esp-idf:000]: W (82291) wifi: Haven't to connect to a suitable AP now!
[21:02:38][C][wifi:283]:   SSID: ''[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[21:02:38][W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 8 - ASSOC_LEAVE
[21:02:38][W][wifi_esp32:336]: Event: Disconnected ssid='myFRITZ!xxx' bssid=E8:DF:70:1C:xx:xx reason='Association Leave'
[21:02:38]`

closed 2020-10-22 by stale[bot]