esphome/issues #1251: pulsecounter not working

tiswi, 2020-05-27:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32
**Affected component:**

pulse_counter

**Description of problem:**
not counting pulses / stays 0
when trying to use the pulsecounter its not registrating any pulses:

**Problem-relevant YAML-configuration entries:**
```
sensor:
  - platform: pulse_counter
    name: 'Solar Power'
    id: solar_power
    update_interval: 5s
    pin:
      number: GPIO13
      mode: INPUT_PULLDOWN
    count_mode:
      rising_edge: DISABLE
      falling_edge: INCREMENT
    filters:
      - multiply: 0.06

```

**Logs (if applicable):**

```
[18:17:14][D][pulse_counter:159]: 'Solar Power': Retrieved counter: 0.00 pulses/min
[18:17:14][D][sensor:092]: 'Solar Power': Sending state 0.00000 pulses/min with 2 decimals of accuracy
[18:17:19][D][pulse_counter:159]: 'Solar Power': Retrieved counter: 0.00 pulses/min
[18:17:19][D][sensor:092]: 'Solar Power': Sending state 0.00000 pulses/min with 2 decimals of accuracy
[18:17:24][D][pulse_counter:159]: 'Solar Power': Retrieved counter: 0.00 pulses/min
[18:17:24][D][sensor:092]: 'Solar Power': Sending state 0.00000 pulses/min with 2 decimals of accuracy
[18:17:29][D][pulse_counter:159]: 'Solar Power': Retrieved counter: 0.00 pulses/min
[18:17:29][D][sensor:092]: 'Solar Power': Sending state 0.00000 pulses/min with 2 decimals of accuracy
[18:17:34][D][pulse_counter:159]: 'Solar Power': Retrieved counter: 0.00 pulses/min
[18:17:34][D][sensor:092]: 'Solar Power': Sending state 0.00000 pulses/min with 2 decimals of accuracy
[18:17:39][D][pulse_counter:159]: 'Solar Power': Retrieved counter: 0.00 pulses/min
[18:17:39][D][sensor:092]: 'Solar Power': Sending state 0.00000 pulses/min with 2 decimals of accuracy
[18:17:44][D][pulse_counter:159]: 'Solar Power': Retrieved counter: 0.00 pulses/min
[18:17:44][D][sensor:092]: 'Solar Power': Sending state 0.00000 pulses/min with 2 decimals of accuracy
[18:17:49][D][pulse_counter:159]: 'Solar Power': Retrieved counter: 0.00 pulses/min
[18:17:49][D][sensor:092]: 'Solar Power': Sending state 0.00000 pulses/min with 2 decimals of accuracy
[18:17:54][D][pulse_counter:159]: 'Solar Power': Retrieved counter: 0.00 pulses/min
[18:17:54][D][sensor:092]: 'Solar Power': Sending state 0.00000 pulses/min with 2 decimals of accuracy
[18:17:59][D][pulse_counter:159]: 'Solar Power': Retrieved counter: 0.00 pulses/min
[18:17:59][D][sensor:092]: 'Solar Power': Sending state 0.00000 pulses/min with 2 decimals of accuracy

```

**Additional information and things you've tried:**
I tried following code to check there were any pulses:
```
binary_sensor:
  - platform: gpio
    name: pulse
    id: pulse
    pin:
      number: GPIO13
      mode: INPUT_PULLDOWN
    on_press:
    - lambda: |-
        static int num_executions = 0;
        ESP_LOGD("main", "I am at execution number %d", num_executions);
        num_executions += 1;

```
and this was working fine with an increasing counter after each pulse:

```
[18:13:25][D][binary_sensor:036]: 'pulse': Sending state ON
[18:13:25][D][main:566]: I am at execution number 518
[18:13:25][D][binary_sensor:036]: 'pulse': Sending state OFF
[18:13:32][D][binary_sensor:036]: 'pulse': Sending state ON
[18:13:32][D][main:566]: I am at execution number 519
[18:13:32][D][binary_sensor:036]: 'pulse': Sending state OFF
[18:13:39][D][binary_sensor:036]: 'pulse': Sending state ON
[18:13:39][D][main:566]: I am at execution number 520
[18:13:40][D][binary_sensor:036]: 'pulse': Sending stat
```

glmnet, 2020-06-16:
This is an interesting one, testing your code I found out that the pulse counter works, its just that it is pulling up, not pulling down.

In ESP32 pulse counter is done in hardware, uses esp-idf API for this, and it seems to have a bug, see here https://www.esp32.com/viewtopic.php?t=6346

I hacked ESPHome code to setup pin after setup the counter and then the pulse counter works as expected, pulling down.

It feels odd though.

Basically move this line https://github.com/esphome/esphome/blob/3fd130869e81d962c4aebf3b91064d82727bffb5/esphome/components/pulse_counter/pulse_counter_sensor.cpp#L49 to line 96

closed 2020-12-05 by stale[bot]

i7i5, 2021-10-04:
I have exact the same problem trying to count pulses with my esp32.

Originally I got hope that separating the pulse counters within these 2 Prs would help sort of:
https://github.com/esphome/esphome/pull/1777
https://github.com/esphome/esphome/pull/1778

But they got auto closed due to tests in wrong file smth (1777) and an import fault caused by renaming in 1778.

My instance, fresh installation.
Home Assistant 2021.9.7
ESPHome: 2021.9.2