esphome/issues #1255: NTC sensor not returning the correct temperature

SeByDocKy, 2020-05-28:
Hassio

ESP32

ESP32 WROOM Dev kit 1.0

https://next.esphome.io/components/sensor/ntc.html

Returning bad temperatures

**Problem-relevant YAML-configuration entries:**
```
 - platform: adc
    id: sensor_resistance
    pin: GPIO032
    update_interval: 5s
    attenuation: 0db
    
  - platform: resistance
    sensor: sensor_resistance
    configuration: UPSTREAM
    resistor: 10.0kOhm
    name: resistance_efs
    id: resistance_efs
    
    
  - platform: ntc
    sensor: sensor_resistance
     calibration:
        - 200.720kOhm -> 6.0°C
        - 57.094kOhm -> 22.0°C  
        - 28.766kOhm -> 37.0°C
    name: temperature_efs  
    unit_of_measurement: "°C"
    icon: "mdi:water-percent"
    accuracy_decimals: 2

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

I used instead a basic template sensor as:

```
- platform: template
    name: temperature_efs
    unit_of_measurement: "°C"
    icon: "mdi:water-percent"
    accuracy_decimals: 2  
    update_interval: 5s  
    lambda: |-
      float A = -0.0023372161946098318;
      float B = 0.0006789562966514373;
      float C = -1.3022483012452408e-06;
      float LR = log(id(resistance_efs).state);
      return  (1.0/(A + B*LR + C*LR*LR*LR) - 273.15);
````

And readings are ok. A,B,C have been fitted with three couples (Ri,Ti) correspondly to the wikipedia
https://fr.wikipedia.org/wiki/Relation_de_Steinhart-Hart

ideal2545, 2020-06-08:
Did you ever resolve this using the normal NTC platform? I think I'm running into the same issue

SeByDocKy, 2020-06-08:
> Did you ever resolve this using the normal NTC platform? I think I'm running into the same issue

No .....  I am waiting the bugfix..... and still running my template sensor instead as workaround

ideal2545, 2020-06-09:
> 
> 
> > Did you ever resolve this using the normal NTC platform? I think I'm running into the same issue
> 
> No ..... I am waiting the bugfix..... and still running my template sensor instead as workaround

Hi SeByDocKy, I got this to work using the calibration by doing ohm instead of Kohm

```
sensor:
  - platform: ntc
    sensor: resistance_sensor
    calibration:
      # b_constant: 2553
      # reference_temperature: 25°C
      # reference_resistance: 1514kOhm
      - 2966.00439Ohm -> 3.3°C
      - 1965.15833Ohm -> 22.7°C
      - 438.65070Ohm -> 75.5°C      
    name: NTC Temperature
```

SeByDocKy, 2020-06-10:
> Hi SeByDocKy, I got this to work using the calibration by doing ohm instead of Kohm
>
> ```
> sensor:
>   - platform: ntc
>     sensor: resistance_sensor
>     calibration:
>       # b_constant: 2553
>       # [...]

Ho Thanks .... I have to try :)

SeByDocKy, 2020-06-14:
I tried in ohms and for me it's not working at all :( .... I tried many values by shifing the dot no success.

FreeWillyB, 2020-07-22:
Man I wish I checked here earlier. I just spent 3 hours banging me head against the wall trying to figure out why this amazingly simple circuit wasn’t working. Gonna have to try some of y’all’s fixes to see if that will work

closed 2020-12-05 by stale[bot]