esphome/issues #1256: D1 mini  - as3935 - anyone got it to work?

Sc0th, 2020-05-29:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker 1.15.0-dev

**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos D1 Mini

**Affected component:**

as3935

**Description of problem:**

Has anyone managed to get one of these working? I have tried and tried but not a squeak out output, I live in Singapore so it has lots of opportunities, but no matter what I do, nothing...

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

esphome:
  name: d1_mini_detect
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: ""
  password: !secret password

  manual_ip:
    static_ip: 
    gateway: 
    subnet: 
    dns1: 
    dns2: 

mqtt:
  broker: mqtt
  discovery: false

logger:
  level: VERY_VERBOSE

ota:

i2c:
  sda: D4
  scl: D5

as3935_i2c:
  irq_pin: D6

sensor:
 - platform: wifi_signal
   name: "WiFi Signal"
   retain: false
 - platform: as3935
   lightning_energy:
     name: "Lightning Energy"
   distance:
     name: "Distance Storm"

binary_sensor:
  - platform: as3935
    name: "Storm Alert"

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

`May 29 17:12:43 [V][as3935:175]: Calling read_interrupt_register_
May 29 17:12:43 [VV][i2c:056]: Beginning Transmission to 0x03:
May 29 17:12:43 [VV][i2c:093]:     Writing 0b00000011 (0x03)
May 29 17:12:43 [VV][i2c:061]:     Transmission ended. Status code: 0x00
May 29 17:12:43 [VV][i2c:083]: Requesting 1 bytes from 0x03:
May 29 17:12:43 [VV][i2c:113]:     Received 0b00011100 (0x1C)`
```

**Additional information and things you've tried:**

I get the above in the logs on constant repeat so it's seeing something, I have no idea if this is a good something or a bad something.

This is my last ditch effort to get it working before it goes in 'the drawer' insights, suggestions, tales of success would be much appreciated!

HugoVonHory, 2020-06-08:
I have five D1 minis in sensors with no issues. First flash over USB and next OTA. Example config (without sensors):

`esphome:
  name: example
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: 'name'
  password: 'pwd'
  manual_ip:
    static_ip: 10.0.0.x
    gateway: 10.0.0.y
    subnet: 255.255.255.0

logger:
 level: WARN

api:
  password: 'apipwd'
  id: 'sample_sensor'
  
ota:
  password: 'otapwd'`

edwardscaleb9, 2020-06-22:
I'm trying to get one working with an Olimex ESP32-POE.  I can see the sensor, but I'm constantly getting "Disturber was detected - Try increasing the spike rejection value".

closed 2020-11-03 by stale[bot]