esphome/issues #1258: ESPHome complier hard crashes HassOS 0.110.3 and/or core

jazzmonger, 2020-05-29:
## The problem

Clicking UPLOAD on ESPHOME crashes HassIO and HOME ASSISTANT Core.
I've reproduced it 3 times in a row:

[image]

This causes a hard crash of HassOS.  the 3rd time I was logged into the Pi command line as root on a console and pressed enter repeatedly to see when/where the lockup occurred.  It happens right at the last line of the ESPhome complier (api_server.cpp)  every time.

## Environment

Raspberry Pi 3b

arch | aarch64
-- | --
chassis | embedded
dev | false
docker | true
hassio | true
host_os | HassOS 3.13
installation_type | Home Assistant
os_name | Linux
os_version | 4.19.114-v8
python_version | 3.7.7
supervisor | 225
timezone | America/Los_Angeles
version | 0.110.3
virtualenv | false

- Home Assistant Core release with the issue: 0.110.3
- Last working Home Assistant Core release (if known): 0.110.0
- Operating environment (Home Assistant/Supervised/Docker/venv): HassOS 3.13  
- Integration causing this issue: 
- Link to integration documentation on our website: 
ESPHOME

## Problem-relevant `configuration.yaml`

```yams
esphome:
  name: coop
  platform: ESP8266
  board: nodemcuv2
wifi:
  ssid: "XX"
  password: "XX"
#  power_save_mode: none
  # ...
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.1.145
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.1.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0
    dns1: 192.168.1.250
    dns2: 8.8.8.8
# Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Coop ESP Fallback hotspot"
    password: "xx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
#  ap:
#    ssid: "Coop ESP Fallback hotspot"
#    password: "xx"

logger:
  level: DEBUG
# Enable Home Assistant API
api:
ota:
#-----------------------------
binary_sensor:
#-----------------------------
  - platform: gpio
    name: "Coop Bottom Button"
    pin:
      number: GPIO00
      inverted: true
      mode: INPUT_PULLUP
#
  - platform: gpio
    name: "Coop Top Button"
    pin:
      number: GPIO02
      inverted: true
      mode: INPUT_PULLUP
#
  - platform: gpio
    name: "Coop Outside Reed SW"
    device_class: door
    pin:
      number: GPIO04
      inverted: false
      mode: INPUT_PULLUP
#
  - platform: gpio
    device_class: door
    name: "Coop Inside Reed SW"
    pin:
      number: GPIO05
      inverted: false
      mode: INPUT_PULLUP
#-----------------------------
switch:
#-----------------------------
  - platform: output
    inverted: false
    name: "Coop Door Inside"
    output: 'coop_out_12'
#
  - platform: output
    name: "Coop Door Outside"
    inverted: false
    output: 'coop_out_14'
#
#-----------------------------    
output:
#-----------------------------
#
  - platform: gpio
    pin: 12
    inverted: true
    id: 'coop_out_12'
#
  - platform: gpio
    pin: 14
    inverted: true
    id: 'coop_out_14'
#Uptime Sensor-----------------------------
sensor:
#-----------------------------
  - platform: uptime
    name: Coop ESP8266 Upensor
    filters:
      - lambda: return x / 3600.0;
    unit_of_measurement: "hours"
    accuracy_decimals: 2
    update_interval: 600s #10 mmin

```

## Traceback/Error logs

```txt

```

## Additional information

erwinf, 2020-06-01:
I have the issue compiling for esp32. Tried it multiple times and it crashes at the api_server.cpp line.

brandond, 2020-06-01:
You're most likely running out of memory. What else are you running on your pi? Can you compile successfully if you stop other things beforehand?

jazzmonger, 2020-06-02:
Not much.  Running TOP shows Im not running out of memory... wouldn't ESPHOME account for this if it was the case and just terminate?  crashing the entire system is pretty extreme....

truglodite, 2020-06-03:
I had the same problem when I was using a pi3b+. In fact it crashed often right at the same time as yours does (api_server.cpp.o), but occasionally would crash on something later in compilation. I upgraded to a pi4-4gb and haven't had the problem since. 

Unfortunately, it seems like a waste of a pi4. The pi3b+ was more than enough for everything else going on in my HA system (which is probably average to above average in size/complexity). We need at least a pi4 just to be able to reliably compile esphome hex files, every great once in a while when the need arises. I guess there is be some currently unknown memory issue that could be fixed with maybe more research. Whether it's a problem in HA or esphome code, there is a very worthy cause to fix it... freeing up a bunch of underutilized pi4s for more power hungry applications.

There's always going to be folks running HA on full sized machines for various reasons (usually for handling video cameras, etc). You can't really do those things on a pi4 anyways. So kind of interesting, that the pi4's only true fit in the HA sphere is just to fullfill compiling esphome. :P

jazzmonger, 2020-06-03:
I just updated HOME ASSISTANT HASSOS to 4.3, and I'm on 100.4.  Here's a pic of the pi at crash time.  It crashes in the same place each time.

[image]

brandond, 2020-06-03:
It's either overheating (unlikely due to CPU throttling), drawing too much power and browning out the supply, or using so much memory that it becomes unresponsive while swapping. It is not recommended to run swap on a SD card as it will burn it out very quickly; I would disable swap so that you can at least let the OOM killer terminate things quickly.

Compiling is one of the more resource-intensive things you can do on a device. It pushes the cpu, memory, and disk simultaneously. If your host crashes when compiling, it is not an issue with the compiler, or the software you are compiling (esphome) - it is an issue with the os configuration, kernel, or hardware.

truglodite, 2020-06-03:
I have to agree it is likely an issue with hassos and/or the kernel. Unfortunately, that doesn't bode well for a solution to this issue. Hassos and even more so other os's used to host ha have very broad use cases that esphome has little to do with. It would take dev's pulling out of more popular issues to fix it. In that context, I think a pi4 isn't that much more expensive that pi3.

As a data point, I figure I should mention my pi3b+ was running a canakit wallwart (known to be reliable), and I was running hassos installed on an ssd drive w/ usb-sata adaptor.

jazzmonger, 2020-06-04:
I'd love to test it now that I just migrated everything to my Pi 4 but ESPHOME is MISSING from HOME ASSISTANT Add-ons.  I've tried adding the repository manually for the production and dev versions to no avail.

I can still control devices through the esphome API (which is odd) but no ESPHome integration...

arch | armv7l
-- | --
chassis | embedded
dev | false
docker | true
hassio | true
host_os | HassOS 4.8
installation_type | Home Assistant
os_name | Linux
os_version | 5.4.42-v7l
python_version | 3.7.7
supervisor | 226
timezone | America/Los_Angeles
version | 0.110.4
virtualenv | false

truglodite, 2020-06-04:
Jazzmonger, double check your router configs. That has happened to me at least twice, and both times it was my router ad blocking list at fault (Diversion medium).

jazzmonger, 2020-06-04:
No luck. My Router is just a router.  I disabled AdGuard on ha but still no esphome... maddening.

Any other suggestions?  It was fine on my pi3's until I did this fresh install on a pi4. All other integrations are visible as well as hacs.  Interestingly, after I restored my configuration,  none of the integrations would start (no log entries either so no indication of what was wrong) so I had to uninstall and then add each one back.

jazzmonger, 2020-06-04:
> It's either overheating (unlikely due to CPU throttling), drawing too much power and browning out the supply, or using so much memory that it becomes unresponsive while swapping. It is not [...]

This issue was happening on a pi3 booting haasOS from an ssd, so no SD cards were in the picture.  I'm happy to try just about anything.  

How do you disable swapping on HassOS?

jazzmonger, 2020-06-05:
Moving to the Pi4 and installing the dev branch, I was successful in getting ESPHOME to compile.

arch | armv7l
-- | --
chassis | embedded
dev | false
docker | true
hassio | true
host_os | HassOS 4.9
installation_type | Home Assistant
os_name | Linux
os_version | 5.4.42-v7l
python_version | 3.7.7
supervisor | 227
timezone | America/Los_Angeles
version | 0.110.4
virtualenv | false

trollsoft7, 2020-06-05:
Update to ESPHome 1.14.4. solved the compiler crash for me perfectly. Running now smoothly on RPi4 and 0.110.4 Homeassistant. Thanks!

closed 2020-06-11 by glmnet