esphome/issues #1260: Unable to control entities using HTTP Web Server without access to WAN internet

JeffResc, 2020-06-03:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266
**Affected component:**

HTTP Web Server

**Description of problem:**
Unable to set state of entity without access to WAN on device to download JS/CSS from external server (esphome.io). This becomes an issue when the ISP internet is down and you are unable to easily control your ESP still. I ended up having to use the REST API to control my lights, but I expected it to be more seem less.

**Problem-relevant YAML-configuration entries:**
```yaml
N/A

```

**Logs (if applicable):**

```
N/A
```

**Additional information and things you've tried:**

randybb, 2020-06-03:
Just switch to dev branch where you have more options https://next.esphome.io/components/web_server.html

glmnet, 2020-06-04:
Ahh yes, you now have options to host locally the css / js files (e.g. local web server) or even include them inside the ESP itself. Thanks to the contributors!!

We really need a new version!

closed 2020-06-04 by glmnet