esphome/issues #1261: Adafruit_SPIDevice.h:1:17: fatal error: SPI.h: No such file or directory

SeByDocKy, 2020-06-03:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
HASSIO
Please provide details about your environment below this line. -->

ESP32 WROOM32

**Affected component:**

**Description of problem:**
My configuration file is no more compiling. Please find the error: https://pastebin.com/tmd26KJP

Mainly seems there is a change with Adafruit_BusIO lib

**Problem-relevant YAML-configuration entries:**
```yaml

esphome:
  name: station_meteo
  platform: ESP32
  board: esp32dev
  includes:
    - veml6070_custom_sensor.h
    - tx23_custom_sensor.h
  libraries:
    - "https://github.com/adafruit/Adafruit_VEML6070"
    - "https://github.com/egzumer/Arduino-LaCrosse-TX23-Library"
 
wifi:
  ssid: !secret esphome_ssid
  password: !secret esphome_password

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

time:

  - platform: homeassistant
    id: homeassistant_time  
    
    on_time:
      - seconds: 58
        minutes: 59
        hours: 23
        then:  
          - sensor.template.publish:
              id: template_yesteday_rain
              state: !lambda return id(daily_rain).state;

i2c:
  sda: GPIO22
  scl: GPIO21

binary_sensor:

  - platform: status
    name: "station_meteo_status"
    
    
switch:
  - platform: restart
    name: "restart_station_meteo"    

sensor:

  - platform: pulse_counter
    pin:
      number: GPIO12
      inverted: True
      mode: INPUT_PULLUP    
    unit_of_measurement: 'mm'
    update_interval: 2s
    accuracy_decimals: 3
    name: 'station_meteo_compteur_pluviometre'
    id: rain_meter
    filters:
      - multiply:  25 #0.0250 #0.367    #90/2.25ml om 90 est la valeur pour chaque bascule ave 2s et 2.25ml est le volume de chaque réservoir

  - platform: total_daily_energy
    name: "station_meteo_pluie_du_jour"
    power_id: rain_meter  
    unit_of_measurement: "mm"
    accuracy_decimals: 3
    id: daily_rain
    icon: mdi:water
    
  - platform: template
    name: "station_meteo_pluie_veille"
    id: template_yesteday_rain
    unit_of_measurement: "mm"
    accuracy_decimals: 3
    icon: mdi:water
    update_interval: 10s
    
  - platform: bme280
    temperature:
      name: "station_meteo_temperature"
      oversampling: 16x
      id: station_meteo_temperature
    pressure:
      name: "station_meteo_pression"
      id: station_meteo_pression
    humidity:
      name: "station_meteo_humidite"
      id: station_meteo_humidite
    address: 0x76
    update_interval: 5s

  #- platform: tx20
  #  wind_speed:
  #    name: "station_meteo_vitesse_vent"
  #    unit_of_measurement: "m/s"
  #    icon: "mdi:weather-windy"
  #    accuracy_decimals: 1
  #  wind_direction_degrees:
  #    name: "station_meteo_direction_vent"
  #  pin:
  #    number: GPIO2
  #    #mode: INPUT_PULLUP
  #  id: tx20_id

  - platform: custom
    id: tx23_id
    lambda: |-
      auto TX23 = new TX23CustomSensor();
      App.register_component(TX23);
      return {TX23->speed_sensor, TX23->direction_sensor};

    sensors:
    - name: "station_meteo_vitesse_vent"
      unit_of_measurement: "m/s"
      accuracy_decimals: 1
      icon: "mdi:weather-windy"
    
    - name: "station_meteo_direction_vent"
      unit_of_measurement: "°"
      icon: "mdi:compass-outline"
      id: "station_meteo_direction_vent"
   
      
  - platform: bh1750
    name: "station_meteo_luminosite"
    address: 0x23
    update_interval: 5s
    
  - platform: pulse_counter
    pin: GPIO36
    name: "station_meteo_compteur_geiger"
    update_interval: 15s    
  
  - platform: custom
    lambda: |-
      auto veml6070 = new VEML6070CustomSensor();
      App.register_component(veml6070);
      return {veml6070};
    sensors:
      name: "station_meteo_UV"
      unit_of_measurement: "mW/m²"
      accuracy_decimals: 2
      filters:
      - delta: 10
      - multiply: 0.03
      - filter_out: 65535
      - lambda: |-
          if (x > 600) {
            return 0;
          } else {
            return x;
          }
  - platform: uptime
    name: "up_station_meteo"
    id: uptime_sec
    
  - platform: wifi_signal
    name: "WiFi puissance_station_meteo"
    update_interval: 10s  
    
#  - platform: template
#    name: station_meteo_uptime
#    lambda: |-
#      int seconds = (id(uptime_sec).state);
#      int days = seconds / (24 * 3600);
#      seconds = seconds % (24 * 3600); 
#      int hours = seconds / 3600;
#      seconds = seconds % 3600;
#      int minutes = seconds /  60;
#      seconds = seconds % 60;
#      return { (String(days) + "d " + String(hours) + "h " + String(minutes) + "m "+ String(seconds) + "s").c_str() };
#    icon: mdi:clock-start
#    update_interval: 113s  

#text_sensor:
#  -platform: template
#   name: "station_meteo_fleche_vent"
#   lambda: |-
#     static const char *dirTable[] = {"N", "NNE", "NE", "ENE", "E", "ESE", "SE", "SSE", "S", "SSW", "SW", "WSW", "W", "WNW", "NW", "NNW"};
#     return {dirTable[id(tx23_id).state};

#text_sensor:
#  - platform: template
#    name: "station_meteo_fleche_vent"
#    lambda: |-
#      return {id(tx23_id).get_wind_cardinal_direction()};
#    update_interval: 2s

 

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

SeByDocKy, 2020-08-12:
> Same question I have.I google it find that may Arduino lib version is old and unmatched this header file.I try to find missing header file, but more and more missing file appear in compiling.

Just add in your yaml file

```
esphome:
  name: station_meteo
  platform: ESP32
  board: esp32dev
  libraries:
    - SPI

```

martin3000, 2020-08-12:
Thanks @SeByDocKy , that fixes it!

ChiaSang, 2020-08-13:
> Just add in your yaml file
>
> ```
> esphome:
>   name: station_meteo
>   platform: ESP32
>   board: esp32dev
>   libraries:
>     - SPI
> ```

Yes, I find this solution in esphome community, add missing SPI.H in lib.thank you all the time!

glmnet, 2020-08-14:
This is a custom component library dependency issue which is solved, should this issue be closed?

SeByDocKy, 2020-08-14:
> This is a custom component library dependency issue which is solved, should this issue be closed?

It's probably related to plateformio but I don't know if it comes from the ESPhome's plateformio configuration or plateformio standalone ....

martin3000, 2020-08-14:
Here the problem arose after updating ESPhome with pip3.

closed 2020-12-26 by stale[bot]