esphome/issues #1263: ESPHOME missing from HassOS Add-ons

jazzmonger, 2020-06-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

arch | armv7l
-- | --
chassis | embedded
dev | false
docker | true
hassio | true
host_os | HassOS 4.8
installation_type | Home Assistant
os_name | Linux
os_version | 5.4.42-v7l
python_version | 3.7.7
supervisor | 226
timezone | America/Los_Angeles
version | 0.110.4
virtualenv | false

**ESP (ESP32/ESP8266, Board/Sonoff):**

several
**Affected component:**

ALL - ESPHOME is MISSING from HOME ASSISTANT Add-ons.  I've tried adding the repository manually for the production and dev versions to no avail.

**Description of problem:**
ESPHOME is MISSING from HOME ASSISTANT Add-ons.  I've tried adding the repository manually for the production and dev versions to no avail.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

Caguillo, 2020-06-14:
is there any help here?

jazzmonger, 2020-06-14:
I had to add the dev repository manually and then it showed up

tombrain, 2020-07-30:
> I had to add the dev repository manually and then it showed up

how do i add it manually?

OttoWinter, 2020-07-30:
@tombrain Same as you did when installing - add the repository in the top right.

Though nowadays you can also just install it from the pre-installed community hass.io addons.

This sounds like a bug in the hassio supervisor to me. There was no change whatsoever to the hassio adodn repository at that time, so it's likely because some bug in the supervisor removed the repository.

closed 2020-07-30 by OttoWinter