esphome/issues #1264: Warning that not connected to the internet despite the node on wifi and sending data

Newms118, 2020-06-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

I am running Home Assistant (HASSIO) on a Raspberry Pi 4 4Gb version. 

**ESP (ESP32/ESP8266, Board/Sonoff):**

I'm using a NodeMCU ordered off of amazon (HiLetgo 3pcs ESP8266 NodeMCU CP2102 ESP-12E Internet WiFi Development Board Open Source Serial Wireless Module Works Great for Arduino IDE/Micropython (Large))

**Affected component:**

**Description of problem:**

I'm doing some really simple work here.  I'm following this video to make BRUH sensors: https://www.youtube.com/watch?v=9Yu57vjz7AY.

A few days ago I successfully set up two NodeMCUs with the DHT22 sensor, the compilation went fine, they were able to be flashed OTA after I used the Rpi4 USB to do the initial compilation. 

Today I'm trying to add the PIR and an LED.  Both of my running NodeMCUs have been submitting data to my HA all day and I've been able to track this.  So to install the PIR and LED, I went to the relevant code section on Esphome and added this to the yaml file. Went to upload and compile and I get an error saying, "Warning! You are not connected to the Internet." This repeats twice and I've include the text from the log below (sorry it may not be in the right format, dont know how to submit something in debug mode).  

Not being connected to the internet doesn't make sense, the node is online, submitting data to HA. In case there was something wrong with wifi, I reconnected my node sensor back to the Rpi4 using the USB and tried to compile using USB (like I originally did) instead of OTA.  I got the same error, saying it still cant connect to the internet, despite it still being connected already. 

In case it was something wrong with the chip, I created a new node, using a new NodeMCU with nothing else connected to it. I connected the NodeMCU to the RPi using the USB as done before.  That gave me the second error log shown below. 

Ive search around and cant find any information online or in the forums, so any help would be greatly appreciated.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: temp_sensor_node_1
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "XXXXXXXX"
  password: "XXXXXXXXXXXXXXXXX"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Temp Sensor Node 1"
    password: "ymongU3KFi1y"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
sensor:
  - platform: dht
    pin: D7
    model: dht22
    temperature:
      name: "SenseNode Temp"
    humidity:
      name: "SenseNode Humidity"
    update_interval: 30s
    
binary_sensor:
  - platform: gpio
    pin: D6
    name: "SenseNode Motion"
    device_class: motion
    
light:
  - platform: fastled_clockless
    chipset: WS2812
    pin: D3
[... 157 lines cut ...]

 #include <ESPAsyncTCP.h>
                         ^
compilation terminated.
*** [/data/temp_sensor_node_3/.pioenvs/temp_sensor_node_3/src/esphome/components/api/api_connection.cpp.o] Error 1
In file included from src/esphome/components/api/api_server.cpp:1:0:
src/esphome/components/api/api_server.h:19:25: fatal error: ESPAsyncTCP.h: No such file or directory

*********************************************************************
* Looking for ESPAsyncTCP.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESPAsyncTCP.h"
* Web  > https://platformio.org/lib/search?query=header:ESPAsyncTCP.h
*
*********************************************************************

 #include <ESPAsyncTCP.h>
                         ^
compilation terminated.
*** [/data/temp_sensor_node_3/.pioenvs/temp_sensor_node_3/src/esphome/components/api/api_server.cpp.o] Error 1
========================== [FAILED] Took 8.95 seconds ==========================
```

**Additional information and things you've tried:**
I've tried stopping and restarting Esphome addon.  I've restarted HA.  Nothing has changed.

2016for, 2020-06-04:
+1
Same here
Getting "Warning! You are not connected to the Internet."
Was working fine yesterday

TorbjornZachrisson, 2020-06-04:
+1
Same here
Tried everything. Different computers. Docker and no docker. Beta release.

anthonylavado, 2020-06-04:
We figured out the problem in the Discord. I'll open another issue with how to fix it.

glmnet, 2020-06-04:
Closing. Fixes and temp workarounds https://github.com/esphome/issues/issues/1265

closed 2020-06-04 by glmnet