esphome/issues #1265: PlatformIO fails with "Warning! You are not connected..."

anthonylavado, 2020-06-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Debian 10 (Buster), with pip in a venv
(This issue affects all installs of stable 1.14.3)

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 (nodemcuv2), but the issue affects any new boards or libraries not yet used by the user

**Affected component:**

Any new libraries needed when compiling/building.

**Description of problem:**
In ESPHome stable (1.14.3), PlatformIO is pinned to version 4.0.3.

In PlatformIO's `util.py`, it has two hardcoded IP addresses to check internet connectivity:
https://github.com/platformio/platformio-core/blob/v4.0.3/platformio/util.py#L381-L384

PlatformIO's code will check each IP for a connection on port 80. If any of them works, it continues. If they all fail, it decides you're not connected to the internet, and doesn't try to download anything from the library.

Both the IP Addresses listed currently fail this check. Starting in 4.1.0 and newer, they added more IP Addresses. In the latest one (4.3.4 at the time of writing) there are some DNS names as well (GitHub.com, platformio.org).

Because all checks currently fail, PlatformIO will fail to download any new libraries you haven't used/cached yet.

**Problem-relevant YAML-configuration entries:**

**Logs (if applicable):**

**Additional information and things you've tried:**
If you switch to the development version of ESPHome, PlatformIO 4.3.3 is installed, which will pass the checks.

anthonylavado, 2020-06-04:
I have temporary fix for those with access to the command line of their system.
Assuming you have IPv4 Forwarding turned on, you can use IPTables to redirect the request. It won't last past a reboot, but it will work.

The command is:
`iptables -t nat -A OUTPUT -d 192.30.253.113 -j DNAT --to-destination 31.28.1.238`

References:
https://superuser.com/questions/681705/using-iptables-to-redirect-ip-address
https://www.debuntu.org/how-to-redirecting-network-traffic-to-a-new-ip-using-iptables/

anthonylavado, 2020-06-04:
From the Discord for Hass Add-On users:

> Of thoes who want the EPSHOME Hass addon, go to the add on store where ESPHOME is, there is 3 branches available when you add in the ESPHOME link. Stop the current running ESPHOME, add the dev branch and then run it, running two at once doesnt work unless I assume you change the port

dibyadip-das, 2020-06-04:
I am facing the same problem. Below are my logs
```
INFO Reading configuration /config/esphome/multisensor.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/multisensor
Processing multisensor (board: d1_mini; framework: arduino; platform: espressif8266@2.2.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Looking for ESPAsyncTCP-esphome library in registry
Warning! You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.
Looking for ESPAsyncWebServer-esphome library in registry
Warning! You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.
Warning! You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.
Looking for ArduinoJson-esphomelib library in registry
Warning! You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <Wire> 1.0
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
Compiling /data/multisensor/.pioenvs/multisensor/src/esphome/components/api/api_connection.cpp.o
Compiling /data/multisensor/.pioenvs/multisensor/src/esphome/components/api/api_server.cpp.o
Compiling /data/multisensor/.pioenvs/multisensor/src/esphome/components/api/list_entities.cpp.o
Compiling /data/multisensor/.pioenvs/multisensor/src/esphome/components/api/subscribe_state.cpp.o
In file included from src/esphome/components/api/api_connection.h:7:0,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/api/api_server.h:19:25: fatal error: ESPAsyncTCP.h: No such file or directory

*********************************************************************
* Looking for ESPAsyncTCP.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESPAsyncTCP.h"
* Web  > https://platformio.org/lib/search?query=header:ESPAsyncTCP.h
*
*********************************************************************

 #include <ESPAsyncTCP.h>
                         ^
compilation terminated.
In file included from src/esphome/components/api/list_entities.h:52:0,
                 from src/esphome/components/api/list_entities.cpp:1:
src/esphome/components/api/api_server.h:19:25: fatal error: ESPAsyncTCP.h: No such file or directory

*********************************************************************
* Looking for ESPAsyncTCP.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESPAsyncTCP.h"
* Web  > https://platformio.org/lib/search?query=header:ESPAsyncTCP.h
*
*********************************************************************

 #include <ESPAsyncTCP.h>
                         ^
compilation terminated.
In file included from src/esphome/components/api/api_server.cpp:1:0:
src/esphome/components/api/api_server.h:19:25: fatal error: ESPAsyncTCP.h: No such file or directory

*********************************************************************
* Looking for ESPAsyncTCP.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESPAsyncTCP.h"
* Web  > https://platformio.org/lib/search?query=header:ESPAsyncTCP.h
*
*********************************************************************

 #include <ESPAsyncTCP.h>
                         ^
compilation terminated.
In file included from src/esphome/components/api/subscribe_state.h:47:0,
                 from src/esphome/components/api/subscribe_state.cpp:1:
src/esphome/components/api/api_server.h:19:25: fatal error: ESPAsyncTCP.h: No such file or directory

*********************************************************************
* Looking for ESPAsyncTCP.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESPAsyncTCP.h"
* Web  > https://platformio.org/lib/search?query=header:ESPAsyncTCP.h
*
*********************************************************************

 #include <ESPAsyncTCP.h>
                         ^
compilation terminated.
*** [/data/multisensor/.pioenvs/multisensor/src/esphome/components/api/api_connection.cpp.o] Error 1
*** [/data/multisensor/.pioenvs/multisensor/src/esphome/components/api/api_server.cpp.o] Error 1
*** [/data/multisensor/.pioenvs/multisensor/src/esphome/components/api/list_entities.cpp.o] Error 1
*** [/data/multisensor/.pioenvs/multisensor/src/esphome/components/api/subscribe_state.cpp.o] Error 1
========================== [FAILED] Took 5.80 seconds ==========================
```

nmatt25, 2020-06-04:
same problem here .. hopefully an update will be pushed out

nmatt25, 2020-06-04:
`INFO Reading configuration /config/esphome/master_bedroom_blinds.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/master_bedroom_blinds
Processing master_bedroom_blinds (board: d1_mini; framework: arduino; platform: espressif8266@2.2.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Looking for ESPAsyncTCP-esphome library in registry
Warning! You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.
Looking for ESPAsyncWebServer-esphome library in registry
Warning! You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.
Warning! You are not connected to the Internet.
If you build a project first time, we need Internet connection to install all dependencies and toolchains.
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
Compiling /data/master_bedroom_blinds/.pioenvs/master_bedroom_blinds/src/esphome/components/api/api_connection.cpp.o
Compiling /data/master_bedroom_blinds/.pioenvs/master_bedroom_blinds/src/esphome/components/api/api_server.cpp.o
Compiling /data/master_bedroom_blinds/.pioenvs/master_bedroom_blinds/src/esphome/components/api/list_entities.cpp.o
Compiling /data/master_bedroom_blinds/.pioenvs/master_bedroom_blinds/src/esphome/components/api/proto.cpp.o
In file included from src/esphome/components/api/api_connection.h:7:0,
                 from src/esphome/components/api/api_connection.cpp:1:
src/esphome/components/api/api_server.h:19:25: fatal error: ESPAsyncTCP.h: No such file or directory

*********************************************************************
* Looking for ESPAsyncTCP.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESPAsyncTCP.h"
* Web  > https://platformio.org/lib/search?query=header:ESPAsyncTCP.h
*
*********************************************************************

 #include <ESPAsyncTCP.h>
                         ^
compilation terminated.
*** [/data/master_bedroom_blinds/.pioenvs/master_bedroom_blinds/src/esphome/components/api/api_connection.cpp.o] Error 1
In file included from src/esphome/components/api/api_server.cpp:1:0:
src/esphome/components/api/api_server.h:19:25: fatal error: ESPAsyncTCP.h: No such file or directory

*********************************************************************
* Looking for ESPAsyncTCP.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESPAsyncTCP.h"
* Web  > https://platformio.org/lib/search?query=header:ESPAsyncTCP.h
*
*********************************************************************

 #include <ESPAsyncTCP.h>
                         ^
compilation terminated.
*** [/data/master_bedroom_blinds/.pioenvs/master_bedroom_blinds/src/esphome/components/api/api_server.cpp.o] Error 1
In file included from src/esphome/components/api/list_entities.h:52:0,
                 from src/esphome/components/api/list_entities.cpp:1:
src/esphome/components/api/api_server.h:19:25: fatal error: ESPAsyncTCP.h: No such file or directory

*********************************************************************
* Looking for ESPAsyncTCP.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESPAsyncTCP.h"
* Web  > https://platformio.org/lib/search?query=header:ESPAsyncTCP.h
*
*********************************************************************

 #include <ESPAsyncTCP.h>
                         ^
compilation terminated.
*** [/data/master_bedroom_blinds/.pioenvs/master_bedroom_blinds/src/esphome/components/api/list_entities.cpp.o] Error 1
========================== [FAILED] Took 5.12 seconds ==========================`

firestrife23, 2020-06-04:
Same problem here, using ESPHome add-on for home assistant

anthonylavado, 2020-06-04:
There are two possible fixes mentioned above - you can either use `iptables` to redirect the request and let it succeed, or use the dev version of ESPHome.

jumblies, 2020-06-04:
for anyone running this in docker - the image you need is 
```esphome/esphome:dev```

dibyadip-das, 2020-06-04:
Now my ESP Home integration is working with Dev branch 👍 I am able to compile and upload the bins to respective nodes.

masonevans, 2020-06-04:
If you installed esphome via pip (for example if you're flashing locally instead of using the HA add-on) it does not seem like there's an easy way to switch to the dev version.

Instead, you can edit the platformio file directly.  Assuming you set up a virutal env located at `venv` within your project, edit the file `venv/lib/python3.7/site-packages/util.py`.

On line 382 you'll see the list of IPs.  You can change the second IP, on line 383, to `31.28.1.238`, and everything will work.

ssieb, 2020-06-04:
If you are using pip to install esphome, there's an answer at https://esphome.io/guides/faq.html#how-do-i-use-the-latest-bleeding-edge-version
I would recommend using "pip3" instead though, but it depends on what you used originally.
pip3 install https://github.com/esphome/esphome/archive/dev.zip

glmnet, 2020-06-04:
> From the Discord for Hass Add-On users:

Looks like hass people wants an update on ESPHome 

Where do you find this? seems like they know this issue already.

ivankravets, 2020-06-04:
Ivan from PlatformIO here. What is the reason this projected is fixed to 4.0.3? We recommend using the latest PIO Core version. We don't change our CLI API. It has not been changed since 2014 (the first release).

If you have any issues with the latest PIO Core, please report us and we would be glad to fix ASAP.

ivankravets, 2020-06-04:
@OttoWinter, please help here.

anthonylavado, 2020-06-04:
> Looks like hass people wants an update on ESPHome
> 
> Where do you find this? seems like they know this issue already.
@glmnet Sorry, I don't understand the question? I just found the reason why this isn't working. That's all.

glmnet, 2020-06-04:
Yes, it's crystal clear on your post. Sorry I rushed to ask. I pinned this issue.

innocentoreo, 2020-06-04:
Sorry if I'm posting this in the wrong place (e.g. if it's unrelated). I'm using ESPHome on an RPi 4. I installed it using docker. I had trouble executing the iptables listed above command on my Pi. I kept getting the message: "iptables v1.8.2 (nf_tables): unknown option "--to-destination". It took me a while fix this. The fix I used was to change the version of iptables from iptables-nft to iptables-legacy using the command: update-alternatives --config iptables. Then the command posted above worked. I was unable to fix the issue with a switch to the dev docker image. I think because I am using the esphome/esphome-armv7 image? I switched to esphome/esphome-armv7:dev and that did not fix the problem. Hopefully this information is useful for anyone in the same position to me.

germanotero, 2020-06-04:
I've have deployed the dev version, and it works..but it's dev

TheLastGimbus, 2020-06-04:
Mine dev doesn't work:
```
20-06-04 22:12:59 ERROR (SyncWorker_18) [supervisor.docker.addon] Can't build 15ef4d2f/armv7-addon-esphome-dev:dev: The command '/bin/bash -o pipefail -c pip3 install --no-cache-dir https://github.com/esphome/esphome/archive/dev.zip' returned a non-zero code: 2
```
please fix this asap and push to the main channel :(

glmnet, 2020-06-04:
It doesn’t work for armv7 (pi4) for some reason

glmnet, 2020-06-04:
Yet another workaround is to copy a working node  with the name of your new node. This way PlatformIO will use the cached libraries

TheLastGimbus, 2020-06-04:
It's so ironic that it's open source python code, but yet it's so dig up in all these containers and dockers etc that it's so hard to just edit it.

Is there any option that I could just... edit those two IPs that are checked myself? Through Hass file explorer?

nmatt25, 2020-06-04:
Dev version works for compiling nodes but OTA still fails with a 1 code

myoung34, 2020-06-05:
Using dev for me (from docker hub) results in

```
Processing front_porch_switch (board: esp01_1m; framework: arduino; platform: espressif8266@2.2.3)
--------------------------------------------------------------------------------
CorePackageManager: Installing tool-scons @ ~3.30102.0
Error: Detected unknown package 'tool-scons'
```

jnanadarshan, 2020-06-05:
Can't figureout for my config, please help.
Device - RPi4
OS - Raspbian (32bit), OpenMediaVault on top
Using Home-Assistant and Hassio on Docker. 
esphome is installed as a docker container through Supervisor tab

Tried pulling and replacing existing **esphome/esphome-hassio-armv7:1.14.3**	 with **esphome/esphome:dev** or **esphome/esphome-hassio-armv7:dev** didn't work

tried the iptables redirect (my Pi4 IP is 192.168.31.71 on port 8123) didn't work - sudo iptables -t nat -A OUTPUT -d 192.168.31.71 -j DNAT --to-destination 31.28.1.238

nilvanis, 2020-06-05:
> Tried the **iptables** redirect (my Pi4 IP is 192.168.31.71 on port 8123) didn't work -
> **sudo iptables -t nat -A OUTPUT -d 192.168.31.71 -j DNAT --to-destination 31.28.1.238**
> Is there any mistake in iptables command? The IP should be my Pi4 IP right?
No.
You should use the command as @anthonylavado pasted, that is:
> iptables -t nat -A OUTPUT -d 192.30.253.113 -j DNAT --to-destination 31.28.1.238

The reason behind this command is that apparently that a hardcoded IP 192.30.253.113 is no longer valid and 31.28.1.238 is.
Therefore this command 'translates' all requests sent to 192.30.253.113 and replaces destination address to 31.28.1.238. It is called DNAT (Destination Network Address Translation).

jumblies, 2020-06-05:
> It's so ironic that it's open source python code, but yet it's so dig up in all these containers and dockers etc that it's so hard to just edit it.

Docker is great for deployment; it doesn't work very well when you need to make a small change and have that persist.  This can be interpreted as disparaging docker or even esphome, but all of the cool HASSIO add ons wouldn't exist without docker, at least in a form that could be easily implemented.  

ESPHome is a little like napster.  It makes it so stupid easy to deploy nodes that is has been incredibly successful.  With that comes the loss of appreciation for just how far abstracted esphome is from the bare metal programming it replaced.  Shoot, my first MQTT node took a week worth of debugging to get stable and esphome takes 5 minutes.

Like I said above, it is very frustrating to have it fail, but @OttoWinter completely changed the game with esphome and I'm very grateful.

ivankravets, 2020-06-05:
I still do not understand what do you do. Also, I don't understand why do you do these hacks with `iptables`. This is an example where we try to put a "crutch" instead of fixing the root of the problem. The root of the problem is the next - THERE IS NO INTERNET CONNECTION on DEVICE/Docker Instance etc.

1. We recommend using the latest PlatformIO Core, which is 4.3.4 at this time.
2. Even if you use ancient 4.0.3 you should not have any problem. Please check [these lines](https://github.com/platformio/platformio-core/blob/v4.0.3/platformio/util.py#L381:L384) from 4.0.3 tag 
3. We improved an algorithm to detect Internet connection in the latest versions. See [new lines](https://github.com/platformio/platformio-core/blob/develop/platformio/util.py#L365:L371). That is what I said, we recommend using the latest PIO Core. It does not affect your dev-platforms and just improves stability and adds new features.
4. I don't see a reason to use Docker for PlatformIO Core. If you need to keep it in an isolated environment, please use the native Python Virtual Environment which is automatically supported if you use our [get-platformio.py](https://docs.platformio.org/en/latest/core/installation.html#installation-methods) script. It will install PIO Core into isolated env located in ~/.platformio/penv. You will see the final message and instructions from that installer.

I would like to help here but I still do not understand what do you do. The first thing which comes to my mind that is an incorrect running docker container. It should be run with a Docker bridge network. Again, I'm not sure that we need Docker here. We love Docker at PlatformIO and use it in our DC infrastructure but in this case, it looks like overheads. The same if you will use an AIRPLANE to visit a store in 5km from you.

jumblies, 2020-06-05:
@ivankravets I think, but can't say for sure, that docker is used for the larger hassio architecture which all runs in docker containers.

It's clear that this isn't a platformio problem. If you're still getting issues in PIO, you should just reference this thread and close them.

TheLastGimbus, 2020-06-05:
@ivankravets 
> Even if you use ancient 4.0.3 you should not have any problem. Please check these lines from 4.0.3 tag

I can ping the `"193.222.52.25"  # dl.platformio.org` ip with normal ping on my computer, but it neither of them works form Python's `socket` on port 80. Form my raw pc, not docker.
About Docker, I think it's like @jumblies said - it's better for HASS to just run every add-on in Docker.

I can ping IPs from new version (and github.com, of course) - both ping and `socket`
So ESPHome devs - plz just bump PIO version to newest on main channel and let me have my lights again :laughing:

ivankravets, 2020-06-05:
> So ESPHome devs - plz just bump PIO version to newest on main channel and let me have my lights again 

Could you make a PR? Where are these hardcoded lines?

TheLastGimbus, 2020-06-05:
Looks like someone already did bump it in "docker base", ~2hours ago:
https://github.com/esphome/esphome-docker-base/pull/2
https://github.com/esphome/esphome-docker-base/compare/v2.0.1...v2.1.0

I didn't contribute anything to HASS yet, so I don't really know what finally goes to users in HASS's add-on store, but it looks like those lines:
https://github.com/esphome/esphome/blob/dev/requirements.txt#L11
https://github.com/esphome/esphome/blob/master/requirements.txt#L12
https://github.com/esphome/esphome/blob/dev/setup.py#L43
https://github.com/esphome/esphome/blob/master/setup.py#L44

Are different in dev vs master branch. Would bumping them, and some version code somewhere, do the job?

anthonylavado, 2020-06-05:
As of 11:08 GMT today (2020-06-05), there is a new version of ESPHome released that appears to bump to PlatformIO Core 4.3.4: https://github.com/esphome/esphome/releases/tag/v1.14.4

This resolves the issue successfully. I'll close this issue by the end of the day, but we should keep it pinned so that everyone knows to update.

anthonylavado, 2020-06-05:
Hi @ivankravets, perhaps I can help shed some light here (though I am a new user of ESPHome).

> I still do not understand what do you do.

ESPHome (as viewed by somebody who just started using it 10 days ago) is a set of structured templates and tools that make it easy to program and manage a set of ESP* devices. It has a web interface that makes it very easy to watch over all your nodes in one place, and deploy changes to them as needed.

>Also, I don't understand why do you do these hacks with `iptables`. This is an example where we try to put a "crutch" instead of fixing the root of the problem. The root of the problem is the next - THERE IS NO INTERNET CONNECTION on DEVICE/Docker Instance etc.

But the issue is that there  *is* an internet connection in all of these cases. I am in Canada, and when I try to open a connection on Port 80 to 192.30.253.113, there is no response. It times out. If I try to open a connection to 193.222.52.25, it is flat out refused. To be safe, I've tried this across four different networks, with findings confirmed by others.

Thus, the sane fix at the moment is to make sure we can redirect to an address that works for the time being. Even if it is temporary. Depending on how some may have this installed, it's not always possible to get to the `util.py` and make the change manually.

> 1. We recommend using the latest PlatformIO Core, which is 4.3.4 at this time.
This is now updated, so the software is using 4.3.4. It's important to note that like much of other OSS, this is volunteer maintained so a newer release may have been held up by a few things.

> 2. Even if you use ancient 4.0.3 you should not have any problem. Please check [these lines](https://github.com/platformio/platformio-core/blob/v4.0.3/platformio/util.py#L381:L384) from 4.0.3 tag
See as above, connections on port 80 don't respond from the two listed IP addresses, at least in Canada.

> 3. We improved an algorithm to detect Internet connection in the latest versions. See [new lines](https://github.com/platformio/platformio-core/blob/develop/platformio/util.py#L365:L371). That is [...]
I wouldn't have the IP Addresses hardcoded myself, but just maybe a line to GitHub and api.platformio.org. After all, IP addresses may change, but the FQDNs *should* succeed.

> 4. I don't see a reason to use Docker for PlatformIO Core. If you need to keep it in an isolated environment, please use the native Python Virtual Environment which is automatically supported if you [...]
As @jumblies helpfully mentioned, it's not that people are using Docker specifically for PIO Core, but rather for ESPHome. It just happens to be that the last minor before today was published on [November 1st 2019](https://pypi.org/project/esphome/#history), at which time PIO Core was at 4.0.3 ([version history](https://pypi.org/project/platformio/#history)). Though 1.14.3 came later, the solution was tested and working, so why change if its not needed yet? Yes, this happened, but it *was* working.

> I would like to help here but I still do not understand what do you do. The first thing which comes to my mind that is an incorrect running docker container. It should be run with a Docker bridge [...]

Though I don't use them, Docker containers are wildly popular in the Home Assistant/Home Autmation world as a way to bring in working side systems as add-ons, which is very useful in a wide varied world of home hardware. It helps people who are running their system on everything from a Raspberry PI/SBC, to full systems.

I'm happy to keep chatting elsewhere if you'd like to know more 🙂. I'm @anthonylavado most places.

closed 2020-06-06 by anthonylavado