esphome/issues #1266: SIM800l configured but still unable to send or receive sms

Georgeibe, 2020-06-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESPHome (DEV) Vmware virtual enviroment, Docker
**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMCU 8266
**Affected component:**

https://esphome.io/components/sim800l.html

**Description of problem:**
I have configured SIM800L module according to guidelines but was not able to send or receive sms. I have tried different pins for TX RX and 115200 baud rate. 

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: node1
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "DOR"
  password: "XXXXXX"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Node1 Fallback Hotspot"
    password: "XXXXX"

captive_portal:

# Enable logging
logger:
   baud_rate: 0

ota:

# Enable Home Assistant API
api:
  services:
  - service: send_sms
    variables:
      recipient: string
      message: string
    then:
    - sim800l.send_sms:
        recipient: !lambda 'return recipient;'
        message: !lambda 'return message;'

uart:
  baud_rate: 9600
  tx_pin: TX
  rx_pin: RX

sim800l:
  on_sms_received:
    - lambda: |-
        id(sms_sender).publish_state(sender);
        id(sms_message).publish_state(message);
    - logger.log:
        format: "Received '%s' from %s"
        args: [ 'message.c_str()', 'sender.c_str()' ]

text_sensor:
- platform: template
  id: sms_sender
  name: "Sms Sender"
- platform: template
  id: sms_message
  name: "Sms Message"

```

**Logs (if applicable):**

```
[00:00:26][I][app:101]: ESPHome version 1.15.0-dev compiled on Jun  4 2020, 23:59:29
[00:00:26][C][wifi:415]: WiFi:
[00:00:26][C][wifi:283]:   SSID: [redacted]
[00:00:26][C][wifi:284]:   IP Address: 10.202.1.140
[00:00:26][C][wifi:286]:   BSSID: [redacted]
[00:00:26][C][wifi:287]:   Hostname: 'node1'
[00:00:26][C][wifi:291]:   Signal strength: -56 dB ▂▄▆█
[00:00:26][C][wifi:295]:   Channel: 6
[00:00:26][C][wifi:296]:   Subnet: 255.255.255.0
[00:00:26][C][wifi:297]:   Gateway: 10.202.1.1
[00:00:26][C][wifi:298]:   DNS1: 10.202.1.1
[00:00:26][C][wifi:299]:   DNS2: (IP unset)
[00:00:26][C][uart_esp8266:071]: UART Bus:
[00:00:26][C][uart_esp8266:073]:   TX Pin: GPIO12
[00:00:26][C][uart_esp8266:076]:   RX Pin: GPIO13
[00:00:26][C][uart_esp8266:078]:   Baud Rate: 9600 baud
[00:00:26][C][uart_esp8266:079]:   Bits: 8
[00:00:26][C][uart_esp8266:080]:   Parity: NONE
[00:00:26][C][uart_esp8266:081]:   Stop bits: 1
[00:00:26][C][uart_esp8266:085]:   Using software serial
[00:00:26][C][template.text_sensor:020]: Template Sensor 'Sms Sender'
[00:00:26][C][template.text_sensor:020]: Template Sensor 'Sms Message'
[00:00:26][C][logger:175]: Logger:
[00:00:26][C][logger:176]:   Level: DEBUG
[00:00:26][C][logger:177]:   Log Baud Rate: 0
[00:00:26][C][logger:178]:   Hardware UART: UART0
[00:00:26][C][sim800l:259]: SIM800L:
[00:00:26][C][sim800l:260]:   RSSI: 0 dB
[00:00:26][C][captive_portal:169]: Captive Portal:
[00:00:26][C][ota:029]: Over-The-Air Updates:
[00:00:26][C][ota:030]:   Address: node1.local:8266
[00:00:26][C][api:095]: API Server:
[00:00:26][C][api:096]:   Address: node1.local:6053
[00:00:27][D][sim800l:216]: Unhandled: +CPIN: READY - 0
[00:00:31][D][sim800l:216]: Unhandled: Call Ready - 0
[00:00:41][W][sim800l:062]: Not ack. 3 AT+CMGF=1
[00:00:41][D][sim800l:216]: Unhandled: OK - 0
[00:00:56][W][sim800l:062]: Not ack. 3 AT+CMGF=1
[00:00:56][D][sim800l:216]: Unhandled: ERROR - 0
[00:00:56][D][sim800l:216]: Unhandled: RDY - 0
[00:00:56][D][sim800l:216]: Unhandled: +CFUN: 1 - 0
[00:00:57][D][sim800l:216]: Unhandled: +CPIN: READY - 0
[00:01:01][D][sim800l:216]: Unhandled: Call Ready - 0
[00:01:02][D][sim800l:216]: Unhandled: SMS Ready - 0
[00:01:11][W][sim800l:062]: Not ack. 3 AT+CMGF=1
[00:01:11][D][sim800l:216]: Unhandled: OK - 0
[00:01:12][D][sim800l:216]: Unhandled: Call Ready - 0
[00:01:26][W][sim800l:062]: Not ack. 3 AT+CMGF=1
[00:01:26][D][sim800l:216]: Unhandled: ERROR - 0
[00:01:26][D][sim800l:216]: Unhandled: RDY - 0
[00:01:26][D][sim800l:216]: Unhandled: +CFUN: 1 - 0
[00:01:27][D][sim800l:216]: Unhandled: +CPIN: READY - 0
[00:01:31][D][sim800l:216]: Unhandled: Call Ready - 0
[00:01:41][W][sim800l:062]: Not ack. 3 SMS Ready
[00:01:41][D][sim800l:216]: Unhandled: AT+CMGF=1 - 0
[00:01:41][D][sim800l:216]: Unhandled: OK - 0
[00:01:47][I][ota:046]: Boot seems successful, resetting boot loop counter.
[00:01:56][W][sim800l:062]: Not ack. 3 AT+CMGF=1
[00:01:56][D][sim800l:216]: Unhandled: OK - 0
[00:01:58][D][sim800l:216]: Unhandled: Call Ready - 0
[00:02:00][D][sim800l:216]: Unhandled: SMS Ready - 0
[00:02:26][W][sim800l:062]: Not ack. 3 AT+CMGF=1
...

```

**Additional information and things you've tried:**
Switched TX and RX. Checked sim card from mobile phone,

glmnet, 2020-06-12:
Ahhh looks like your SIM800L is browning out, i.e. not enough power on the module and it reboots itself. 
Can you send a picture of your setup?

I have mine with a "big" voltage regulator, the switching ones which say are rated for 3Amps, the regulator input is 12V, and I believe it is configured at 3.7V

Georgeibe, 2020-06-13:
Thank you so much. I have switched to the 3A power supply and it works like a charm. I am supplying 5v to the SIM800 module.

glmnet, 2020-06-13:
5V seems too much. You should work se a regulable dc dc, that’s the easiest way. Set to 4v

closed 2020-06-14 by glmnet