esphome/issues #1267: SDS011 update_intervall can't be never

maurice-schleussinger, 2020-06-05:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

newest Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266

**Affected component:**

SDS011 https://esphome.io/components/sensor/sds011.html

**Description of problem:**
The sensor component SDS011 has config variable `update_interval` which has the configuration type time(https://esphome.io/guides/configuration-types.html#config-time). If I set it to `never`the error below occurs.

(Context: I want to disable the sensor during the night and handle updates myself via UART commands.)

**Problem-relevant YAML-configuration entries:**
```yaml
uart:
  rx_pin: D6
  tx_pin: D7
  baud_rate: 9600

sensors:
- platform: sds011
    update_interval: never
    pm_2_5:
      name: "Feinstaub PM 2,5 (pm25) Schlafzimmer"
    pm_10_0:
      name: "Feinstaub PM 10 (pm10) Schlafzimmer"
```

**Logs (if applicable):**

```
INFO Reading configuration /config/esphome/schlafzimmer_bme280.yaml...
INFO Detected timezone 'CET' with UTC offset 1 and daylight savings time from 28 March 02:00:00 to 31 October 03:00:00
Failed config

sensor.sds011: [source /config/esphome/schlafzimmer_bme280.yaml:52]
  platform: sds011
  
  Unknown value 'never', valid options are 'us', 'microseconds', 'ms', 'milliseconds', 's', 'sec', 'seconds', 'min', 'minutes', 'h', 'hours', 'd', 'days'.
  update_interval: never [source /config/esphome/schlafzimmer_bme280.yaml:53]
  pm_2_5: 
    name: Feinstaub PM 2,5 (pm25) Schlafzimmer
  pm_10_0: 
    name: Feinstaub PM 10 (pm10) Schlafzimmer

```

**Additional information and things you've tried:**

maurice-schleussinger, 2020-06-05:
Okay, so the interval has to be between 1-30 minutes. Any other value is not valid for this sensor. 
Can we change the wording in the documentation from 

> If update_interval is between 1-30 minutes, the SDS011 periodically turns on for 30s before each measurement. 

to something like "update_interval must be between 1-30 minutes…"?

napieraj, 2020-10-03:
the issue arises from the update interval being pushed to the SDS011 register itself, as such `never` is not a valid value, the documentation or code would need to be updated to either better clarify or handle manual polling as you request

closed 2020-10-03 by napieraj