esphome/issues #1269: Upgrade to 1.14.4 cause ESP32 to lockup with DHT11 sensor

faspina, 2020-06-05:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32/NodeMcu32

**Affected component:**
DHT11 

https://esphome.io/components/sensor/dht.html

**Description of problem:**
DHT11 sensor was working with 1.14.3 , upgraded to 1.14.3 , NODEMCU lockedup and unresponsive,  took out the following code and it worked. Also simplified the DHT sensor and took out the filters, same result. Tried different GPIO , same result.  Works fine on my original nodemcu  

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
  - platform: dht
    model: DHT11
    pin: GPIO21
    update_interval: 30s     
    temperature:
      name: "Office"
      filters:
        - filter_out: nan
        - heartbeat: 15s
        - sliding_window_moving_average:
            window_size: 2
            send_every: 2
            send_first_at: 1
        - lambda: return x * (9.0/5.0) + 32.0; # report temperature in Fahrenheit
      unit_of_measurement: "°F"
    humidity:
      name: "Office Humidity"
      filters:
        - filter_out: nan
        - heartbeat: 15s
        - sliding_window_moving_average:
            window_size: 2
            send_every: 2
            send_first_at: 1
      accuracy_decimals: 1  
      

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

faspina, 2020-06-05:
Should be noted this happens on two different devices since 1.14.4

johnboiles, 2020-06-05:
I'm also getting an ESP32 lockup with the DHT22 after upgrading to 1.14.4. I pulled my ESP32 down from the shelf, plugged in a USB cable, and uploaded the esphome firmware again to get the logs.

@faspina could you connect with a usb cable and try to re-install your esphome firmware and post the logs? That will help us be certain you and I are seeing the same issue. I think it's very likely that it's the exact same thing.

Downgrading to 1.14.3 and reinstalling solves the issue for me.

[This](https://github.com/johnboiles/homeassistant-config/blob/master/esphome/air_sensor.yaml) is the esphome yaml I'm using.

<details>
  <summary>Click to expand logs!</summary>

```  
> esphome air_sensor.yaml run
INFO Reading configuration air_sensor.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d air_sensor
Processing air_sensor (board: nodemcu-32s; framework: arduino; platform: espressif32@1.12.1)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif32 3.10004.200129 (1.0.4)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
Dependency Graph
|-- <AsyncTCP-esphome> 1.1.1
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <FS> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.6
|   |-- <AsyncTCP-esphome> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <Wire> 1.0.1
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
Compiling .pioenvs/air_sensor/src/main.cpp.o
Linking .pioenvs/air_sensor/firmware.elf
Retrieving maximum program size .pioenvs/air_sensor/firmware.elf
Checking size .pioenvs/air_sensor/firmware.elf
Building .pioenvs/air_sensor/firmware.bin
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  13.2% (used 43216 bytes from 327680 bytes)
Flash: [======    ]  55.6% (used 910354 bytes from 1638400 bytes)
========================================================================== [SUCCESS] Took 6.90 seconds ==========================================================================
INFO Successfully compiled program.
Found multiple options, please choose one:
  [1] /dev/cu.usbserial-0001 (CP2102 USB to UART Bridge Controller - CP2102 USB to UART Bridge Controller)
  [2] Over The Air (air_sensor.local)
(number): 1
INFO Running:  platformio run -d air_sensor -t upload --upload-port /dev/cu.usbserial-0001
Processing air_sensor (board: nodemcu-32s; framework: arduino; platform: espressif32@1.12.1)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif32 3.10004.200129 (1.0.4)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
Dependency Graph
|-- <AsyncTCP-esphome> 1.1.1
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <FS> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.6
|   |-- <AsyncTCP-esphome> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
[... 144 lines cut ...]
[08:25:04]Backtrace: 0x4012df2b:0x3ffaf6a0 0x4012e01e:0x3ffaf760 0x4012fca5:0x3ffaf790 0x4012fd3b:0x3ffaf800 0x4015000d:0x3ffaf850 0x4015110e:0x3ffaf870 0x4015111d:0x3ffaf8b0 0x401168dd:0x3ffaf8d0 0x40116cec:0x3ffaf910 0x401179dd:0x3ffaf980 0x40117a96:0x3ffaf9d0 0x40112fa5:0x3ffafa20 0x401132c0:0x3ffafa70 0x401161c9:0x3ffafa90 0x401161ea:0x3ffafab0 0x40115da8:0x3ffafad0 0x40116072:0x3ffafaf0 0x4009034f:0x3ffafb10 0x40089915:0x3ffafb50
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x4012df2b: hmac_sha1_vector at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/wpa_supplicant/src/crypto/sha1.c:70 (discriminator 1)
WARNING Decoded 0x4012e01e: hmac_sha1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/wpa_supplicant/src/crypto/sha1.c:111
WARNING Decoded 0x4012fca5: pbkdf2_sha1_f at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/wpa_supplicant/src/crypto/sha1-pbkdf2.c:54
WARNING Decoded 0x4012fd3b: pbkdf2_sha1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/wpa_supplicant/src/crypto/sha1-pbkdf2.c:91
WARNING Decoded 0x4015000d: wpa_set_bss
WARNING Decoded 0x4015110e: wpa_config_bss
WARNING Decoded 0x4015111d: wpa_sta_connect
WARNING Decoded 0x401168dd: cnx_sta_connect_led_timer_cb
WARNING Decoded 0x40116cec: cnx_sta_connect_led_timer_cb
WARNING Decoded 0x401179dd: cnx_remove_all_rc
WARNING Decoded 0x40117a96: cnx_start_handoff_cb
WARNING Decoded 0x40112fa5: clear_bss_queue
WARNING Decoded 0x401132c0: scan_inter_channel_timeout_process
WARNING Decoded 0x401161c9: chm_end_op
WARNING Decoded 0x401161ea: chm_end_op_timeout_process
WARNING Decoded 0x40115da8: wifi_station_stop
WARNING Decoded 0x40116072: ieee80211_timer_do_process
WARNING Decoded 0x4009034f: ppTask
WARNING Decoded 0x40089915: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
[08:25:05]
[08:25:05]Rebooting...
```
</details>

faspina, 2020-06-05:
I am running ESPhome on docker on a unraid server, so I have to plug a cable in and pull my nodemcu in . may take me a bit to get around to it its probably the same issue

mario-tux, 2020-06-05:
The same here with DHT2 on ESP32 working with previous release.

mbo18, 2020-06-06:
Same here, ESP32, DHT22, ESPHome 1.14.4. Disabling the DHT sensor has solved this issue but no more temp and humidity report :(

asadenton, 2020-06-07:
I am seeing this too. 1.14.4 works fine on most of my devices but the one that has a DHT and Dallas is failing if I include either of those in the config. I am getting the same error message as above :"Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)". I think [1276](https://github.com/esphome/issues/issues/1276) is also related

csmart, 2020-06-07:
See https://github.com/esphome/issues/issues/1276#issuecomment-640146210

Gymnae, 2020-06-07:
Even though the devices did not appear in home assistant nor esphome I was able to flash them without DHT11 sensor code via OTA and bring them back to properly reporting themselves. 
This could be a workaround for the moment.

glmnet, 2020-06-14:
Duplicate of #1276

closed 2020-06-14 by glmnet