esphome/issues #1270: Incorrect icon for battery sensor

kloggy, 2020-06-05:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

  platform: ESP32
  board: mhetesp32devkit
**Affected component:**

  - platform: xiaomi_cgg1

**Description of problem:**
The battery sensor created by ESPHome in HA has an mdi icon assigned to it which means that adding
`device class: battery`
in the `customize` section for the sensor doesn't work. By which I mean Lovelace does not display the icon adapted to the percentage charge remaining.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
  - platform: xiaomi_cgg1
    mac_address: xxxxx
    temperature:
      name: "Kitchen Temperature"
    humidity:
      name: "Kitchen Humidity"
    battery_level:
      name: "Kitchen Battery Level"
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

kloggy, 2020-10-04:
Still an issue as far as I am aware

closed 2021-02-15 by stale[bot]

roooodcastro, 2021-05-13:
This is still an issue, it's also happening with me. Can this issue please be opened again?

glmnet, 2021-05-14:
Can you try with latest version?