esphome/issues #1275: upgrading to 1.14.4 killed sonoff basic wifi

pauldogg, 2020-06-06:
Upgrading sonoff basic to 1.14.4 causes wifi to stop working. device seems to work, but captive portal and normal wifi no longer seem to work
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

hassio 0.110.5 rpi4 esphome 1.14.4
**ESP (ESP32/ESP8266, Board/Sonoff):**

sonoff basic defined as   platform: ESP8266
  board: esp01_1m
**Affected component:**

https://esphome.io/changelog/v1.14.0.html

**Description of problem:**
Upgrading to esphome 1.14.4 from 1.14.3 caused wifi to no longer connect, and captive portal does not kick in.
Re-uploading my previously saved bin of the 1.14.3 compile, solved the issue
**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

esphome:
  name: fountain_01
  platform: ESP8266
  board: esp01_1m
#  board: esp8285
#  arduino_version: 2.4.2

wifi:
  ssid: "XXX"
  password: "XXX"
#  output_power: 20dB

  # Optional manual IP
  manual_ip:
    static_ip: 192.168.X.XXX
    gateway: 192.168.X.X
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "fountain_01"
    password: "XXX"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "XXX"

ota:
  password: "XXX"

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "Hummingbird Fountain button"
    on_press:
      then:
        - switch.toggle: fountain01_relay
    on_double_click:
      then:
        - light.toggle: fountain01_light
#      then:
#        - homeassistant.service:
#            service: light.turn_off
#            data:
#              entity_id: group.all_lights

switch:
  - platform: restart
    name: "Hummmingbird Fountian Restart"

  - platform: gpio
    name: "Hummingbird Fountain pump"
    pin: GPIO12
    id: fountain01_relay

output:
  - platform: esp8266_pwm
    pin: GPIO14
    id: redgpio
  - platform: esp8266_pwm
    pin: TX
    id: greengpio
  - platform: esp8266_pwm
    pin: RX
    id: bluegpio

status_led:
  pin:
    number: GPIO13
    inverted: yes

light:
  - platform: rgb
    name: "Hummingbird Fountain Light"
    red: redgpio
    green: greengpio
    blue: bluegpio
    id: fountain01_light

    effects:
      - random:
      - strobe:
      - flicker:

time:
  - platform: homeassistant

sun:
  latitude: XXX
  longitude: XXX

  on_sunrise:
    - then:
        - light.turn_off:
#            brightness: 0%
            id: fountain01_light
            transition_length: 300s

        - switch.turn_on:
            id: fountain01_relay

  on_sunset:
    - then:
        - light.turn_on:
            brightness: 100%
            id: fountain01_light
            transition_length: 300s

        - delay: 3h
        - switch.turn_off:
            id: fountain01_relay
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

I thought this could be related to the wifi output issue discussed here
https://github.com/esphome/issues/issues/455#issuecomment-553247563

but changing the output power did not fix this.
Any help would be greatly appreciated.
Thanks

smoke007, 2020-06-09:
I have many Sonoff Basics and have had this problem for a while.  The only way I've been able to stop this issue is by disabling #logger:, I usually do a Clean Build, then Compile, and Flash with a programmer.  I was hoping this issue might go away in 1.14.4, but I just verified this issue is still present for me on my Sonoff Basics.

glmnet, 2020-06-12:
These issues were correlated to a wifi tx power, you might try experimenting with the value there

pauldogg, 2020-06-13:
> Edit - I was able to successfully update all 19 Sonoff Basic R2 from 1.14.3 to 1.14.4, while Logger was disabled, without killing the wifi. 

I was able to update to 1.14.4 if I comment out logger.

Thanks for that!

closed 2020-06-13 by pauldogg